DJ2LS/FreeDATA #60: Support for translations

DJ2LS, 2021-08-18:
Support for different languages in a separate file

DJ2LS, 2023-03-23:
https://www.npmjs.com/package/i18n-js

DJ2LS, 2023-08-06:
@Mashintime I'd like to work on this, soon. Are you interested, working on this?

Mashintime, 2023-08-06:
A previous project I've worked on implemented this, and personally feel it's a giant waste of time.  Unless electron/node implement it in an easier way I personally have no interest.  Feel this may be better for a 2nd beta or initial release as things are still pretty fluid development wise.

DJ2LS, 2023-08-06:
I wouldn't call it a complete waste of time as long as many people in the world are not speaking English well enough. 
Nevertheless priority isn't too high at this - it's more because it's the oldest issue here. I'm wondering how's the complexity of i18n 🤔

DJ2LS, 2023-09-30:
closing this one now as its not making sense enough. Better focusing on other topics

petrkr, 2025-02-24:
I was looking if someone already no requested this...

Because my friend does not know English and asked why it is not translated. So maybe after 2 years you can reopen with low priority, but keep it at mind.

For me is English fine, but in Czech because of Russian invasion between 68-89 they forced learn Russian language and forbidden learn English, thus lot of 50+ years old HAMs does not know English more than "hello" or "thank you", which make complicated to overtalk them to use this application

DJ2LS, 2025-02-25:
@petrkr thanks for your comment. The biggest problem is time. Would you be interested in working on this?

petrkr, 2025-02-25:
If there will be already some i18n framework with po translation files, then rewritting text could be done step-by-step, not all texts and all translations at once.

My problem is I really do not know JavaScript nor vue. So if you can do in limited time some skeleton like 'here instedad text you will use _translate('id-texr', 'defaultvalue') then anyone can help by pull requests.

DJ2LS, 2025-02-25:
So you are talking about an implementation example?

petrkr, 2025-02-25:
Yes.

Include library, show example 1 string usage and en.po / en.mo 

Rest can be done by others step-by-step.

I can also do that, but since I do not know vue framework it will be much more time for me just about learning framework itself and then implementation

DJ2LS, 2025-02-25:
Okay, I think I can do this. I will reopen this issue, but I really need your help then :-)

LA3QMA, 2025-02-25:
I did some research today to see if this is something I could try to start
playing around with if I get someone to help.

Using localisation the translations are in different JSON files where
english is default/fallback.

Also using Pythons built-in "gettext" can be used but then I think this has
to be compiled into .mo files.

Anyhow both methods make it relatively easy for other users to help by pull
requests.

The JSON method once adopted requires less work/maintenance.
The "proper" way using i.e "gettext" requires the compilation of the .mo
files.

Both the server and the GUI need this and JSON/"gettext" is both
supported for javascript.

DJ2LS, 2025-02-25:
I think we should separate this work - GUI and Server. While the server can stay for now in English, it might make more sense for the GUI.

DJ2LS, 2025-02-25:
the branch "ls-i18n" contains a first working version. Have a look in gui settings how to do this. 
At least you have to do the following things:

- Create a new language file with <iso>_<name>.json like en_English.json
- create new variable files inside templates like {{ $t('settings_enable') }}
- create variable inside language. 

Notes:

- Language files will be detected automatically. 
- Let's start with English settings first. Please think about a good naming. like <categorie>_<setting or whatever>
- As soon as we have all variables created in English, we can continue with the specific language files, as the variable names are static.

petrkr, 2025-02-25:
Do not develop wheel again, use .po anf .mo with i18n

Because there are lot of already made software aimed for translations and also we can put to some opensource community.

If you will do it is some own format, noone else can edit it and it will be lot of pain to keep format.

Server so think does not need translations.

Poedit for local

https://weblate.org/en/ for cooperation

DJ2LS, 2025-02-25:
So question is - can we use i18next or another i18 package for doing this easily out of the box

petrkr, 2025-02-25:
I littlebit investigate how Gettext works and it grab data for .pot files from source codes. Do not know how make it able to create .pot from Vue sources yet. But since it get strings from soruce codes it will be always up to date, because you will not forgot to add them by hand.

Now I working on secure element from tropicsquare, so I will take a look later if you both will not be faster

DJ2LS, 2025-02-26:
[doing some research, it seems, that some of the Gettext ](https://www.npmjs.com/package/gettext-parser) This library is doing all we need it seems - converting Gettext to json. It seems, most i18n libraries are using json as their data format. But we could use such a tool for using po files, instead.

DJ2LS, 2025-02-26:
I meanwhile also see the benefits of Gettext files. Thanks for mentioning, @petrkr

DJ2LS, 2025-02-26:
Well, the po convertion really bloats code it seems. Most easy and simple is still using json.

DJ2LS, 2025-02-26:
using the tool `i18next-conv` seems to help here.
`i18next-conv -l de -s de_Deutsch.json -t de_Deutsch.po ` converts to po file from json and back via `i18next-conv -l de -s de_Deutsch.po -t de_Deutsch.json`

DJ2LS, 2025-02-28:
We should use nested keys:

`{
    "settings": {
        "audio": {
            "quality": "Audioqualität"
        }
    }
}`

`{{ t('settings.audio.quality') }}`

closed 2025-03-10 by DJ2LS