DJ2LS/FreeDATA #96: Trouble with Ubuntu 20.04

frspin, 2021-12-26:
With Ubuntu 20.04 AppImage don't show any audio device.
Downloading source from GitHub and using Wiki instructions give an error in libcodec2:

/lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.32' not found

Downloading libcoded2 as in optional step in Wiki instructions show a different error:

codec2/build_linux/src/libcodec2.so.1.0: undefined symbol: fdmdv_8_to_48_short

and Hamlib library used in Ubuntu 20.04 is old for FreeDATA TNC!

How can I try this new modem?

Regards

Franco Spinelli
IW2DHW

DJ2LS, 2021-12-26:
@frspin , thanks for your opened issue! 
I'm doing a minor change of the code, actually, so I'm not surprised a lot of things are not working. 
Great to see, you downloaded the source code and running FreeDATA manually, this helps me  a lot!
Please delete and download codec2 again. https://github.com/drowe67/codec2/pull/265 has been merged some hours ago. This should solve the problem with `codec2/build_linux/src/libcodec2.so.1.0: undefined symbol: fdmdv_8_to_48_short`.

Please let me know, if you need any further help!

DJ2LS, 2021-12-26:
`and Hamlib library used in Ubuntu 20.04 is old for FreeDATA TNC!` this is beacause of the old hamlib version which is shipped with ubuntu 20.04 . If you don't have a really new radio, this shouldnt be a problem.

frspin, 2021-12-26:
Downloaded new version of libcodec2 and modified codec2.py for search codec2 library only in downloaded version.
Codec2 error is gone and waterfall is showing now.

For hamlib I get no model in pull down choice. I have typed "123" which is model number for my Yaesu 897D in Ubuntu 20.04 hamlib version.

Error now is :

2021-12-26 16:43.57 [error    ] [DMN] Hamlib: rig not supported...
2021-12-26 16:43.57 [error    ] [TNC] Hamlib - can't open rig  e=<class 'NotImplementedError'>

Testing rig setting I get now this errors:

[{'deviceid': '123', 'deviceport': '/dev/ttyUSB0', 'pttprotocol': 'RIG', 'pttport': '/dev/ttyUSB0', 'serialspeed': '38400', 'pttspeed': '38400', 'data_bits': '8', 'stop_bits': '1', 'handshake': 'None'}]
2021-12-26 15:51:19 [error    ] [DMN] Hamlib: rig not supported...
Wrong number or type of arguments for overloaded function 'Rig_set_conf'.
  Possible C/C++ prototypes are:
    Rig::set_conf(token_t,const_char_string)
    Rig::set_conf(char const *,char const *)

2021-12-26 15:51:19 [error    ] [DMN] Hamlib: Can't open rig   e=<class 'NotImplementedError'>

See attached screenshot for empty drop down choice  of rig

Regards

Franco Spinelli
IW2DHW
[image]

frspin, 2021-12-26:
I correct my previous message.
Typing RIG a drop-down list appear, without scroll bar, and I am able to get RIG_MODEL_FT897.

Now rig open OK

I suppose that "Rig" setting for PTT mean "CAT controlled"

Now that TNC is running, how can I test it?
There is a frequency where there are working signals or a file to try?

Regards

Franco Spinelli
IW2DHW

DJ2LS, 2021-12-26:
Great to see you solved the problem.
There is no frequency, because you are the first one trying to test it. But its normal data. So if you want you could send on 7040 and maybe I can receive you ( I have a bad antenna here.... ).

DJ2LS, 2021-12-26:
yes, RIG should be something like cat. Its mostly used, if you dont have usb device between you radio and laptop.
You can click the "test button" . It will become green if its working and PTT should toggle.

frspin, 2021-12-26:
OK

I will try examples in test directory and become acquainted  with modem and PTT.

After this I will try "on air" and write you a message for receiving my CQ CQ CQ

Regards

Franco Spinelli
IW2DHW

DJ2LS, 2021-12-26:
you could also use the beacon mode which is a little bit more comfortable

frspin, 2021-12-26:
OK

Now I need to go out but maybe tomorrow I will send you a message and transmit a beacon on 7040.
I am in JN45op and I have a 40 mt dipole oriented est-west.

Franco Spinelli
IW2DHW

DJ2LS, 2021-12-26:
@frspin I changed a lot of code today and merged a #95 , so please do a `git pull` before starting the tnc next time so everything is up to date. Please let me know if any more problems occured with ubuntu 20.04 and feel free opening a [discussion](https://github.com/DJ2LS/FreeDATA/discussions)

frspin, 2021-12-26:
> @frspin <https://github.com/frspin> I changed a lot of code today and
> merged a #95 <https://github.com/DJ2LS/FreeDATA/pull/95> , so please do
> a |git pull| before starting the tnc next time so [...]

Some problems with rig settings:

- audio port is not saved between sessions
- ttyUSB0 for rig control not saved between sessions
- what is setup for CAT control in PTT port?
- testing rig setup give a "yellow" result with:

[{'devicename': 'RIG_MODEL_FT897', 'deviceport': '/dev/ttyUSB0',
'pttprotocol': 'RIG', 'pttport': '/dev/ttyUSB0', 'serialspeed': '38400',
'pttspeed': '38400', 'data_bits': '8', 'stop_bits': '1', 'handshake':
'None'}]
2021-12-26 19:34:52 [info     ] [DMN] Hamlib device openend
status=SUCCESS
2021-12-26 19:34:52 [warning  ] [DMN] Hamlib PTT               status=NO
SUCCESS

- setting 'pttport' to  'n/a [26a9] don't change anything. Parameter
  remain at "/dev/ttyUSB0'
- obviously no beacon because there is non PTT

Other dialog with rig (frequency, mode) is correct

Regards

Franco Spinelli
IW2DHW

DJ2LS, 2021-12-26:
Settings will be saved when starting the tnc itself. But I'm thinking about saving this immediately.
The audio devices will not be saved at the moment, because they are normally changing a lot. But I will add this to the todo list! 

Hamlib integration is a little bit tricky. 
So it seems your device can be opened, but ptt seems not to toggle. How do you normally use your radio and ptt? With wsprx for example? And how do you connect it with your computer?

I need to optimise the Hamlib Integration with additional settings and will try doing this tomorrow.

frspin, 2021-12-27:
Il 26/12/21 21:52, DJ2LS ha scritto:
> Settings will be saved when starting the tnc itself. But I'm thinking
> about saving this immediately.

Yes, all setting (but no audio device) is saved at tnc start. This is
working.

> The audio devices will not be saved at the moment, because they are
> normally changing a lot. But I will add this to the todo list!

If you can save device name and not device number, it will be consistent
between runs.

> 
> Hamlib integration is a little bit tricky.
> So it seems your device can be opened, but ptt seems not to toggle. How
> do you normally use your radio and ptt? With wsprx for example? And how
> do you connect it with your computer?

I use my run with Fldigi (but with rigctl) and wsjt-x. But wsjt-x run a
builtin version of hamlib.
In other test done with hamlib I have found that Python3 port of Ubuntu
20.04 version there is some problems. I can get frequency but I can't
set frequency.
I will do some more test with python 3.8 and hamlib

> 
> I need to optimise the Hamlib Integration with additional settings and
> will try doing this tomorrow.

In another my python project I have used a rigctl and subprocess for
controlling my rig due to previous problem with hamlib and python.

Regards

Franco Spinelli
IW2DHW

frspin, 2021-12-27:
Some results with hamlib used from Python.

It seems that 3.3 version is not working from python 3.8

I can define rig, open it, get info for frequency and also PTT. 
But  I can't "set" any parameter.

>>> my_rig.get_ptt()
rig_get_ptt called
ft897: cache invalid
serial_flush called
write_block called
write_block(): TX 5 bytes
0000    00 00 00 00 f7                                      .....           
read_block called
read_block(): RX 1 bytes
0000    ff                                                  .               
0

but
>>> my_rig.set_ptt(Hamlib.RIG_PTT_RIG,1)
rig_set_ptt called
>>> my_rig.set_ptt(Hamlib.RIG_PTT_RIG,0)
rig_set_ptt called

and no PTT on rig!

Also, in rig.py file, there is a 

            # set rig mode to USB
            self.my_rig.set_mode(Hamlib.RIG_MODE_USB)
            return True

and

    def set_mode(self, mode):
        return 0

But for my rig, mode for CAT control need to be PKTUSB but, also setting it, no PTT at all.

So it seem that hamlib 3.3 is unusable from Python 3.8

Regards

Franco Spinelli
IW2DHW

DJ2LS, 2021-12-27:
@frspin thanks for  having closer look at Hamlib.
Some month ago I did tests with rigctl and rigctld as well, but decided staying at the python binding because its a little bit easier to control in my opinion. The core problem is, that there's only an outdated version available for Ubuntu 20.04 and some other distributions like Debian or Suse. The future idea is, using the Hamlib API directly. But for this I need some more help. This is the only way which is stable for the user without the need for installing additional software, I think. 
So I'm not surprised there are problems with python and the Hamlib python binding.

If you want, you can install the latest Hamlib version from source. I wrote a guide which is in the wiki. 

I also rewrote the rig.py file some days, so I need to update a lot of the code as well (specially the advanced settings). I will do this step by step the next days.

Thanks for your help, Franco!

frspin, 2021-12-27:
Il 27/12/21 12:02, DJ2LS ha scritto:
> @frspin <https://github.com/frspin> thanks for having closer look at Hamlib.
> Some month ago I did tests with rigctl and rigctld as well, but decided
> staying at the python binding because its a [...]

C and C++ version of hamlib 3.3 is correctly working so i think that a
interim solution can be a usage of libhamlib.so with ctypes as already
you do with libasound and libcodec2.
You need only open(), set_mode() and set_ptt() functions I suppose.

With this "interface" to libhamlib.so you are free from different
versions and bindings. And, if needed, you can provide a libhamlib.so as
you already are doing for libcodec2

> 
> If you want, you can install the latest Hamlib version from source. I
> wrote a guide which is in the wiki.

I have downloaded 4.4 version and compiled it but I will test it before
install and this is not possible (at least as far as I know).
If I change hamlib version, removing repo version, I need to rebuild
many other programs, as qsstv, direwolf, fldigi and many other and I do
this step only if I am safe with new hamlib version.

> 
> I also rewrote the rig.py file some days, so I need to update a lot of
> the code as well (specially the advanced settings). I will do this step
> by step the next days.
> 
> Thanks for your help, Franco!

Keep me informed on your progress!
I will monitor your GitHub repo

Regards

Franco Spinelli
IW2DHW

DJ2LS, 2021-12-27:
> C and C++ version of hamlib 3.3 is correctly working so i think that a interim solution can be a usage of libhamlib.so with ctypes as already you do with libasound and libcodec2. You need only [...]

Yes, I already did some tests with ctypes and libhamlib.so but got stuck. I also created an issue for this #76 #71 
But this is a topic where I need to focus on, I guess.

> I have downloaded 4.4 version and compiled it but I will test it before
install and this is not possible (at least as far as I know).
If I change hamlib version, removing repo version, I need to rebuild
many other programs, as qsstv, direwolf, fldigi and many other and I do
this step only if I am safe with new hamlib version.

I totally understand you! 

> Keep me informed on your progress! I will monitor your GitHub repo

Audio and USB devices will now be saved to config and selected automatically. I also updated the hamlib advanced settings and added a "USB" option.

frspin, 2021-12-27:
Il 27/12/21 13:32, DJ2LS ha scritto:
> 
> Yes, I already did some tests with ctypes and libhamlib.so but got
> stuck. I also created an issue for this #76
> <https://github.com/DJ2LS/FreeDATA/issues/76> #71
> <https://github.com/DJ2LS/FreeDATA/issues/71>
> But this is a topic where I need to focus on, I guess.
> 
> 
> Audio and USB devices will now be saved to config and selected
> automatically. I also updated the hamlib advanced settings and added a
> "USB" option.

Now TNC start only with some sound card selected for input.

I have 4 sound card:

- a HDA Intel PCA on main board
- a Ensoniq AudioPCI
- a Webcam USB
- a USB Creative

Selecting any sound card for input save it and start TNC but not for USB
Creative, connected to my rig!

Selecting this sound card TNC don't start and so no possibility to save it.

After this in config.json file there is some mistake:

{
  "tnc_host": "192.168.178.163",
  "tnc_port": "3000",
  "daemon_host": "192.168.178.163",
  "daemon_port": "3001",
  "mycall": "IW2DHW",
  "mygrid": "JN45op",
  "deviceid": "RIG_MODEL_FT897",
  "deviceport": "/dev/ttyUSB0",
  "serialspeed": "38400",
  "ptt": "RTS",
  "spectrum": "scatter",
  "tnclocation": "localhost",
  "rx_audio": "pulse [7db9]",
  "tx_audio": "front [cb24]",
  "pttprotocol": "RIG",
  "pttport": "/dev/ttyS4",
  "data_bits": "8",
  "stop_bits": "2",
  "handshake": "None"
}

And this config don't change if I change audio card setup and start TNC

Regards

Franco Spinelli
IW2DHW

DJ2LS, 2021-12-27:
okay, I found the problem with the wrong audio card and hopefully fixed it. 
The problem with "USB Creative" sounds interesting :thinking: 
I have a lot of problems with ubuntu automatically selecting audio devices. You could check ubuntus audio settings to see, if NOT a needed audio device is selected.

frspin, 2021-12-27:
Now audio device is correctly saved and also "USB Creative" is working.

Back to rig setting in Advanced setting I set 8 bit, 2 stop bit a no handshake and this settings are saved.
If I test my rig using "Test settings" button I get this message on daemon window:

[{'devicename': 'RIG_MODEL_FT897', 'deviceport': '/dev/ttyUSB0', 'pttprotocol': 'RIG', 'pttport': '/dev/ttyUSB0', 'serialspeed': '38400', 'data_bits': '8', 'stop_bits': '1', 'handshake': 'None'}]

with wrong  'stop_bits': '1',

Closing TNC and closing daemon don't change anything.

After starting TNC GUI and pressing "Beacon" button I get this error message in daemon window:

############ START OF ERROR #####################
SOCKET COMMAND ERROR: {"type" : "BROADCAST", "command" : "START_BEACON", "parameter": "","timestamp" : 1640616686515}
<class 'ValueError'>
<class 'ValueError'> sock.py 106
############ END OF ERROR #######################

Ping and CQ don't show any error message but, as usual, "rig don't transmit" as per hamlib error!

Regards

Franco Spinelli
IW2DHW

DJ2LS, 2021-12-27:
> ############ START OF ERROR #####################
SOCKET COMMAND ERROR: {"type" : "BROADCAST", "command" : "START_BEACON", "parameter": "","timestamp" : 1640616686515}
<class 'ValueError'
<class 'ValueError'> sock.py 106
############ END OF ERROR #######################

This errors are actually common but shouldn't affect usability that much. Its more a debugging output for me. 

> But. after changing line 140 of rig.py file from
self.my_rig.set_mode(Hamlib.RIG_MODE_USB)
to
self.my_rig.set_mode(Hamlib.RIG_MODE_PKTUSB)
error is gone.

I disabled this setting now

DJ2LS, 2021-12-27:
@frspin There are some more problems with Hamlib it seems 🤔

frspin, 2021-12-27:
> This errors are actually common but shouldn't affect usability that
> much. Its more a debugging output for me.
>
>     But. after changing line 140 of rig.py file from [...]

I don't know what is happening but, after a "git pull", now there is no
audio device, no possibility to select an input audio device and no
serial or usb device selection for rig setup.

Config file is unmodified.

Franco Spinelli
IW2DHW

DJ2LS, 2021-12-27:
Okay, maybe you can solve this by opening and closing everything. I'm testing here the commits as well on my second laptop and it seems to work here. But I still need to fix some network problems, so sometimes the GUI is stuck which can be solved by reopening everything

frspin, 2021-12-27:
Il 27/12/21 17:10, DJ2LS ha scritto:
> Okay, maybe you can solve this by opening and closing everything. I'm
> testing here the commits as well on my second laptop and it seems to
> work here. But I still need to fix some network problems, so sometimes
> the GUI is stuck which can be solved by reopening everything

Closed and reopened all. Also re-cloned repo from GitHub and logout from
desktop and login.

No change: no audio devices, no possibility to select an input audio
device a no serial port.

Also rebooted my PC.

No change

Franco Spinelli
IW2DHW

DJ2LS, 2021-12-27:
mh, this is really weird 🤔 sorry for the trouble! Is there anything which will be displayed in the cli?

frspin, 2021-12-27:
Il 27/12/21 17:26, DJ2LS ha scritto:
> mh, this is really weird 🤔 sorry for the trouble! Is there anything
> which will be displayed in the cli?

No new message:

In daemon window:

***@***.***:/mnt/Prog/ham/FreeDATA/tnc$ python3 daemon.py
2021-12-27 17:29.09 [warning  ] [TNC] Hamlib outdated          found=3.3
recommend=4.1
2021-12-27 16:29:09 [info     ] [DMN] Starting...              python=3.8
2021-12-27 16:29:09 [warning  ] [DMN] Hamlib outdated          found=3.3
recommend=4.1
2021-12-27 16:29:09 [info     ] [DMN] Starting TCP/IP socket   port=3001
2021-12-27 16:29:16 [debug    ] [DMN] Client connected         ip=127.0.0.1

And in GUI window

***@***.***:/mnt/Prog/ham/FreeDATA/gui$ npx electron main.js
libva error: /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so init failed
[10305:1227/172915.776570:ERROR:sandbox_linux.cc(374)]
InitializeSandbox() called with multiple threads in process gpu-process.
Trying to start daemon binary
Can't start daemon binary
--> this is only working with the app bundle and a precompiled binaries

Both window show messages and errors as previous runs

Franco Spinelli
IW2DHW

DJ2LS, 2021-12-27:
please do a `git pull` I fixed this

frspin, 2021-12-27:
Il 27/12/21 17:52, DJ2LS ha scritto:
> please do a |git pull| I fixed this

Done and config problems are gone.

Now there is a problem with libcodec2 finding
I have a libcodec2, 0.9 version, installed from Ubuntu repo and used by
audacity, ffmpeg, gimp, gstream, mplayer, opencpn, sane and many other.

So I need to modify "libname" list in codec2.py, removing first line
"libcodec2.so" and forcing to search in
"codec2/build_linux/src/libcodec2.so.1.0"

After this (already done but lost in recent updates) all problems are
gone except "hamlib problems".

Regars

Franco Spinelli
IW2DHW

DJ2LS, 2021-12-27:
Okay, just to be sure I didn't forget something:
You are able access the radio, but you can't toggle the ptt?

frspin, 2021-12-27:
Il 27/12/21 18:11, DJ2LS ha scritto:
> Okay, just to be sure I didn't forget something:
> You are able access the radio, but you can't toggle the ptt?

Yes, and also set frequency, but this in not a problem
Frequency and mode can be selected directly on radio

Franco Spinelli
IW2DHW

DJ2LS, 2021-12-27:
> Yes, and also set frequency, but this in not a problem
Frequency and mode can be selected directly on radio

The frequency is "read only", so it acts as an indicator, if hamlib integration is working well.

frspin, 2021-12-28:
As an interim solution I can set rig device as 
"RIG_MODEL_NETRIGCTL"
and  device port as
"localhost:4532"

and run rigctld on my Ft897D rig
but, setting this in config.json file, result in "n/a [26a9]" value in port field on GUI and my config value is not used.

Can you accept a "localhost:4532" value as a interim solution for PTT problem?

Regards
Franco Spinelli
IW2DHW

DJ2LS, 2021-12-28:
This sounds interesting. I will do some tests to see if we can use this easily and without minor changes.

frspin, 2021-12-28:
I have tested rigctld solution, as in my previous mail, using a test Python program and current version of Hamlib have problems in setting PTT also with device 2 NETRIGCTL.

As a "temporary solution" I have done a my_rig.py file removing any hamlib call and using only rigctl with subprocess.Popen() and hard coding my rig.

This is not a solution but now PTT goes On or Off and modem is working.

If you need a test I can now transmit a Beacon or a CQ on 7040000

Regards
Franco Spinelli
IW2DHW

DJ2LS, 2021-12-28:
Could you send me the command for rigctl you used? There must be a bug or reason somewhere why it’s not working 🤔

frspin, 2021-12-28:
Il 28/12/21 14:46, DJ2LS ha scritto:
> Could you send me the command for rigctl you used? There must be a bug
> or reason somewhere why it’s not working 🤔

On a CLI window:

rigctld -m 123 -r /dev/ttyUSB0 --serial-speed=38400 --ptt-type=RIG

and this start rigctld for my Ft897D with hamlib 3.3 version

On a second CLI window

rigctl -m 2 -r localhost:4532

and, at prompt, command used are

t  -> for getting PTT state
T [0|1] -> for setting PTT state

and this is working

Done same but with Python 3.8 and python binding for hamlib don't work
Test program in Python is attached
Output from this is:

./py3test.py: Python 3.8.10; Hamlib 3.3

freq :			7040000.0
status after set_freq:			-11 - Feature not available
status after set_ptt:			-11 - Feature not available
freq:			7040000.0
mode:			
bandwidth:		1
Backend copyright:	LGPL
Model:			NET rigctl
Manufacturer:		Hamlib
Backend version:	1.0
Backend status:		Stable
Rig info:		None

My "rig.py" using only rigctl commands is also attached.
With a little of work it can be modified for getting rig model and rig
port/speed from config.

Regards
Franco Spinelli
IW2DHW

DJ2LS, 2021-12-28:
Could you send me the python file to dj2ls@protonmail.com? 
if rigctl is working, then it should be possible with rigctld as well to access the ptt 🤔 

I'm thinking about a fallback option for rigctl, but on the other side there's no big difference between rigctl and using the Hamlib python binding, because the Hamlib binding has the same dependency as rigctl. 
I really want to get this working! It would be awesome, if you could test FreeDATA :-) 

So these settings are correct? I will do some debugging to see if I missed some settings.
'devicename': 'RIG_MODEL_FT897', 
'deviceport': '/dev/ttyUSB0', 
'pttprotocol': 'RIG', 
'pttport': '/dev/ttyUSB0', 
'serialspeed': '38400', 
'data_bits': '8', 
'stop_bits': '2', 
'handshake': 'None'

frspin, 2021-12-28:
Il 28/12/21 15:18, DJ2LS ha scritto:
> Could you send me the python file to ***@***.***
> ***@***.***>?

Done

> if rigctl is working, then it should be possible with rigctld as well to
> access the ptt 🤔

Configuring rigctl with config parameters use actual GUI and don't
require rigctld

> 
> I'm thinking about a fallback option for rigctl, but on the other side
> there's no big difference between rigctl and using the Hamlib python
> binding, because the Hamlib binding has the same dependency as rigctl.
> I really want to get this working! It would be awesome, if you could
> test FreeDATA :-)

With rigctl and my_rig.py I am "near" to solve the problem!

> 
> So these settings are correct? I will do some debugging to see if I
> missed some settings.
> 'devicename': 'RIG_MODEL_FT897',
> 'deviceport': '/dev/ttyUSB0',
> 'pttprotocol': 'RIG',
> 'pttport': '/dev/ttyUSB0',
> 'serialspeed': '38400',
> 'data_bits': '8',
> 'stop_bits': '2',
> 'handshake': 'None'

Setting are correct for my rig.

Regards

Franco Spinelli
IW2DHW

DJ2LS, 2021-12-28:
Thanks, @frspin , I got your rig file! Nice idea! There seem to be two radio models available! In your my_rig file you used the `897D` radio, but int the GUI you selected only the `897` model? Maybe this could be another reason for problems.

[image]

frspin, 2021-12-28:
Il 28/12/21 15:55, DJ2LS ha scritto:
> Thanks, @frspin <https://github.com/frspin> , I got your rig file! Nice
> idea! There seem to be two radio models available! In your my_rig file
> you used the |897D| radio, but int the GUI you selected only the |897|
> model? Maybe this could be another reason for problems.
> 

No, for hamlib there is no difference from Ft897 and Ft897D. Are same
rig, with only a major frequency stability for Ft897D rig.

Selecting Ft897D from GUI give a "0" as rig number and all other
programs use 123 (rig number for Ft897)

Now Beacon from modem is working and wattmeter show power out!

Franco Spinelli
IW2DHW

DJ2LS, 2021-12-28:
@frspin I created a PR #97 for this and also uploaded your my_rig as "rigctl". I also updated some parts so you can type in numbers as well in the GUI and they will be recognized by rig and rigctl!

frspin, 2021-12-30:
Update for PTT and Yaesu Ft897D

I am now able to control PTT for this rig using hamlib 3.3 from Ubuntu repo!

Required modifications in rig.py file are all in set_ptt() function:

    def set_ptt(self, state):
        if state:
            #self.my_rig.set_ptt(self.hamlib_ptt_type, 1)
            self.my_rig.set_ptt(Hamlib.RIG_VFO_CURR, 1)
        else:
            #self.my_rig.set_ptt(self.hamlib_ptt_type, 0)
            self.my_rig.set_ptt(Hamlib.RIG_VFO_CURR, 0)
        return state

set_ptt() require as first parameter a "vfo", not a mode for PTT control, and my Ft897D require setting it to RIG_VFO_CURR.

I don't have other rigs for testing but I suppose that this setting is correct for all rigs. RIG_VFO_CURR is current vfo, without changes.

Regards

Franco Spinelli
IW2DHW

closed 2022-01-05 by DJ2LS