DJ2LS/FreeDATA #102: Bidirectional piping instead of virtual audio devices for ctests

DJ2LS, 2022-01-04:
Bidirectional piping instead of virtual audio devices for ctests, so we can run the tests on GithubActions

Ideas:

- python audio dummy module
- mkfifo

drowe67, 2022-01-04:
Sending samples over UDP on localhost might work too.  Then `netcat` can be used to glue things together.

DJ2LS, 2022-05-01:
@drowe67 I decided not to update the actual ctests from pyAudio library to sounddevice library ( bot using portaudio ) within the latest version 0.4.0 . 
I'm not happy with running tests with virtual sound devices as this is not reliable enough for running the ctests. I'm thinking about named pipes or netcat for sending data like done with our simple stdio tests via pipes. For doing this we have to completely redesign them - so I don't want to spent too much time on the actual virtual audio device tests.

closed 2022-05-21 by kronenpj