DJ2LS/FreeDATA #106: Buffer overflow / decoding slower than audio

DJ2LS, 2022-01-06:
This is a big problem because it crashes the entire application.
It happens if we are decoding not as fast as audio will be added to the buffer.
If the entire application Stucks for some reason the buffer overflow occurs earlier .
This occurs on old MacBook (2008) and on laptop (2016). So there's a need for having a look at this. 

Thoughts:
- Fill buffer only if not full.
- Move decoding into own thread for each mode
- Solve application stuck moments. Specially if new data arrived or network problems occur 
- Instead of decoding simultaneously, we could decide when to decode a specific mode. This lowers datarate. 
- Instead of decoding all modes, select modes which will be simultaneously decoded

DJ2LS, 2022-01-07:
Today I changed the audio part a little bit so every decoding channel is now an own thread. This improves decoding time a little bit, but maybe still not enough.

DJ2LS, 2022-01-07:
@drowe67 I spent a half day for doing tests with a really old macbook ( 2008 ). Interesting, how  decoding is working on older, slow cpu devices. I improved our decoding part a little bit, so its now running faster and more in parallel. The old macbook can only decode 2 modes in parallel until it is running into a bufferoverflow. It will be interesting to see, how this performs on a laptop from 2016, where the problem also occurs. 
I also increased the audio buffer for a test, but with increasing buffer size and therefore more data within, the overall data processing is slowing down until its running into a buffer overflow.

Maybe I need to think about an improvement of the protocol, so we can decide, which modes we are listening to in parallel :thinking:

drowe67, 2022-01-07:
@DJ2LS it would be interesting to measure the modem CPU using the codec2 test programs, and compare to other machines.  I just use `time` as a first pass.  It's probably the acquisition code in the modem - much could be done to optimise this (e.g. vectorisation).

drowe67, 2022-01-07:
The CPU of the modem may also vary over time, for example at the end of each frame the FEC decoder will run which may block the CPU for a little while.

DJ2LS, 2022-01-10:
I added some time measurement to test test_rx and test_multimode_rx
Interesting: If screen is going to standby, we can decode without problems. So its just a small amount of CPU power which is not available during normal usage, which causes buffer overflows :smile:

DJ2LS, 2022-01-14:
Important first step:
Decoding of all modes only if datachannel opened. Else decode only datac0.
This should reduce cpu load  while in idle state. 

When in single mode transmission decode only for datachannel mode

drowe67, 2022-01-14:
> I added some time measurement to test test_rx and test_multimode_rx Interesting: If screen is going to standby, we can decode without problems. So its just a small amount of CPU power which is not available during normal usage, which causes buffer overflows smile

This doesn't sound like a CPU load issue, more a thread priority/real time/latency issue.  If it was a CPU load it would always fail, and you would see CPU pinned at 100% (or a permanently high `loadav`).

Perhaps the thread or process running the receiver is not being called in time to handle the incoming data.  

These problems can be tricky to debug.  I found it really important to find the root cause before making design changes.  On freedv-gui I added some debug counters that would increment when a buffer over or under flowed.  I could then poll these counters from time to time without upsetting the program flow.  I think we used larger buffers to overcome similar problems, and perhaps adjusted thread priorities.

Another very useful tool would be a test that reliably reproduces the problem.

drowe67, 2022-01-14:
> I added some time measurement to test test_rx and test_multimode_rx Interesting: If screen is going to standby, we can decode without problems. So its just a small amount of CPU power which is not available during normal usage, which causes buffer overflows smile

This sounds very similar to the occasional virtual sound card fails :thinking: 

Is it possible to run the tests with aplay/arecord is used for sound I/O?  Perhaps we need some more work in the PyAudio area.

DJ2LS, 2022-01-15:
@drowe67 I did this test as a OTC test with the entire tnc. I'm not sure, if I did tests with `test_multimode_rx` as well, but I think so. I'm not sure how to reproduce this problem instead of using my old MacBook. A friend has had a similiar problem with a newer laptop.  I think it may be best to count buffer overflows from now on, as you did it in freedv-gui. 
I also played with buffer size, but the problem was, that if too much data has been in the buffer, processing speed has been too slow and it continuous increased until overflow. The big buffer size also increases decoding delays, which might crash the protocol if too high. 
I will do the following as next step:
- add buffer overflow counters for every data transmission
- tests with flushing buffer after overflow ( currently we are not appending data if buffer full to prevent app crash )

DJ2LS, 2022-02-04:
This shouldn't (hopefully) occur anymore with latest protocol version because of reduced parallel decoding. 
However, we need to test this to figure out minimum hardware requirements

- [ ] Raspberry Pi 4
- [ ] MacBook Late 2008

closed 2022-04-03 by DJ2LS