DJ2LS/FreeDATA #114: TCI support

DJ2LS, 2022-01-14:
https://github.com/maksimus1210/TCI/blob/master/doc/TCI_interface_1.6.pdf

I hope that the TCI protocol is implemented relatively soon in the project:

TCI – Transceiver Control Interface is the network interface for control, data transfer and synchronization between a transceiver/receiver, logger, digital software, skimmer, other software and external power amplifier, band filters, antenna switch, radio station controller and other devices.

https://github.com/maksimus1210/TCI

Software like MSHV, JTDX etc is using this. It's not limited to SunSDR tranceivers. 

Using TCI you don't need any external/thirdparty VAC as this goes trough websocket

73 de LA3QMA
Kai Gunter

DJ2LS, 2022-03-04:
https://github.com/Hamlib/Hamlib/issues/736

LA3QMA, 2023-02-16:
> [Hamlib/Hamlib#736](https://github.com/Hamlib/Hamlib/issues/736)

They have changed the link for the protocol to: https://github.com/ExpertSDR3/TCI
I can ask if they can remove/update the old as this is not up to date regarding audio in/out

DJ2LS, 2023-06-16:
closing this as completed. Further adjustments can be done later.

PR #909 (2025-03-12): remove tci support 
Since TCI is only used by a very few persons, this PR won't affect lots of users. closes #114
closed 2023-06-16 by DJ2LS