DJ2LS/FreeDATA #117: Add Tune or TX button

kt4wo, 2022-01-14:
Add a "Tune" or "TX" button to enable easy
setting of audio levels to the radio.

drowe67, 2022-01-14:
In term of Tx levels the instructions in the FreeDV User Manual might be useful - set the drive to just tickle the ALC.

DJ2LS, 2022-01-14:
@kt4wo when talking about "Tune" are you talking about triggering the antenna tuner?
Regarding to audio levels and buttons, I try to keep the gui as simple as possible so the gui isn't overloaded with functions. However, there's a need for a common settings button. Let's see where it fits best.

@drowe67 thanks for the reminder, I will link this to the wiki / readme the next days!

drowe67, 2022-01-14:
@DJ2LS I think this request is about a way to make a continuous test transmission (say for 10 or 20 seconds) in order to adjust the TX drive, and make sure the audio path and radio is configured OK.

kt4wo, 2022-01-14:
***@***.*** I think this request is about a way to make a continuous
test transmission (say for 10 or 20 seconds) in order to adjust the TX
drive, and make sure the audio path and radio is configured OK."

Yes--- exactly.

DJ2LS, 2022-01-15:
In this case I would say, the ping options may be best, as its doing measurement of SNR as well and we are getting an answer from a remote station. 
In my over the air (OTA) test I used the ping to see if everything is working fine. 
@kt4wo but please let us know, if you have made other experiences or how we could improve this

commit 04f6c49a (2022-04-02): audio tuning and chat optimisation

closes #117
closed 2022-04-02 by DJ2LS