DJ2LS/FreeDATA #121: Hamlib TCPIP support

pe1rrr, 2022-01-17:
Currently the only way to arrange a PTT device is by assuming the radio is connected to the same system as the audio device, in my case however the ability to connect to a hamlib socket on a remote system is necessary to control the radio's PTT and such.

Cheers
Red

DJ2LS, 2022-01-17:
adding this to the todo list

DJ2LS, 2022-01-17:
@pe1rrr I started working on this and made surprisingly faster progress than I thought. Its already working for a dummy rig. Some more smaller changes and optimization and it can be tested

DJ2LS, 2022-01-18:
@pe1rrr I uploaded a version today, which is highly experimental.  Please expect some problems and failures.

pe1rrr, 2022-01-18:
[image]

In the pic attached I can't tell what the fields are for when selecting rigctld, I guessed host & port, so tried to start the TNC with those set, and seem to be correct.

When starting the TNC the settings appear to still try to load the serial port configuration, as well, the connection refused is returned although a manual telnet to the port mentioned (4599 is a custom defined instance of a rigctld emulator within LinBPQ) which connects successfully, so the TNC is struggling to open the socket here for some reason.
 
```

2022-01-18 19:01:09 [warning  ] [DMN] Stopping TNC
[{'radiocontrol': 'rigctld', 'devicename': '', 'deviceport': '/dev/ttyAMA0', 'pttprotocol': 'RIG', 'pttport': '/dev/ttyAMA0', 'serialspeed': '9600', 'data_bits': '8', 'stop_bits': '1', 'handshake': 'None', 'rigctld_port': '4599', 'rigctld_ip': '192.168.1.28'}]
2022-01-18 19:01:20 [error    ] [DMN] Hamlib: Can't open rig   e=<class 'ConnectionRefusedError'> error=ConnectionRefusedError(111, 'Connection refused')

```

I also had this message appear in the console, though it may be unrelated.

```

Traceback (most recent call last):
  File "main.py", line 16, in <module>
    import data_handler
  File "/mnt/disk1/SuperBoxStuff/git/FreeDATA/tnc/data_handler.py", line 17, in <module>
    import modem
  File "/mnt/disk1/SuperBoxStuff/git/FreeDATA/tnc/modem.py", line 24, in <module>
    import codec2
  File "/mnt/disk1/SuperBoxStuff/git/FreeDATA/tnc/codec2.py", line 94, in <module>
    api.freedv_rawdatapostambletx.argtype = [c_void_p, c_char_p, c_char_p]
  File "/usr/lib/python3.7/ctypes/__init__.py", line 369, in __getattr__
    func = self.__getitem__(name)
  File "/usr/lib/python3.7/ctypes/__init__.py", line 374, in __getitem__
    func = self._FuncPtr((name_or_ordinal, self))
AttributeError: /usr/local/lib/libcodec2.so: undefined symbol: freedv_rawdatapostambletx

```

Meanwhile, when starting the TNC to save the settings

pe1rrr, 2022-01-18:
Ignore the last line it was a snip from a deleted paragraph.

DJ2LS, 2022-01-18:
Ah, could you please delete the settings file, which is in ~/.config/FreeDATA/config.json ? I had to update the settings file, so It could be possible this caused the problem.

DJ2LS, 2022-01-18:
Yes, the fields are IP and PORT. Sorry for not changing the fields in the GUI. I had to save the current state, because of some work on the socket integration on another place. I just want to give you the option for testing this :smiley:

pe1rrr, 2022-01-18:
Ok, deleted the config.json and now the fields populate  with the default of 127.0.0.1, which is better than undefined :-) I put my IP in and port and 'started' the TNC in order to save, as I am not sure if the Test PTT works before saving the TNC settings.

```

2022-01-18 19:20:36 [warning  ] [DMN] Stopping TNC
[{'radiocontrol': 'rigctld', 'devicename': 'RIG_MODEL_DUMMY_NOVFO', 'deviceport': '/dev/ttyAMA0', 'pttprotocol': '', 'pttport': '/dev/ttyAMA0', 'serialspeed': '9600', 'data_bits': '8', 'stop_bits': '1', 'handshake': 'None', 'rigctld_port': '4599', 'rigctld_ip': '192.168.1.28'}]
2022-01-18 19:20:39 [error    ] [DMN] Hamlib: Can't open rig   e=<class 'ConnectionRefusedError'> error=ConnectionRefusedError(111, 'Connection refused')

```

By the way, do I have to rebuild anything after doing a git pull on the repo? I am not that familiar with these build tools (npm and python) so let me know if there is more to do on that end.

DJ2LS, 2022-01-18:
no, you don't have to rebuild after a git pull. This is the easy part of python with all its pros and cons :-) 
For testing you dont have to save the settings by starting the tnc. Just press Test PTT.
but I dont know why the connection cant be established. You could test this maybe by using localhost/127.0.0.1
Another option would be, there is a bug with the ip setting. After saving settings, restart the tnc and try again.

pe1rrr, 2022-01-18:
I think I have found the issue:

```

pi@superbox:~/git/FreeDATA/tnc/lib/hamlib/linux $ file libhamlib.so 
libhamlib.so: ELF 64-bit LSB pie executable, x86-64, version 1 (SYSV), dynamically linked, BuildID[sha1]=64cf4bcf6c130783ec361e5b40feb1d34fa2a422, with debug_info, not stripped

pi@superbox:~/git/FreeDATA/tnc/lib/hamlib/linux $ ls
libhamlib.so  python3.8

pi@superbox:~/git/FreeDATA/tnc/lib/hamlib/linux $ uname -a
Linux superbox 5.10.60-v7l+ #1449 SMP Wed Aug 25 15:00:44 BST 2021 armv7l GNU/Linux

```

Ah.

pe1rrr, 2022-01-19:
I think this will turn out to be a PEBCAK issue, I have just rebuilt Hamlib from latest git repo src but missed the --with-python-bindings (doh).

DJ2LS, 2022-02-02:
Feature introduced. Further testing and bug hunt needed. Closing this issue for now.

closed 2022-02-02 by DJ2LS