DJ2LS/FreeDATA #126: RTS PTT never changes status

UgoPoddine, 2022-01-19:
Hello everyone,

I have configured the FreeDATA PTT for asserting RTS signal, with "dummy rig model", in the same way I'm currently doing for example in FreeDV without issue (see attachment).
The PTT test is always green, but, at harware level, the RTS never changes status.

Any help appreciated
Thank-you
Best regards
Ugo

[PTT is not asserting Serial RTS.docx](https://github.com/DJ2LS/FreeDATA/files/7898934/PTT.is.not.asserting.Serial.RTS.docx)

UgoPoddine, 2022-03-14:
Hello everyone,

just confirming that this issue seems not to be solved in the last release.
This unfortunately prevents the use on RaspberryPI and old radios, where GIPIO (driven by the RTS) is used for asserting PTT.

Congratulation for the interesting project, my best 73s
IU1IPB Ugo

DJ2LS, 2022-03-14:
Hi Ugo, thanks for your reply. Sorry for not fixing this problem yet. It's hard for me to reproducing it, but I didn't forget it! 
As an alternative you could use VOX driven ptt as long as RTS isn't working.

UgoPoddine, 2022-03-14:
Thank-you. You are welcome 🙂
It was only to make you aware.
If you need help in this particular issue, let me know.
I found in the past VOX based Ptt of my ts180s built in 1979 a bit slow in response for digital modes, but of course we are dealing now with asynchronous "ping pong" like connections, that probably your suggestion is good

Thank-you
Ugo

DJ2LS, 2022-03-14:
Ugo, if needed we could integrate a pilot tone for opening VOX. Overall speed would be slower but still possible, I think

DJ2LS, 2022-03-24:
@UgoPoddine , i will drop native Hamlib integration, soon for moving to rigctld.

Did you test ptt with rigctld? If so, could you do me the favour and sending me a working command for testing it?

UgoPoddine, 2022-03-24:
Hello Simon,

I will try to help you as soon as possible, of course.
But please consider that my experience in using Hamlib is minimal : my radio is a fully analogue built in 1979 (no CAT) and the Raspberry that acts on it simply runs a small Ptt daemon that takes RTS from virtual serial port in bank2 and switchs GPIO connected to the Ptt line of the radio and nothing more, without the need of hamlibs.
But I have a couple of Linux machines, not connected to radio, where I can do some tests.

I will keep you informed
73, Ugo

UgoPoddine, 2022-04-05:
Hello Simon,

Small update here.
The PTT with direct integration of RS232 RTS without Hamlib doesn't run yet.
But with Rigctld running in background, the Raspberry PI is now able to switch the PTT and to send the first CQ.

I will try now to find someone on 40m to exchange somethin.

Thank you very much
73, Ugo

DJ2LS, 2022-04-06:
Hi Ugo, great that its now working. Could you send me the command you used for rigctld and the envorinment? So we could add it to the wiki and it may help others.

UgoPoddine, 2022-04-06:
Hello Simon,

No problem.  The direct method is "nice to have" (especially with less powerful systems, where CPU is rare), but with the new hamlib tool integration we "PIs" are quite ok.

Surely less urgent
Ugo

DJ2LS, 2022-05-01:
hi @UgoPoddine , it seems this problem is solved? So I'm going to close it for now. If it occurs again, let us open this issue again

UgoPoddine, 2022-05-01:
Hello Simon, 

I have not yet installed the yesterday version and it will take sometime.
In the previous version the situation was the following :

A) using Hamlib Rigctld the RTS could be set (almost a workaround)
B) in direct mode (my preferred due to the poor resources of the PI3) no way to set RTS 

On my opinion, without urgency, it could be better to adjust also the direct mode (if not yet solved in the yesterday release), also because "direct mode" setting shows RTS as PTT possible mode, but I will let you to evaluate :-)
At the moment, in some way, in any case, I can transmit, also if, at the moment, I had never found somebody in Europe answering the pings.

To thank-you in any case for your activity, feel free to ask if you need some test

73s
Ugo

DJ2LS, 2022-05-01:
Okay, Ugo, thanks for your reply! 
The idea is, completely remove support for direct mode and moving to "rigctld" only. So we can avoid some failures because hamlet / rigctld is more stable than the python implementation.

Just as a notice: With the latest releases you can enable a "QRV" response. If you are transmitting a CQ, stations hearing you can respond you with a "QRV" so you know, they are online. In this case you don't have to ping to a specific callsign.

closed 2022-10-04 by DJ2LS

UgoPoddine, 2022-10-11:
> Il giorno 24 mar 2022, alle ore 06:49, DJ2LS

Hello Simon,

I made the required test, if correctly understood.

The following command (red part), on my Raspberry, starts hamlib rigctld 
tool as need in my scenario :

/home/pi/wsjt-x/wsjtx-2.5.4/build/hamlib-prefix/bin/./rigctl -m 1 -p 
/dev/ttyAMA0 -P RTS

This starts the interactive daemon from command line, on dummy radio, my 
RS232 port and with PTT on RTS, as your app should do.

With "t" command you can check the PTT status, with "T" command (0/1) 
you can set or unset PTT, "q" to quit.

Of course, forget about the path of rigctl on my machine : I'm forced to 
call it explicitly due a version mismatch.

Let me know if it helps.

73, IU1IPB Ugo

PS : I'm at currently at :

rigctl Hamlib 4.5 SHA=18548e

version

DJ2LS, 2022-10-11:
Hi Ugo, did you run a „git pull“ for the latest version? There’s now support for controlling rigctld from gui Simon Von meinem iPhone gesendet

UgoPoddine, 2022-10-11:
Hi Simon, 

The version has been updated three days ago,  but I haven't checked the changes yet.
In any case, untill now since several months, using Hamlin my PTT worked well.
In the past my attempt was using direct mode.
I will try as soon as possible.

Thanks
Ugo

DJ2LS, 2022-10-11:
Yes Ugo, direct connection has been removed with latest version. Could you do me the favour and testing if this is working for you as well?Thanks,Simon Von meinem iPhone gesendet