DJ2LS/FreeDATA #130: no longer reading rigctl parameters from gui

pe1rrr, 2022-01-24:
It seems that the variable set for the rigctl host and port isnt being read from the config file anymore (this is on the ls-leave-prototype branch) 

create 48<->8 kHz resampler
create audio_buffer:  9600
create audio_buffer:  9600
create audio_buffer:  9600
2022-01-24 19:14:16 [debug    ] [TNC] starting pyaudio callback
2022-01-24 19:14:16 [critical ] [RIGCTLD] Could not connect to rigctld! ip=localhost port=4532
2022-01-24 19:14:16 [debug    ] Rigctl intialized
2022-01-24 19:14:16 [info     ] [TNC] Starting TCP/IP socket   port=3000
Exception in thread HAMLIB_THREAD:
Traceback (most recent call last):
  File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.9/threading.py", line 892, in run
    self._target(*self._args, **self._kwargs)
  File "/mnt/disk1/SuperBoxStuff/git/FreeDATA/tnc/modem.py", line 464, in update_rig_data
    static.HAMLIB_BANDWITH = self.hamlib.get_bandwith()
  File "/mnt/disk1/SuperBoxStuff/git/FreeDATA/tnc/rigctld.py", line 76, in get_bandwith
    bandwith = data[1]
IndexError: list index out of range
2022-01-24 19:14:18 [debug    ] [TNC] Client connected         ip=127.0.0.1 port=37680

When running the TNC in direct mode it throws the issue with not finding the lib:

2022-01-24 19:13:57 [warning  ] [RIG] Python Hamlib binding not found error=ImportError('libpython3.7m.so.1.0: cannot open shared object file: No such file or directory')
2022-01-24 19:13:57 [warning  ] [RIG] Trying to open rigctl
2022-01-24 19:13:57 [warning  ] [RIG] Rigctl found! Please try using this version=4.5~git

What is some good news is at least if I choose the TCPIP rigctl (and it fails because of the host/ip being wrong aka default) it can still proceed into TNC operation, which I will test on its own shortly.

DJ2LS, 2022-01-24:
Thanks for the error report. I didn't focus on Hamlib integration on ls-leave-prototype, yet. I will have a look the next days/weeks. This (Hamlib integration) will need some more time

DJ2LS, 2022-01-24:
Bug found, will be fixed with next push to ls-leave-prototype

DJ2LS, 2022-01-24:
I will close this, so we are keeping issue list a little bit more clean. I solved this issue with #128

closed 2022-01-24 by DJ2LS

pe1rrr, 2022-01-24:
The IP is now being correctly set on the TNC from the GUI but the port is not, it is still trying the default port (4532) and not my port (4599 in this case).

2022-01-24 21:22:29 [warning  ] [RIGCTLD] Re-Trying to establish a connection to rigctld! attempt=1 ip=192.168.1.28 port=4532
2022-01-24 21:22:30 [warning  ] [RIGCTLD] Re-Trying to establish a connection to rigctld! attempt=2 ip=192.168.1.28 port=4532
2022-01-24 21:22:31 [warning  ] [RIGCTLD] Re-Trying to establish a connection to rigctld! attempt=3 ip=192.168.1.28 port=4532
2022-01-24 21:22:32 [warning  ] [RIGCTLD] Re-Trying to establish a connection to rigctld! attempt=4 ip=192.168.1.28 port=4532
2022-01-24 21:22:33 [warning  ] [RIGCTLD] Re-Trying to establish a connection to rigctld! attempt=5 ip=192.168.1.28 port=4532

I can confirm that at least network traffic is flowing albeit to  the wrong port :)

22:23:44.348739 IP 192.168.1.27.52618 > 192.168.1.28.4532: Flags [S], seq 2742403232, win 64240, options [mss 1460,sackOK,TS val 2728558710 ecr 0,nop,wscale 10], length 0
22:23:44.349125 IP 192.168.1.28.4532 > 192.168.1.27.52618: Flags [R.], seq 0, ack 2742403233, win 0, length 0

pe1rrr, 2022-01-24:
Testing the TNC the rigctl timeouts/polling interfere with the TNC's communication so it will stall a CQCQCQ or PING command until it manages to get in between the polls.

https://youtu.be/TqeAyoQrap0

DJ2LS, 2022-01-24:
Okay, fixed the port problem now - hopefully.
I tried to reproduce the problem with the CQ and PING which forces a network problem, but can't reproduce this. We might have a closer look at this, the next days. 
Enough work for today :-)

pe1rrr, 2022-01-24:
> Okay, fixed the port problem now - hopefully.
> 
> I tried to reproduce the problem with the CQ and PING which forces a network problem, but can't reproduce this. We might have a closer look at this, the next days. 
> 
> Enough work for today :-) 
> 
> 

Ok good news, the rigctl seems to work for small files but I have encountered a situation during a larger (46kb jpg) file that caused the rigctl routine to lose connection and then go full throttle polling without reconnecting. I have reproduced this twice.

I checked the rigctld daemon (listener) and that was still working just fine, there was also no network traffic during this situation on that port while monitoring with tcpdump so I think something broke the pipe/socket internally?

Smaller file transfers worked great-ish. Good work!

pe1rrr, 2022-01-25:
Here is a screen grab with my voice this time explaining whats going on, its unedited so you should skip to the part where it goes bad with rigctl errors.

https://youtu.be/z2TNR3I4BDg

DJ2LS, 2022-01-25:
Mh, interesting, the daemon is loosing the connection. This shouldn't occur. I need to implement a reconnect mechanism into the socket module. I thought this will already be handled by the socket library of python 🤔

DJ2LS, 2022-01-25:
Which raspberry pi are you using?