DJ2LS/FreeDATA #131: CPU load: Option to turn off waterfall/scatter

pe1rrr, 2022-01-28:
Would it be possible to add the ability to turn off the waterfall/scatter displays as these are possibly contributing to higher than necessary CPU load under test on RPI4.  

Is it possible to run the daemon completely without any GUI running? I.e headless?

DJ2LS, 2022-01-28:
Yes, disabling scatter and gui is planed to reduce network load.
I will keep this issue open as feature request 

The daemon/tnc can be used without gui. Further information will be available, soon: https://wiki.freedata.app

pe1rrr, 2022-01-28:
> Yes, disabling scatter and gui is planed to reduce network load.
> I will keep this issue open as feature request 
> 
> The daemon/tnc can be used without gui. Further information will be available, soon: https://wiki.freedata.app

The wiki is off to a good start w/doc for the TNC commands, Keep up the great work. 👍

DJ2LS, 2022-02-02:
Introduced feature with latest commit.

closed 2022-02-02 by DJ2LS