DJ2LS/FreeDATA #132: GUI bails when daemon segfaults after long file-write timeout.

pe1rrr, 2022-02-05:
Sending two identical 4.7Kb files from A->B causes the receiving GUI to:

- the receiver does not report any incoming file on the 2nd attempt
- The daemon gets stuck until it hits a segfault (about 5 minutes)
- the GUI then loses connection to the daemon so goes blurry

This is reproducible so I am wondering what causes the file writing to time-out because the daemon gets the fully decoded dump of the file on the debug screen, [image]

DJ2LS, 2022-02-06:
Hm, Interesting 🤔 As this error already occurred before, I did some improvements to the network, which are not yet published. This hopefully solves the problem, the gui is stuck in blurred mode, which is an indicator for a lost network connection. 
But the segfault is concerning me a little bit. 

@pe1rrr Could you post the file you are trying to send here in this issue? I really want to reproduce this for getting the origin of the segfault.
My raspberry pi arrived, so I can use a native arm device now for testing! 
However, I need some days of time for this ( busy because of volunteer work and exams starting  )

pe1rrr, 2022-02-06:
> Hm, Interesting 🤔 As this error already occurred before, I did some improvements to the network, which are not yet published. This hopefully solves the problem, the gui is stuck in blurred mode, which is an indicator for a lost network connection. 
> But the segfault is concerning me a little bit. 

I don't know how the setup really works but from observation I can tell there's about two or three types of network threads and its when the one thread handling the transfer of the file to a file handle is eventually dying, for some reason this causes the TNC to disconnect from the GUI, sort of. It blurs it, but I can  still see the GUI doing something behind the blurred screen.

> @pe1rrr Could you post the file you are trying to send here in this issue? I really want to reproduce this for getting the origin of the segfault.
> My raspberry pi arrived, so I can use a native arm [...]

Steps:

- Start the TNC on both File Send and File Receive side.
- Send the EXE file to the remote station once.
- Once file has successfully transferred and the GREEN status pop up has shown, check the date timestamp on the file and make a note if it.
- Send the file again from the same location to the same remote station without touching the remote station's GUI.
- Observe if the GUI shows any status of INCOMING FILE, if not, wait for the file to finish transferring, and then wait. It will segfault after about 5 minutes.
- If the GUI does show the INCOMING FILE and GREEN success message afterwards, the bug has failed to reproduce, try sending the file again without resetting any of the TNCs or GUIs.

Hope this helps.
Good luck!

It is this file: https://www.dropbox.com/s/l6fgqxsyaoqguzu/RedHeat%20-%20Gravity%20Assist.exe?dl=0

DJ2LS, 2022-02-10:
After doing some debugging I came to the decision:
- no problem with file transfer
- no problem with tnc ( tnc still alive, responds to pings and file transfer. everything as expected )
- daemon crashes ( gui looses connection then )
- possible crash caused by error `libgomp could not create thread pool destructor`

ToDo:
- remove blurr screen #134 
- stay connected to tnc even if daemon not running
- find problem within daemon

DJ2LS, 2022-02-12:
Latest commit removes the blurred screen and gui is now able to connect to a headless tnc, running without the daemon. 
The crashing daemon seems only occuring on raspberry pis

DJ2LS, 2022-02-15:
latest debugging shows, that the problem could be caused by PyAudio

DJ2LS, 2022-02-15:
We can start and terminate PyAudio 254 times. So this seems to be a possible bug with PyAudio 🤔

DJ2LS, 2022-02-16:
possible fix: self compiling PortAudio from source

DJ2LS, 2022-02-17:
Running audio reloading part as subprocess seems to fix this problem

DJ2LS, 2022-02-17:
@pe1rrr It seems I fixed the problem after hours of debugging. Could you do me the favor and testing this on your raspberry Pi as well? Thanks!

DJ2LS, 2022-02-17:
aaah audio broken, I need to fix something else!

pe1rrr, 2022-02-17:
Oh whoops I need to do a bunch of re-testing as I just finished finding a couple of problems, one of which is related to the file transfer- i will get back to you in a bit!

DJ2LS, 2022-02-17:
No hurry! On my Pi the problem seems to be fixed. at least you just have to run another daemon instance on another port.
`./daemon --port 5000`

Just let it run for 10 or 15 minutes. This can be done in parallel. And hopefully it survives

pe1rrr, 2022-02-17:
> No hurry! On my Pi the problem seems to be fixed. at least you just have to run another daemon instance on another port.
> 
> `./daemon --port 5000`
> 
> 
> 
> Just let it run for 10 or 15 minutes. This can be done in parallel. And hopefully it survives

I have two RPI4s running back-to-back with the TNCs so parallel is no issue :) so far so good, 10 minutes into a test session. ~~I've left them idle while i write this and will report at the end whether they have crashed or not as it will take me about 5 minutes to write this.~~

Ok daemon is still running fine

There is an issue with the GUI still where it will fail to get "file successful" from the daemon, the daemon does a the cleanup, the CRC is good, but it doesn't ever receive the status message or the file. I have produced this bug in the video I am posting, and i have my mic on so you can hear me explaining what I am doing, but its 10 minute long video, you can skip to the last 3 minutes or so if need be.

In the video:
I was testing the file transfer several times because earlier today I was doing a test on apparently a fault that you subsequently fixed while i was testing (lol) but it was interesting- it was failing when a audio underrun forced re-negotiation, and that would always result in a unsuccessful file transfer, but you committed a fix already and that appears to have been this issue.

Anyway, after failing to reproduce that issue with the auto-negotiation screwing up the file transfer, i noticed that when I went to send the same file back from the Pi that had just received a file succesfullly, it would cause the problem outlined above with the GUI.

I hope this makes sense! Here's the video: https://youtu.be/Cp3RdBrzfpM

DJ2LS, 2022-02-18:
@pe1rrr I'd like to close this issue, as it seems the multiprocessing solution with PyAudio solved the problem of a crashing deamon on the the raspberry pi

DJ2LS, 2022-02-18:
> I have two RPI4s running back-to-back with the TNCs so parallel is no issue :) so far so good, 10 minutes into a test session. ~I've left them idle while i write this and will report at the end [...]

This problem should have been solved by the latest commits 🤔

DJ2LS, 2022-02-24:
Problem didn't occur anymore

closed 2022-02-24 by DJ2LS