DJ2LS/FreeDATA #139: Rigctl (tcpip) settings not overriding defaults

pe1rrr, 2022-03-15:
After updating and removing the config.json I have tried to get the rigctl working to my custom IP and port but the daemon doesn't seem to get it and so it reports errors "connection refused" and shows the ip (localhost) and port (4352), those are the defaults.

DJ2LS, 2022-03-16:
Hi @pe1rrr ,
the remote ip feature is working, but you might need to press enter after changing the ip and then restart the application. Then it should save the ip and port. 
I need to optimise this, but I'm focused on some other areas as well. So it might take some time until fixed this. 

A lot of things changed the last weeks. Please move from branch `ls-leave-prototype` to `main` when running from source. As an alternative you could run the app bundle, which has an included auto updater. But it seems the precompiled binaries won't run on raspberry pi, yet.

DJ2LS, 2022-04-15:
@pe1rrr Does this problem still occur with the latest version?

pe1rrr, 2022-04-27:
this issue appear to be fixed

DJ2LS, 2022-04-27:
Great, I will close it for now. If it occurs again, let us open it again and track down the failure 🔍

pe1rrr, 2022-05-01:
Here's some logs:

2022-05-01 22:39:34 [debug    ] [C2 ] creating audio buffer    size=9600
2022-05-01 22:39:34 [debug    ] [C2 ] creating audio buffer    size=9600
2022-05-01 22:39:34 [debug    ] [C2 ] creating audio buffer    size=4800
2022-05-01 22:39:34 [debug    ] [C2 ] creating audio buffer    size=4800
2022-05-01 22:39:34 [info     ] opened audio devices
2022-05-01 22:39:34 [debug    ] [TNC] starting pyaudio callback
2022-05-01 22:39:34 [warning  ] [RIGCTLD] Connection to rigctld refused! Reconnect... e=ConnectionRefusedError(111, 'Connection refused') ip=localhost port=4532
2022-05-01 22:39:34 [error    ] [RIGCTLD] Can't connect to rigctld! ip=localhost port=4532
2022-05-01 22:39:34 [info     ] [TNC] Starting TCP/IP socket   port=3000
2022-05-01 22:39:35 [debug    ] [SCK] Client connected         ip=127.0.0.1 port=37326
2022-05-01 22:39:35 [warning  ] [RIGCTLD] Connection to rigctld refused! Reconnect... e=ConnectionRefusedError(111, 'Connection refused') ip=localhost port=4532
2022-05-01 22:39:36 [warning  ] [RIGCTLD] Connection to rigctld refused! Reconnect... e=ConnectionRefusedError(111, 'Connection refused') ip=localhost port=4532
2022-05-01 22:39:36 [warning  ] [RIGCTLD] Connection to rigctld refused! Reconnect... e=ConnectionRefusedError(111, 'Connection refused') ip=localhost port=4532

After startup the TNC appears to be OK with the rigctl, but after issuing a ping command from the GUI it begins to spew the above errors that reference a default setting that isn't from my input to the GUI's settings for rigctl.

pe1rrr, 2022-05-01:
2022-05-01 22:44:40 [debug    ] [SCK] CMD                      command={'type': 'get', 'command': 'test_hamlib', 'parameter': [{'radiocontrol': 'rigctld', 'devicename': 'RIG_MODEL_DUMMY_NOVFO', 'deviceport': '/dev/ttyAMA0', 'pttprotocol': 'USB', 'pttport': '/dev/ttyAMA0', 'serialspeed': '9600', 'data_bits': '8', 'stop_bits': '1', 'handshake': 'None', 'rigctld_port': '4599', 'rigctld_ip': '192.168.1.28'}]}
2022-05-01 22:44:40 [info     ] [RIGCTLD] Connected to rigctld! ip=192.168.1.28 port=4599
2022-05-01 22:44:40 [debug    ] Rigctl intialized
2022-05-01 22:44:40 [info     ] [DMN] Hamlib PTT               status=SUCCESS

### start TNC

2022-05-01 22:45:00 [debug    ] [SCK] CMD                      command={'type': 'set', 'command': 'start_tnc', 'parameter': [{'mycall': 'PE1RRR-1', 'mygrid': 'jo21lo', 'rx_audio': '0', 'tx_audio': '0', 'radiocontrol': 'rigctld', 'devicename': 'RIG_MODEL_DUMMY_NOVFO', 'deviceport': '/dev/ttyAMA0', 'pttprotocol': 'USB', 'pttport': '/dev/ttyAMA0', 'serialspeed': '9600', 'data_bits': '8', 'stop_bits': '1', 'handshake': 'None', 'rigctld_port': '4599', 'rigctld_ip': '192.168.1.28', 'enable_scatter': 'False', 'enable_fft': 'True', 'enable_fsk': 'False', 'low_bandwith_mode': 'False', 'tuning_range_fmin': '-50.0', 'tuning_range_fmax': '50.0', 'tx_audio_level': '125', 'respond_to_cq': 'True'}]}
2022-05-01 22:45:00 [debug    ] [DMN] TNC Startup Config : mycall value=PE1RRR-1
2022-05-01 22:45:00 [debug    ] [DMN] TNC Startup Config : mygrid value=jo21lo
2022-05-01 22:45:00 [debug    ] [DMN] TNC Startup Config : rx_audio value=0
2022-05-01 22:45:00 [debug    ] [DMN] TNC Startup Config : tx_audio value=0
2022-05-01 22:45:00 [debug    ] [DMN] TNC Startup Config : radiocontrol value=rigctld
2022-05-01 22:45:00 [debug    ] [DMN] TNC Startup Config : devicename value=RIG_MODEL_DUMMY_NOVFO
2022-05-01 22:45:00 [debug    ] [DMN] TNC Startup Config : deviceport value=/dev/ttyAMA0
2022-05-01 22:45:00 [debug    ] [DMN] TNC Startup Config : pttprotocol value=USB
2022-05-01 22:45:00 [debug    ] [DMN] TNC Startup Config : pttport value=/dev/ttyAMA0
2022-05-01 22:45:00 [debug    ] [DMN] TNC Startup Config : serialspeed value=9600
2022-05-01 22:45:00 [debug    ] [DMN] TNC Startup Config : data_bits value=8
2022-05-01 22:45:00 [debug    ] [DMN] TNC Startup Config : stop_bits value=1
2022-05-01 22:45:00 [debug    ] [DMN] TNC Startup Config : handshake value=None
2022-05-01 22:45:00 [debug    ] [DMN] TNC Startup Config : rigctld_port value=4599
2022-05-01 22:45:00 [debug    ] [DMN] TNC Startup Config : rigctld_ip value=192.168.1.28
2022-05-01 22:45:00 [debug    ] [DMN] TNC Startup Config : enable_scatter value=False
2022-05-01 22:45:00 [debug    ] [DMN] TNC Startup Config : enable_fft value=True
2022-05-01 22:45:00 [debug    ] [DMN] TNC Startup Config : enable_fsk value=False
2022-05-01 22:45:00 [debug    ] [DMN] TNC Startup Config : low_bandwith_mode value=False
2022-05-01 22:45:00 [debug    ] [DMN] TNC Startup Config : tuning_range_fmin value=-50.0
2022-05-01 22:45:00 [debug    ] [DMN] TNC Startup Config : tuning_range_fmax value=50.0
2022-05-01 22:45:00 [debug    ] [DMN] TNC Startup Config : tx_audio_level value=125
2022-05-01 22:45:00 [debug    ] [DMN] TNC Startup Config : respond_to_cq value=True
2022-05-01 22:45:00 [warning  ] [DMN] Starting TNC             port=/dev/ttyAMA0 rig=RIG_MODEL_DUMMY_NOVFO
2022-05-01 22:45:00 [info     ] [DMN] TNC started              path=source
2022-05-02 00:45.01 [info     ] [C2 ] Searching for libcodec2...
2022-05-02 00:45.01 [warning  ] [C2 ] Libcodec2 found but not loaded e=OSError('lib/codec2/macOS/libcodec2.1.0.dylib: invalid ELF header') path=lib/codec2/macOS/libcodec2.1.0.dylib
2022-05-02 00:45.01 [warning  ] [C2 ] Libcodec2 found but not loaded e=OSError('lib/codec2/win64/libcodec2.dll: invalid ELF header') path=lib/codec2/win64/libcodec2.dll
2022-05-02 00:45.01 [warning  ] [C2 ] Libcodec2 found but not loaded e=OSError('lib/codec2/win32/libcodec2.dll: invalid ELF header') path=lib/codec2/win32/libcodec2.dll
2022-05-02 00:45.01 [info     ] [C2 ] Libcodec2 loaded         path=lib/codec2/linux/armv7l/libcodec2.so.1.0
2022-05-01 22:45:01 [info     ] [TNC] Starting FreeDATA        author=DJ2LS version=0.4.0-alpha year=2022
2022-05-01 22:45:01 [debug    ] [C2 ] create 48<->8 kHz resampler
2022-05-01 22:45:01 [debug    ] [C2 ] creating audio buffer    size=9600
2022-05-01 22:45:01 [debug    ] [C2 ] creating audio buffer    size=9600
2022-05-01 22:45:01 [debug    ] [C2 ] creating audio buffer    size=9600
2022-05-01 22:45:01 [debug    ] [C2 ] creating audio buffer    size=4800
2022-05-01 22:45:01 [debug    ] [C2 ] creating audio buffer    size=4800
2022-05-01 22:45:01 [info     ] opened audio devices
2022-05-01 22:45:01 [debug    ] [TNC] starting pyaudio callback
2022-05-01 22:45:01 [warning  ] [RIGCTLD] Connection to rigctld refused! Reconnect... e=ConnectionRefusedError(111, 'Connection refused') ip=localhost port=4532
2022-05-01 22:45:01 [error    ] [RIGCTLD] Can't connect to rigctld! ip=localhost port=4532
2022-05-01 22:45:01 [info     ] [TNC] Starting TCP/IP socket   port=3000
2022-05-01 22:45:01 [debug    ] [SCK] Client connected         ip=127.0.0.1 port=37466
2022-05-01 22:45:02 [warning  ] [RIGCTLD] Connection to rigctld refused! Reconnect... e=ConnectionRefusedError(111, 'Connection refused') ip=localhost port=4532
[... cut ...]
2022-05-01 22:45:15 [warning  ] [RIGCTLD] Connection to rigctld refused! Reconnect... e=ConnectionRefusedError(111, 'Connection refused') ip=localhost port=4532
2022-05-01 22:45:15 [warning  ] [RIGCTLD] Connection to rigctld refused! Reconnect... e=ConnectionRefusedError(111, 'Connection refused') ip=localhost port=4532
2022-05-01 22:45:16 [warning  ] [RIGCTLD] Connection to rigctld refused! Reconnect... e=ConnectionRefusedError(111, 'Connection refused') ip=localhost port=4532
2022-05-01 22:45:17 [warning  ] [RIGCTLD] Connection to rigctld refused! Reconnect... e=ConnectionRefusedError(111, 'Connection refused') ip=localhost port=4532

DJ2LS, 2022-05-03:
Bug found which may be the origin of this issue:

https://github.com/DJ2LS/FreeDATA/blob/c42be2e1d01a7bb83cc628f0687d310585f12a94/tnc/modem.py#L203

DJ2LS, 2022-05-19:
@pe1rrr can you verify, this is working now?

DJ2LS, 2022-05-19:
found the bug :mechanical_arm:

closed 2022-05-19 by DJ2LS

pe1rrr, 2022-05-19:
Can confirm, fixed

DJ2LS, 2022-05-19:
Great, one more problem solved!

commit 2f767a0c (2022-05-19): fixed #139
commit a1c09c7f (2022-05-19): Merge pull request #195 from DJ2LS/ls-fix-139

fixed #139