DJ2LS/FreeDATA #141: USB port indication are confusing in GUI or FreeDATA it can't find them.

computer-wizard, 2022-03-24:
OS is: Ubuntu 20.04. Latest version FreeDATA (v0.2.0-alpha1)

The USB port listing in the GUI are confusing, I see USB Serial controller D with an ID's what is not right anyhow for transceiver
and PTT. devices I use.
I'm using two Prolific USB to serial converters, one connected to the transceiver (FT-847) to control the frequency and another one used for the PTT (Using RTS to active this)  What I want to see is the Linux device names  like ttyUSB0, ttyUSB1.  It seems that FreeDATA can't find these devices as I don't see them  in the drop-down menus of the GUI.  I tried to enter these devices manually in config.json like  "deviceport": "/dev/ttyUSB1",  for the device connected to the transceiver and "pttport": "/dev/ttyUSB0", for the PTT device but that doesn't help.

DJ2LS, 2022-03-24:
Hi @computer-wizard and thanks for your feedback!

The USB port names are the names which are provided by the usb devices ( for easier device selection ),but are confusing or misleading for some manufacturers. I added a checksum at the end of the name, because some usb devices of some radios occur twice with the same name but different behavior. For this reason just selecting the USB port night not work, actually.

However, it would be best, if we have both - the device description and the port as well, which will be displayed?

computer-wizard, 2022-03-24:
I understand, for Linux the USB device names as used in /dev like ttyUSB0, ttyUSB1, etc.  would be the best and that is also what I see in other programs. Thanks for you effort making this program!

DJ2LS, 2022-03-26:
@computer-wizard I checked some code parts - it should'nt be a problem, selecting the deviceport manually within the config file. If the device is then available, it should be selected by default.
However, I adjusted the way, the devices will be displayed. The next release includes these changes and from now on, the ports are visible as well.

commit e8869e17 (2022-03-26): display device port

display the device ports as well closes #141
closed 2022-03-26 by DJ2LS