DJ2LS/FreeDATA #142: no quiskdigital.input and output audio devices reported

pearcedotsgatgmaildotcom, 2022-03-26:
hank you for this.

I have compiled and tried it out on my network connected hl2 .. running quisk as the "radio" .. I can ptt ok but are unable to access the quisk digital input device or the quisk digital output . monitor device.I tried defining input and output as pulse but still no audio to or from the hl2 .. which also has a hr50 attached and gives an external indication of modulation on tx .. quisk also reports the rd drive current and power on tx .. both of which are zero indicating no modulation .. nor can I hear myself on a remote sdr but confirmed quisk ok by successfully tx a carrier.

In js8 the are quisk input output monitor devices in the sound io menus ...
and js8 works fine in my set up as does fldigi 

Not knowing the code of Freedata could someone give me a few hints of where to look .. and/or what to do.

FreeData modem going ok I think as I can tx to the control computer sound card and can see my audio whistle picked up on the mic and displayed on the FreeDATA waterfall.

Any hints appreciated ... I want to try a few trans tasman tests using kiwisdrs in Australia as reception points for later decoding ..

de zl1any stephen

DJ2LS, 2022-03-27:
@pearcedotsgatgmaildotcom , thank for opening an issue! I didnt work with quisk, yet. 
Does quisk create an own audio device, which can be selected? When using `pulse`, are you able seeing the devices in `pavucontrol` ? FreeDATA uses the portaudio library, which should display every audio device, which is available :thinking:

pearcedotsgatgmaildotcom, 2022-03-27:
Thanks for the reply

Yes qusik seems to open devices which are accessible a pavucontrol ..
enclosed some screen shots of pavucontrol after I fire up quisk (I didn't
turn on the hl2 hence the "no device" report in quisk)
Interestingly firefox is also reported as you can see .

I tried running list_audio_devices.py from the tools provided with FreeDTA
with quisk running but it does not seem to report quisk devices so the
problem might be in the python code.
I will try to scrutinize the js8call code to see how they do it.

Fldigi seems a bit different and uses xml-rpc calls .. which I don't fully
understand but there is mention of them in the configuration menu of quisk
with port numbers ..

Altogether this area seems to lack standardization ...

I'll also have a troll thru the quisk code and maybe contact the author to
see if he has any comments although I think he might be getting a bit feed
up with maintenance queries ...

Thanks for you efforts .. what is the programming methodology you are using
... the daemon /tnc code is written in python but the interface seems to be
Java ... I presume you have taken this approach to
maximise future portability to other os? (android etc)

i am keen to contribute and to learn .. the idea of an open source
codec/modem for hf digital data transfer with better performance than
pactor or vara appeals to me

I would like to get it running with the quisk/hl2/hr50 set up I have here
as I think this is a good platform to allow easy expansion to other set ups
and is physically quite tidy.

Quisk on some levels seems a bit of a throw together ... but despite that
is really robust and reliable.

As I said in my last email the ptt rigctl stuff seems basically ok ..
although a little buggy .. I can trigger the tx but sometimes it seems to
fail to disengage at the end of a tx

73s

de zl1any stephen

I

DJ2LS, 2022-03-30:
Hi Stephen,

> Thanks for the reply Yes qusik seems to open devices which are accessible a pavucontrol .. enclosed some screen shots of pavucontrol after I fire up quisk (I didn't turn on the hl2 hence the "no [...]

I did some tests with quick and it seems the problem is the audio library, FreeDATA uses. I'm not sure, if I can solve this problem the next time as the audio stuff is a little bit critical and I have a lot of other FreeDATA-stuff wich needs my attention as well. But I will keep this problem in my mind. Maybe theres a solution for this problem! 

>  what is the programming methodology you are using ... the daemon /tnc code is written in python but the interface seems to be Java ... I presume you have taken this approach to maximise future portability to other os? (android etc)

Yes, you're right! The reason for this is portability to other OS and easiness of learning the programming language - and therefore contributing to the project. Python is used for the TNC, Java, Javascript and HTML for the GUI. All of these languages are easy to use, widely supported and documented with a huge range of public support. 

> i am keen to contribute and to learn .. the idea of an open source codec/modem for hf digital data transfer with better performance than pactor or vara appeals to me I would like to get it running [...]

Yes, I understand your interest in open-source projects like FreeDATA. However, its not easy being competitive with great programs like VARA. Sadly its not open sourced and only available for windows. FreeDATA has to long way and huge challenges to handle before it could become competitive. However, FreeDATA has some unique features, which might be of interest somewhen. But we need some more time...

Simon

DJ2LS, 2022-04-08:
@pearcedotsgatgmaildotcom The latest update should bring back some basic pulse support. Could you test, if you can use quick audio devices within pavaucontrol / pulse?

Please not, the quisk devices will still not displayed within FreeDATA GUI, you need to select them from a pulse audio manager

pearcedotsgatgmaildotcom, 2022-04-08:
Thanks Simon

I'll try out this coming week .. away from qth for a few days at moment

73s

Stephen

pearcedotsgatgmaildotcom, 2022-04-12:
Hi Simon

This works with the 0.3.3 app image ... I had some problems with the source
initially throwing a unable to connect error to the 127.0.0.1 ip and then
unable to find electron-update and not
running at all.

Anyway I can copy myself on a sdr .. here are some audio files through that
link of pings and cqs ..
all I need now is someone to talk to .. hi hi ... the platform seems pretty
robust ...

the pulse audio devices are ALSA-plugin python 3.8  going to quiskdig input
and monitor quisk dig op ...

I'll try on a transtasman sdr when it gets dark

it looks like it will work pretty well .. albeit the software is v. alpha
.. you look like you made a good choice of development platforms

I presume and api for the daemon will come eventually ..

Is the daemon stable now? Looks like most of the work is in the gui.

Thanks again ... can you give me any references that could help me get into
the programming methodology for the java npx electron stuff?

73s

Stephen

ps can one feed recorded audio files back into the tnc .. like fldigi does
... this is quite good for checking performance with a remote sdr

DJ2LS, 2022-04-12:
Hi Stephen, great quisk audio devices are now available for you!

For getting in contact with possible other testers I suggest writing an announcement at the mailing list! 

Regarding to a api and programming: it depends on what you want to do. Did you see the wiki? https://wiki.freedata.app/ ? There you have a list of all network commands. So you could control the daemon and tnc with nearly every other script or programming language who supports network sockets.

Regarding to audio files: this is not working yet, but you could use some loopback devices. However, for testing I recommend using two laptops or virtual machines with shared audio. I'm using two laptops , two radios, both are connected via cable or antenna. 

If it's okay for you I'd like to close this issue, now. 
Feel free contacting me via email if you need further support with developing something for FreeDATA 🙂

Simon, DJ2LS