DJ2LS/FreeDATA #143: PTT doesn't work - version 0.3.0-alpha1

computer-wizard, 2022-03-29:
OS: Ubuntu 20.04 LTS  - FreeDATA version 0.3.0-alpha1 (latest version)

PTT device setting is /dev/ttyUSB1 but transmitter is not keyed. I'm using a separated USB to serial converter for this. FreeDATA 
audio is modulating the transmitter if I key the transmitter manually. There is no way to set the PTT method if you use a separate PTT serial port in the GUI. I'm using RTS.

DJ2LS, 2022-03-30:
Hi, @computer-wizard , thanks for the feedback. My idea is, dropping direct Hamlib integration and moving to rigctld ( Hamlib precompiled Network application) for more Network based flexibility and stability, somewhen the next releases. Could you do me the favour and sending me a working example of using rigctld with the help of the [wiki](https://wiki.freedata.app/en/usage/hamlib)?
Then we have a working configuration which would help me when debugging.

computer-wizard, 2022-03-31:
Hi Simon,
Thanks for the info about rigctld and the ptt works with that. I just use,
rigctld -p /dev/ttyUSB0  and that's all, no need to use the -m option for the rig as I can't use ptt over cat anyhow
.

computer-wizard, 2022-03-31:
Here is the rigctld line what includes the transceiver model and rig device, so that the correct frequency is read back for display in the GUI. ptt device is also included.

rigctld -m 101 -p /dev/ttyUSB0 -P RTS -r /dev/ttyUSB1 -s 57600

It should be nice if FreeDATA itself can start rigctld after starting the tnc in future releases :-)

DJ2LS, 2022-04-02:
@computer-wizard Thanks!
Yes, starting rigctld via gui is the direction I'm heading to :-)

DJ2LS, 2022-04-15:
> Here is the rigctld line what includes the transceiver model and rig device, so that the correct frequency is read back for display in the GUI. ptt device is also included.
>
> rigctld -m 101 -p [...]

@computer-wizard could you do me the favor and opening a feature request for this?

computer-wizard, 2022-04-15:
Done!

DJ2LS, 2022-04-15:
Thanks 🙂
Can this issue be closed then?

DJ2LS, 2022-04-17:
Closed because of #168

closed 2022-04-17 by DJ2LS