DJ2LS/FreeDATA #146: b6cf6e7 on PI/OS GUI Regression

UgoPoddine, 2022-04-03:
Hello,

I have just upgraded the GUI to b6cf6e7 on my RaspberryPI 3+ running PI/OS Bullseye.
Unfortunately the soundcards settings have disappered and there is no way to having them back (the small drop-down list for the soundcards shows nothing). Please note that I'm using ALSA without PulseAudio, with well runnig PortAudio without any issue (aplay -l shows correctly my soundcards and Fldigi detects them correctly).
The same happens with the USB port, but here I han still insert the PPT RS232 port manually again.

Thank-you
73 IU1IPB Ugo

DJ2LS, 2022-04-03:
Hi Ugo, thanks for the report! Could you do me the favor and starting the daemon from cli and posting the messages? This sounds like the daemon isn't providing any audio device anymore in your case.

UgoPoddine, 2022-04-04:
Of course : here the small log.

thank-you
Ugo

Traceback (most recent call last):
  File "/home/pi/FreeDATA/FreeDATA/tnc/./daemon.py", line 29, in <module>
    import audio
  File "/home/pi/FreeDATA/FreeDATA/tnc/audio.py", line 6, in <module>
    import sounddevice as sd
ModuleNotFoundError: No module named 'sounddevice'

UgoPoddine, 2022-04-04:
By the way :

 aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: Loopback [Loopback], device 0: Loopback PCM [Loopback PCM]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 0: Loopback [Loopback], device 1: Loopback PCM [Loopback PCM]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 1: Device [Generic USB Audio Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: Device_1 [USB Audio Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

and ...

 arecord -l
**** List of CAPTURE Hardware Devices ****
card 0: Loopback [Loopback], device 0: Loopback PCM [Loopback PCM]
  Subdevices: 7/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 0: Loopback [Loopback], device 1: Loopback PCM [Loopback PCM]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 1: Device [Generic USB Audio Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: Device_1 [USB Audio Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

Ugo

DJ2LS, 2022-04-04:
Okay, you are missing the sounddevice module.
Please run „pip install sounddevice“

If this is not working, try "pip3".
Please note that you need the latest version of codec2! You should recompile it as well, because of a new api function!

Simon

UgoPoddine, 2022-04-04:
Hello Simon,

Issue is solved following your instructions, thank-you.
This evening I will try the new PTT management on the PI.
As you are probably no so far from me, do you have some kind of beacon active on your QTH in order to try some test file transfer in 80 or 40m ?

Thank-you again
Ugo

DJ2LS, 2022-04-04:
Hi Ugo, I'm at work actually.
But I'm and some other German testers are mostly on 40m, 7057 or 7061 USB

closed 2022-04-05 by DJ2LS

DJ2LS, 2022-10-11:
Okay, you are missing the  soubddevice module.
Please run „pip install sounddevice“
Simon