DJ2LS/FreeDATA #149: auto compile codec2 on github release action

DJ2LS, 2022-04-03:
we should automatically build and link libcodec2.
We need to static link some dependencies for crosscompiling https://github.com/drowe67/codec2/pull/307

kronenpj, 2022-05-19:
I'd like to be sure what this is asking for. From what I see in the GitHub actions files, `codec2` is being built from source on Linux and Mac but not Windows. Please tell me if I'm reading these lines incorrectly:
```yml
      - name: Build codec2 Linux
        if: matrix.os == 'ubuntu-20.04'
        working-directory: tnc
        run: |
          git clone https://github.com/drowe67/codec2.git
          cd codec2 && mkdir build_linux && cd build_linux
          cmake ../
          make
```
```yml
      - name: Build codec2 macOS
        if: matrix.os == 'macos-10.15'
        working-directory: tnc
        run: |
          git clone https://github.com/drowe67/codec2.git
          cd codec2 && mkdir build_mac && cd build_mac
          cmake ../
          make
```
If my interpretation is correct, this issue asks for the Windows `codec2` to be compiled.

DJ2LS, 2022-05-20:
Yes, for Mac and Linux codec2 will be already compiled successfully. But we need to integrate it into the correct subfolder so it will be used by the tnc. Unpegged this in the build process as a proof of concept.
Linux and Mac will be the easier part.

For windows we need to write a new build process and also include static linking of needed libraries into the build process. 

I suggest we are doing this step by step as building the app is a pain.
Starting with Linux, testing it, Mac, ... and at last we are working on windows.

We also need to check if we are going to build native windows builds or if we are crosscompiling. Also we need 32bit band 64Bit libraries.

kronenpj, 2022-06-26:
Feel free to take over the `issue_149_fix` branch. I didn't get as far as I'd hoped and re-learned that I'm not a Windows developer.

DJ2LS, 2022-06-27:
Yes, thanks @kronenpj , no worries about windows. Working with GitHub actions is a pain as the commands can't be tested or I don't know how

PR #237 (2022-07-14): Update GithubAction
This PR improves building a new release via Github Actions. Codec2 ( libcodec2 ) will be compiled for different platforms from latest code base. Furthermore the GithubAction steps are now nearly equal for every Matrix step.

libcodec2 --> fixes #149 
- [x] Raspberry Pi Bullseye
- [x] Windows 10
- [x] Windows 11
- [x] MacOS 11
- [x] MacOS 12
- [x] Ubuntu 20.04
- [x] Ubuntu 22.04

FreeDATA

- [x] Windows
- [x] MacOS
- [x] Ubuntu
- [ ] Raspberry Pi

TODO:

- [x] add mingw dependencies to libcodec2 folder
- [x] Compile for x86 and x64
closed 2022-07-14 by DJ2LS