DJ2LS/FreeDATA #153: Audio problems with Ubuntu 18.04

frspin, 2022-04-07:
There are many audio problems with Ubuntu 18.04
At the first run many devices appears in pull-down menu for input/output audio device (but no pulse)
At the second run only first audio device appear in input device and only built-in devices in output device
(in my setup only input located on-board audio device and output over on-board speakers and HDMI).

Using instead latest Appimage no device appear as input and output device

Obviously this change also my config.json file!

Regards
Franco Spinelli
IW2DHW

DJ2LS, 2022-04-07:
Hi Franco,

thanks for your issue report.
I had to change the audio library, because the old one hasn't been updated for more than 5 years now. Sorry if this affects your usability because of dropped pulse support. However, PortAudio will get Pulse support, soon https://github.com/PortAudio/portaudio/pull/336

The empty audio device list sounds like a problem occurred with the new library, are you running FD from CLI or AppBundle?
Please be sure, you installed sound device with "pip install sounddevice" 

Simon

frspin, 2022-04-07:
Il 07/04/22 10:38, DJ2LS ha scritto:
> Hi Franco,
> 
> thanks for your issue report.
> I had to change the audio library, because the old one hasn't been 
> updated for more than 5 years now. Sorry if this affects your usability 
> because of dropped pulse support. However, PortAudio will get Pulse 
> support, soon PortAudio/portaudio#336 
> <https://github.com/PortAudio/portaudio/pull/336>
> 
> The empty audio device list sounds like a problem occurred with the new 
> library, are you running FD from CLI or AppBundle?

Running Appimage no audio device at all, but I never tried Appiame as 
first run.
Running FD from CLI with a simple shell proc I get described behavior.
After the first run any other run, from CLI or with Appimage, give an 
empty audio list.

> Please be sure, you installed sound device with "pip install sounddevice"

I have installed it with

pip3 install sounddevice

***@***.***:/mnt/Prog/ham/FreeDATA$ pip list |grep sounddevice
sounddevice                  0.4.4

***@***.***:/mnt/Prog/ham/FreeDATA$ pip show sounddevice
Name: sounddevice
Version: 0.4.4
Summary: Play and Record Sound with Python
Home-page: http://python-sounddevice.readthedocs.io/
Author: Matthias Geier
Author-email: ***@***.***
License: MIT
Location: /usr/local/lib/python3.8/dist-packages
Requires: CFFI
Required-by:

there is no package "sounddevice" in Ubuntu repo and 20.04 have both 
python 2.7 and python 3.8

If I open a Python3 console and do:

 >>> import sounddevice as sd
 >>> sd.query_devices()
    0 HDA Intel HDMI: 0 (hw:0,3), ALSA (0 in, 8 out)
    1 HDA Intel HDMI: 1 (hw:0,7), ALSA (0 in, 8 out)
    2 HDA Intel HDMI: 2 (hw:0,8), ALSA (0 in, 8 out)
    3 HDA Intel HDMI: 3 (hw:0,9), ALSA (0 in, 8 out)
    4 HDA Intel HDMI: 4 (hw:0,10), ALSA (0 in, 8 out)
    5 HDA Intel PCH: ALC887-VD Alt Analog (hw:1,2), ALSA (2 in, 0 out)
    6 HDA NVidia: HDMI 0 (hw:2,3), ALSA (0 in, 8 out)
    7 HDA NVidia: HDMI 1 (hw:2,7), ALSA (0 in, 8 out)
    8 HDA NVidia: HDMI 2 (hw:2,8), ALSA (0 in, 8 out)
    9 HDA NVidia: HDMI 3 (hw:2,9), ALSA (0 in, 8 out)
   10 HDA NVidia: HDMI 4 (hw:2,10), ALSA (0 in, 8 out)
   11 Ensoniq AudioPCI: ES1371 DAC1 (hw:3,1), ALSA (0 in, 2 out)
   12 hdmi, ALSA (0 in, 8 out)
   13 pulse, ALSA (32 in, 32 out)
* 14 default, ALSA (32 in, 32 out)
 >>>

I get correct list of all my audio device, also pulse as you can see 
(device 13)

Also, for pulse device (13)

 >>> sd.query_devices(device=13,kind="input")
{'default_high_output_latency': 0.034807256235827665, 
'default_low_output_latency': 0.008684807256235827, 
'default_samplerate': 44100.0, 'name': u'pulse', 'max_output_channels': 
32, 'default_low_input_latency': 0.008684807256235827, 
'max_input_channels': 32, 'default_high_input_latency': 
0.034807256235827665, 'hostapi': 0}

 >>> sd.query_devices(device=13,kind="output")
{'default_high_output_latency': 0.034807256235827665, 
'default_low_output_latency': 0.008684807256235827, 
'default_samplerate': 44100.0, 'name': u'pulse', 'max_output_channels': 
32, 'default_low_input_latency': 0.008684807256235827, 
'max_input_channels': 32, 'default_high_input_latency': 
0.034807256235827665, 'hostapi': 0}

If I get your testing code for audio devices (without multiprocessing 
code) I get:

***@***.***:~/tmp$ python3 audio_devices.py

******** input device ********
{'id': 5, 'name': 'HDA Intel PCH: ALC887-VD Alt Analog (hw:1,2)'}
{'id': 13, 'name': 'pulse'}
{'id': 14, 'name': 'default'}

******** output device ********
{'id': 0, 'name': 'HDA Intel HDMI: 0 (hw:0,3)'}
{'id': 1, 'name': 'HDA Intel HDMI: 1 (hw:0,7)'}
{'id': 2, 'name': 'HDA Intel HDMI: 2 (hw:0,8)'}
{'id': 3, 'name': 'HDA Intel HDMI: 3 (hw:0,9)'}
{'id': 4, 'name': 'HDA Intel HDMI: 4 (hw:0,10)'}
{'id': 6, 'name': 'HDA NVidia: HDMI 0 (hw:2,3)'}
{'id': 7, 'name': 'HDA NVidia: HDMI 1 (hw:2,7)'}
{'id': 8, 'name': 'HDA NVidia: HDMI 2 (hw:2,8)'}
{'id': 9, 'name': 'HDA NVidia: HDMI 3 (hw:2,9)'}
{'id': 10, 'name': 'HDA NVidia: HDMI 4 (hw:2,10)'}
{'id': 11, 'name': 'Ensoniq AudioPCI: ES1371 DAC1 (hw:3,1)'}
{'id': 12, 'name': 'hdmi'}
{'id': 13, 'name': 'pulse'}
{'id': 14, 'name': 'default'}

Can be a multiprocess problem or a dialog problem between tnc and gui?

Franco Spinelli
IW2DHW

frspin, 2022-04-07:
For testing multiprocess code in your implementation I have done some more test:

I have inserted your get_audio_device() in my test code, imported multiprocessing and now appear a different list:

spin@franco:~/tmp$ python3 audio_devices.py
******** input device ********
{'id': 5, 'name': 'HDA Intel PCH: ALC887-VD Alt Analog (hw:1,2)'}
******** output device ********
{'id': 0, 'name': 'HDA Intel HDMI: 0 (hw:0,3)'}
{'id': 1, 'name': 'HDA Intel HDMI: 1 (hw:0,7)'}
{'id': 2, 'name': 'HDA Intel HDMI: 2 (hw:0,8)'}
{'id': 3, 'name': 'HDA Intel HDMI: 3 (hw:0,9)'}
{'id': 4, 'name': 'HDA Intel HDMI: 4 (hw:0,10)'}
{'id': 6, 'name': 'HDA NVidia: HDMI 0 (hw:2,3)'}
{'id': 7, 'name': 'HDA NVidia: HDMI 1 (hw:2,7)'}
{'id': 8, 'name': 'HDA NVidia: HDMI 2 (hw:2,8)'}
{'id': 9, 'name': 'HDA NVidia: HDMI 3 (hw:2,9)'}
{'id': 10, 'name': 'HDA NVidia: HDMI 4 (hw:2,10)'}
{'id': 11, 'name': 'Ensoniq AudioPCI: ES1371 DAC1 (hw:3,1)'}
{'id': 12, 'name': 'hdmi'}

As you can see "pulse" and "default" devices are disappeared!

This is also the list appearing in the first run of program.

Another note: in input device there is no 'Ensoniq AudioPCI: ES1371 DAC1 (hw:3,1)'}, my USB card connected to the rig, and this device don't appear in both lists.

This is an error of sounddevice library:

>>> sd.query_devices(device=11)
{'name': 'Ensoniq AudioPCI: ES1371 DAC1 (hw:3,1)', 'hostapi': 0, 'max_input_channels': 0, 'max_output_channels': 2, 'default_low_input_latency': -1.0, 'default_low_output_latency': 0.00870729499688603, 'default_high_input_latency': -1.0, 'default_high_output_latency': 0.03482917998754412, 'default_samplerate': 44100.9521484375}

give max:input_channels:0 but this is not true. 
This device have 2 input channels and 2 output channels, as shown in previous portaudio library and as used in any other program (Fldigi, Wsjt-X, etc.)

Regards
Franco Spinelli
IW2DHW

DJ2LS, 2022-04-07:
Interesting 🤔 I will have a closer look at this!

And yes, I wouldn't be surprised if there a problem with multiprocessing ( which is necessary for fixing a Debian/Raspberry bug only ) 

Thanks for the provided information and doing tests with the audio library! This is really helping me!

frspin, 2022-04-07:
Doing some test, using pyaudio and not sounddevice (pyaudio is updated to 2017 and sounddevice to 2021)
I get some 'Ensoniq AudioPCI: ES1371 DAC1 (hw:3,1), missed also as input device with pyaudio, but pulse and default devices are present also in multiprocess version:

******** input device ********
{'id': 5, 'name': 'HDA Intel PCH: ALC887-VD Alt Analog (hw:1,2)'}
{'id': 13, 'name': 'pulse'}
{'id': 14, 'name': 'default'}
******** output device ********
{'id': 0, 'name': 'HDA Intel HDMI: 0 (hw:0,3)'}
{'id': 1, 'name': 'HDA Intel HDMI: 1 (hw:0,7)'}
{'id': 2, 'name': 'HDA Intel HDMI: 2 (hw:0,8)'}
{'id': 3, 'name': 'HDA Intel HDMI: 3 (hw:0,9)'}
{'id': 4, 'name': 'HDA Intel HDMI: 4 (hw:0,10)'}
{'id': 6, 'name': 'HDA NVidia: HDMI 0 (hw:2,3)'}
{'id': 7, 'name': 'HDA NVidia: HDMI 1 (hw:2,7)'}
{'id': 8, 'name': 'HDA NVidia: HDMI 2 (hw:2,8)'}
{'id': 9, 'name': 'HDA NVidia: HDMI 3 (hw:2,9)'}
{'id': 10, 'name': 'HDA NVidia: HDMI 4 (hw:2,10)'}
{'id': 11, 'name': 'Ensoniq AudioPCI: ES1371 DAC1 (hw:3,1)'}
{'id': 12, 'name': 'hdmi'}
{'id': 13, 'name': 'pulse'}
{'id': 14, 'name': 'default'}

The problem is using a python library for getting and using sound devices in any OS (linux, Windows and MacOS).

Portaudio seen a good solution as library (Fldigi use this library in Linux, Windows ad Mac without any problem). 
Here the problem seem in Python3 interface.

Ensonic ES1371 is present with no input channel and my USB device, where my rig is connected, is not present in both library query.

Using C program for query devices from Portaudio I get 18 devices and, for Es1371 device:

--------------------------------------- device #12
Name                        = Ensoniq AudioPCI: ES1371 DAC2/ADC (hw:3,0)
Host API                    = ALSA
Max inputs = 2, Max outputs = 2
Default low input latency   =   0.0087
Default low output latency  =   0.0087
Default high input latency  =   0.0348
Default high output latency =   0.0348
--------------------------------------- device #13
Name                        = Ensoniq AudioPCI: ES1371 DAC1 (hw:3,1)
Host API                    = ALSA
Max inputs = 0, Max outputs = 2
Default low input latency   =  -1.0000
Default low output latency  =   0.0087
Default high input latency  =  -1.0000
Default high output latency =   0.0348

Note the difference from audio device #12 (2 input and 2 output) #13 (0 input and 2 output)

Also, for my USB audio card:

--------------------------------------- device #15
Name                        = USB Device 0x41e:0x30d3: Audio (hw:5,0)
Host API                    = ALSA
Max inputs = 1, Max outputs = 2
Default low input latency   =   0.0087
Default low output latency  =   0.0087
Default high input latency  =   0.0348
Default high output latency =   0.0348

completely missed in Python library as is missed pulse device

--------------------------------------- device #17
Name                        = pulse
Host API                    = ALSA
Max inputs = 32, Max outputs = 32
Default low input latency   =   0.0087
Default low output latency  =   0.0087
Default high input latency  =   0.0348
Default high output latency =   0.0348

So the only possibility for getting all devices, also USB and Pulse devices, is using Portaudio library and not pyaudio or sounddevice porting for Python, at least in their current state.

Regards

Franco Spinelli
IW2DHW

DJ2LS, 2022-04-07:
Thanks for testing this!

At least the pyaudio and sounddevice libraries are just accessing the Portaudio C API. So there must be a way getting all of the devices somehow. I will have a look at this when back at home from my job.

frspin, 2022-04-07:
Il 07/04/22 16:45, DJ2LS ha scritto:
> Thanks for testing this!
> 
> At least the pyaudio and sounddevice libraries are just accessing the 
> Portaudio C API. So there must be a way getting all of the devices 
> somehow. I will have a look at this when back at home from my job.

Strange!

After some work, running C version of query and using device for a video 
call now same test program with PyAudio show also my USB soundcard

******** input device ********
{'id': 5, 'name': 'HDA Intel PCH: ALC887-VD Analog (hw:1,0)'}
{'id': 6, 'name': 'HDA Intel PCH: ALC887-VD Alt Analog (hw:1,2)'}
{'id': 12, 'name': 'Ensoniq AudioPCI: ES1371 DAC2/ADC (hw:3,0)'}
{'id': 14, 'name': 'WEB CAM: USB Audio (hw:4,0)'}
{'id': 15, 'name': 'USB Device 0x41e:0x30d3: Audio (hw:5,0)'}
{'id': 17, 'name': 'pulse'}
{'id': 18, 'name': 'default'}
******** output device ********
{'id': 0, 'name': 'HDA Intel HDMI: 0 (hw:0,3)'}
{'id': 1, 'name': 'HDA Intel HDMI: 1 (hw:0,7)'}
{'id': 2, 'name': 'HDA Intel HDMI: 2 (hw:0,8)'}
{'id': 3, 'name': 'HDA Intel HDMI: 3 (hw:0,9)'}
{'id': 4, 'name': 'HDA Intel HDMI: 4 (hw:0,10)'}
{'id': 5, 'name': 'HDA Intel PCH: ALC887-VD Analog (hw:1,0)'}
{'id': 7, 'name': 'HDA NVidia: HDMI 0 (hw:2,3)'}
{'id': 8, 'name': 'HDA NVidia: HDMI 1 (hw:2,7)'}
{'id': 9, 'name': 'HDA NVidia: HDMI 2 (hw:2,8)'}
{'id': 10, 'name': 'HDA NVidia: HDMI 3 (hw:2,9)'}
{'id': 11, 'name': 'HDA NVidia: HDMI 4 (hw:2,10)'}
{'id': 12, 'name': 'Ensoniq AudioPCI: ES1371 DAC2/ADC (hw:3,0)'}
{'id': 13, 'name': 'Ensoniq AudioPCI: ES1371 DAC1 (hw:3,1)'}
{'id': 15, 'name': 'USB Device 0x41e:0x30d3: Audio (hw:5,0)'}
{'id': 16, 'name': 'hdmi'}
{'id': 17, 'name': 'pulse'}
{'id': 18, 'name': 'default'}

And sounddevice version show:

***@***.***:~/tmp$ python3 audio_devices_sounddevice.py
******** input device ********
{'id': 5, 'name': 'HDA Intel PCH: ALC887-VD Analog (hw:1,0)'}
{'id': 6, 'name': 'HDA Intel PCH: ALC887-VD Alt Analog (hw:1,2)'}
{'id': 12, 'name': 'Ensoniq AudioPCI: ES1371 DAC2/ADC (hw:3,0)'}
{'id': 14, 'name': 'WEB CAM: USB Audio (hw:4,0)'}
{'id': 15, 'name': 'USB Device 0x41e:0x30d3: Audio (hw:5,0)'}
******** output device ********
{'id': 0, 'name': 'HDA Intel HDMI: 0 (hw:0,3)'}
{'id': 1, 'name': 'HDA Intel HDMI: 1 (hw:0,7)'}
{'id': 2, 'name': 'HDA Intel HDMI: 2 (hw:0,8)'}
{'id': 3, 'name': 'HDA Intel HDMI: 3 (hw:0,9)'}
{'id': 4, 'name': 'HDA Intel HDMI: 4 (hw:0,10)'}
{'id': 5, 'name': 'HDA Intel PCH: ALC887-VD Analog (hw:1,0)'}
{'id': 7, 'name': 'HDA NVidia: HDMI 0 (hw:2,3)'}
{'id': 8, 'name': 'HDA NVidia: HDMI 1 (hw:2,7)'}
{'id': 9, 'name': 'HDA NVidia: HDMI 2 (hw:2,8)'}
{'id': 10, 'name': 'HDA NVidia: HDMI 3 (hw:2,9)'}
{'id': 11, 'name': 'HDA NVidia: HDMI 4 (hw:2,10)'}
{'id': 13, 'name': 'Ensoniq AudioPCI: ES1371 DAC1 (hw:3,1)'}
{'id': 15, 'name': 'USB Device 0x41e:0x30d3: Audio (hw:5,0)'}
{'id': 16, 'name': 'hdmi'}

With USB device as input and output and without pulse and default devices.

Very strange

Franco Spinelli
IW2DHW

DJ2LS, 2022-04-08:
@frspin I did some tests this morning and it seems I found the origin of the problem: 
We need to reset and initialize PortAudio before running the multiprocessing part. Then all devices will be detected.

DJ2LS, 2022-04-08:
@frspin can you confirm the fix?

frspin, 2022-04-08:
Il 08/04/22 11:36, DJ2LS ha scritto:
> @frspin <https://github.com/frspin> can you confirm the fix?

Now some devices are preserved after each run but:

- no USB audio device
- no waterfall at all

After a new reboot USB audio come back and vanish at second run
Still no waterfall at all

This is device list immediately after reboot

******** input device ********
{'id': 0, 'name': 'Ensoniq AudioPCI: ES1371 DAC2/ADC (hw:0,0)'}
{'id': 7, 'name': 'HDA Intel PCH: ALC887-VD Analog (hw:2,0)'}
{'id': 8, 'name': 'HDA Intel PCH: ALC887-VD Alt Analog (hw:2,2)'}
{'id': 14, 'name': 'WEB CAM: USB Audio (hw:4,0)'}
{'id': 15, 'name': 'USB Device 0x41e:0x30d3: Audio (hw:5,0)'}
{'id': 16, 'name': 'sysdefault'}
{'id': 18, 'name': 'samplerate'}
{'id': 19, 'name': 'speexrate'}
{'id': 20, 'name': 'pulse'}
{'id': 21, 'name': 'upmix'}
{'id': 22, 'name': 'vdownmix'}
{'id': 23, 'name': 'ARDOP'}
{'id': 25, 'name': 'default'}
******** output device ********
{'id': 0, 'name': 'Ensoniq AudioPCI: ES1371 DAC2/ADC (hw:0,0)'}
{'id': 1, 'name': 'Ensoniq AudioPCI: ES1371 DAC1 (hw:0,1)'}
{'id': 2, 'name': 'HDA Intel HDMI: 0 (hw:1,3)'}
{'id': 3, 'name': 'HDA Intel HDMI: 1 (hw:1,7)'}
{'id': 4, 'name': 'HDA Intel HDMI: 2 (hw:1,8)'}
{'id': 5, 'name': 'HDA Intel HDMI: 3 (hw:1,9)'}
{'id': 6, 'name': 'HDA Intel HDMI: 4 (hw:1,10)'}
{'id': 7, 'name': 'HDA Intel PCH: ALC887-VD Analog (hw:2,0)'}
{'id': 9, 'name': 'HDA NVidia: HDMI 0 (hw:3,3)'}
{'id': 10, 'name': 'HDA NVidia: HDMI 1 (hw:3,7)'}
{'id': 11, 'name': 'HDA NVidia: HDMI 2 (hw:3,8)'}
{'id': 12, 'name': 'HDA NVidia: HDMI 3 (hw:3,9)'}
{'id': 13, 'name': 'HDA NVidia: HDMI 4 (hw:3,10)'}
{'id': 15, 'name': 'USB Device 0x41e:0x30d3: Audio (hw:5,0)'}
{'id': 16, 'name': 'sysdefault'}
{'id': 17, 'name': 'front'}
{'id': 18, 'name': 'samplerate'}
{'id': 19, 'name': 'speexrate'}
{'id': 20, 'name': 'pulse'}
{'id': 21, 'name': 'upmix'}
{'id': 22, 'name': 'vdownmix'}
{'id': 23, 'name': 'ARDOP'}
{'id': 24, 'name': 'dmix'}
{'id': 25, 'name': 'default'}

And this after first run of FD

******** input device ********
{'id': 6, 'name': 'HDA Intel PCH: ALC887-VD Alt Analog (hw:2,2)'}
{'id': 12, 'name': 'pulse'}
{'id': 13, 'name': 'default'}
******** output device ********
{'id': 0, 'name': 'Ensoniq AudioPCI: ES1371 DAC1 (hw:0,1)'}
{'id': 1, 'name': 'HDA Intel HDMI: 0 (hw:1,3)'}
{'id': 2, 'name': 'HDA Intel HDMI: 1 (hw:1,7)'}
{'id': 3, 'name': 'HDA Intel HDMI: 2 (hw:1,8)'}
{'id': 4, 'name': 'HDA Intel HDMI: 3 (hw:1,9)'}
{'id': 5, 'name': 'HDA Intel HDMI: 4 (hw:1,10)'}
{'id': 7, 'name': 'HDA NVidia: HDMI 0 (hw:3,3)'}
{'id': 8, 'name': 'HDA NVidia: HDMI 1 (hw:3,7)'}
{'id': 9, 'name': 'HDA NVidia: HDMI 2 (hw:3,8)'}
{'id': 10, 'name': 'HDA NVidia: HDMI 3 (hw:3,9)'}
{'id': 11, 'name': 'HDA NVidia: HDMI 4 (hw:3,10)'}
{'id': 12, 'name': 'pulse'}
{'id': 13, 'name': 'default'}

Franco Spinelli
IW2DHW

DJ2LS, 2022-04-08:
Okay, however, we seem to have fixed another bug which caused a missing pulse device. 
It seems it also takes some time (0-10 seconds in may case), until used devices will be released again under some circumstances. FD only displayes devices with audio slots available. But normally the audio devices are directly available. But it seems ,pulse is also blocking some more audio devices after using them. Could you have a look, if the devices will come back after some time without restarting your computer?

I also had the problem of the stopped waterfall, when not selecting the correct input device in pavucontrol. You can change this while running "pulse" device within FD without the need of closing it.

frspin, 2022-04-08:
Il 08/04/22 12:09, DJ2LS ha scritto:
> It seems it also takes some time (0-10 seconds in may case), until used
> devices will be released again under some circumstances. FD only
> displayes devices with audio slots available. But normally [...]

Full set of devices returned after 30' without reboot.

Regards

Franco Spinelli
IW2DHW

frspin, 2022-04-08:
After previous test now devices seem to be persistent

Still non waterfall

Device is "pulse" in input and also in output
In pavucontrol device is redirected to USB audio device
Pavucontrol show activity on device
Waterfall is still missed

All as in attached video (1.2 Mb - if compatible with GitHub)

https://user-images.githubusercontent.com/3316507/162437924-0e0089eb-7200-4f73-9bea-67e5b8e4f8ba.mp4

Regards

Franco Spinelli
IW2DHW

DJ2LS, 2022-04-11:
Interesting, whats happening if you switching the audio device from "mono" to "stereo" ?

frspin, 2022-04-11:
Il 11/04/22 11:19, DJ2LS ha scritto:
> Interesting, whats happening if you switching the audio device from 
> "mono" to "stereo" ?

Not possible. SoundBlaster Play, audio USB device connected to my rig, 
is a "mono only" input (mike input)

Regards

Franco Spinelli
IW2DHW

DJ2LS, 2022-04-11:
What about the sample rate? FreeDATA needs 48000Hz

frspin, 2022-04-11:
Default sample rate is 44100 for this and other audio cards
But Pulse is able to do any needed conversion
I am using it for any digital program, all using Pulse or Portaudio but all using this libraries from C or C++ 

This setup, using Pulse and also using USB audio directly, was working with FD until sound library change.

Franco Spinelli
IW2DHW

DJ2LS, 2022-04-14:
I did some more tests, but I can't reproduce it. The new sound library isn't doing things that different, compared to the old one because of the fact, they are using the same C library 🤔 
Yes, you're right regarding to the sample rate. And I'm forcing usage of 48000Hz when opening the audio device, so there shouldn't be a problem in this are.

frspin, 2022-04-14:
Il 14/04/22 09:56, DJ2LS ha scritto:
> I did some more tests, but I can't reproduce it. The new sound library 
> isn't doing things that different, compared to the old one because of 
> the fact, they are using the same C library 🤔

Now I am away from my home and unable to do more tests.

What can I do, when back to home, is put a mike in front of radio 
speaker and try another sound card.

For now pulse in already present in device list and I can use pulse for 
change audio input to FD for testing the "waterfall missing problem"

> Yes, you're right regarding to the sample rate. And I'm forcing usage of 
> 48000Hz when opening the audio device, so there shouldn't be a problem 
> in this are.

My setup was working with old Appimages and I can download old version 
and test it. I need to cancel auto-update for this.
Appimage have all library built-in so also any update in system 
libraries can't have effect.

Regard

Franco Spinelli
IW2DHW

DJ2LS, 2022-04-14:
> Il 14/04/22 09:56, DJ2LS ha scritto:
> Now I am away from my home and unable to do more tests. What can I do, when back to home, is put a mike in front of radio speaker and try another sound card. For now pulse in already present in device list and I can use pulse for change audio input to FD for testing the "waterfall missing problem"

No hurry, take your time!

> My setup was working with old Appimages and I can download old version and test it. I need to cancel auto-update for this. Appimage have all library built-in so also any update in system libraries can't have effect. Regard Franco Spinelli IW2DHW

Yes, that's the point which is confusing me 🤔

frspin, 2022-04-20:
Back to home and back to test this strange problem

Also with 0.1.0-alpha waterfall is missed!
So I suppose a problem with some libraries or commands not in Appimage.

Can be "npm"?

My npm version command show:

spin@franco:/mnt/Prog/ham/FreeDATA$ npm version
{
  npm: '8.7.0',
  node: '16.14.0',
  v8: '9.4.146.24-node.20',
  uv: '1.43.0',
  zlib: '1.2.11',
  brotli: '1.0.9',
  ares: '1.18.1',
  modules: '93',
  nghttp2: '1.45.1',
  napi: '8',
  llhttp: '6.0.4',
  openssl: '1.1.1m+quic',
  cldr: '40.0',
  icu: '70.1',
  tz: '2021a3',
  unicode: '14.0',
  ngtcp2: '0.1.0-DEV',
  nghttp3: '0.1.0-DEV'
}

Regards

Franco Spinelli
IW2DHW

frspin, 2022-04-26:
Il 14/04/22 10:08, DJ2LS ha scritto:
> Il 14/04/22 09:56, DJ2LS ha scritto:
>     Now I am away from my home and unable to do more tests. What can I
>     do, when back to home, is put a mike in front of radio speaker and
>     try another [...]

As per my previous post, also old Appimages don't show waterfall.

I have removed any node and npm package from my PC, removed also same 
files in /usr/local/bin and /usr/local/lib.
Redone also git clone command but no change at all after re-install.

Waterfall is not showing.

So I suppose a change in some library in my Ubuntu 20.04

I can install a 22.04 virtual machine and do some test on this virtual 
machine. I need to study for this how to redirect my USB sound card (my 
be Pulse?) to this machine.

Regards

Franco Spinelli
IW2DHW

frspin, 2022-04-26:
Solved
It was a my fault
In config.json file from old run "spectrum": "waterfall" was not set.
And I I have not set it in new GUI!
Setting this and now waterfall is showing and "pulse" device is present.

Regards
Franco Spinelli
IW2DHW

closed 2022-04-26 by DJ2LS

DJ2LS, 2022-10-11:
Hi Franco, it’s possibly a problem with your audio settings I guess 🤔 as the library changes only happened with the newer releases. As a test you could install all required packages manually (see  https://wiki.freedata.app )

Did you try with pavucontrol?

I had this problem also when using pulse. But with the latest release this has been working as expected 🤔 really interesting.

frspin, 2022-10-11:
Il 08/04/22 12:09, DJ2LS ha scritto:
> Okay, however, we seem to have fixed another bug which caused a missing
> pulse device.
> It seems it also takes some time (0-10 seconds in may case), until used
> devices will be released again under [...]

After an hour device returned!
I have run FD at 13.37 and, at second run, USB device are missed.
Now, after 7', USB device is still missed.
If this is the problem, I can suppose it is a problem of minutes, not of 
seconds.

> 
> I also had the problem of the stopped waterfall, when not selecting the 
> correct input device in pavucontrol. You can change this while running 
> "pulse" device within FD without the need of closing it.

Device is "pulse" in input and also in output
In pavucontrol device is redirected to USB audio device
Pavucontrol show activity on device
Waterfall is still missed

All as in attached video (1.2 Mb - if compatible with GitHub)

Regards

Franco Spinelli
IW2DHW

frspin, 2022-10-11:
Il 11/04/22 12:29, DJ2LS ha scritto:
> What about the sample rate? FreeDATA needs 48000Hz

Default sample rate for this card is 44100 but Pulse is able to do any 
required conversion. I am using it for any digital program, all using 
Pulse or Portaudio but all using this libraries from C or C++

This setup, using Pulse and also using USB audio directly, was working 
until sound library change.

Franco Spinelli
IW2DHW

DJ2LS, 2022-10-11:
Are you running from source? Latest audio from source is broken. Fix is ready to be merged

DJ2LS, 2022-10-11:
Possible Fix is now merged to main branch.