DJ2LS/FreeDATA #156: Have TNC respond to ping without DX Call - Can anybody here me?

kronenpj, 2022-04-10:
**Is your feature request related to a problem? Please describe.**
A nice feature of JS8Call allows the operator to send a 'heartbeat' message and any station on frequency has the option to respond with a signal report for that station. It allows an operator to have an idea of how far and how well the station is heard.

**Describe the solution you'd like**
I'd like to propose that the TNC respond to a PING packet without a DX Call.

**Describe alternatives you've considered**
Adding a new packet type isn't necessary as the current behavior for an un-directed PING is to ignore the packet. 

**Additional context**
I would be happy to provide a pull request that extends the TNC code to this end, if it's something you believe would be useful. I wouldn't try to extend the content of the PING response to include the Signal-to-Noise ratio until a later request.

DJ2LS, 2022-04-10:
Thanks for the idea! This shouldn't be that hard to implement, we just need to open the "callsign crc filter".

How should we handle the situation, if several stations are receiving the signal and going to respond?  Are there some best practices yet for this? Maybe a random timer?

Thanks, Simon

kronenpj, 2022-04-10:
A random timer is probably the only way to do it. There will be inevitable collisions but I don't know of another way, unless we do something based on a has of the station's callsign or location. Right now I'm using `time.sleep(randrange(1, 100) / 100.0)` for a 0.01 to 1 second delay.

Until I figure out the configuration option, I added a condition for `frametype == 210` at line 209:
`if helpers.check_callsign(self.mycallsign, bytes(bytes_out[1:3])) or helpers.check_callsign(self.mycallsign, bytes(bytes_out[2:4])) or frametype == 200 or frametype == 210 or frametype == 250:`
This should address the callsign CRC filter.

DJ2LS, 2022-04-10:
We could use the beacon as a heartbeat maybe. If we received a ping with a special crc, calculated by an empty Callsign or some wildcard character, we then answer with a beacon. But answering with a ping ack may be the cleanest way in my opinion. 
However, I like the idea, but we should make it an option, the user can decide if it's station will answer for "empty pings" or not.

DJ2LS, 2022-04-10:
Sorry I accidentally edited instead of quoting ...

> A random timer is probably the only way to do it. There will be inevitable collisions but I don't know of another way, unless we do something based on a has of the station's callsign or location. Right now I'm using `time.sleep(randrange(1, 100) / 100.0)` for a 0.01 to 1 second delay.
> 
> 
> 
> Until I figure out the configuration option, I added a condition for `frametype == 210` at line 209:
> 
> `if helpers.check_callsign(self.mycallsign, bytes(bytes_out[1:3])) or helpers.check_callsign(self.mycallsign, bytes(bytes_out[2:4])) or frametype == 200 or frametype == 210 or frametype == 250:`
> 
> This should address the callsign CRC filter.
> 
> 
> 
> 
> I suggest modifying helpers.check_callsign(self.mycallsign, bytes(bytes_out[1:3])) within helpers.py for doing this from my actual point of view. It returns true if we received a valid crc and also adds additional crc for different SSIDs
> 
> We also should use  a longer delay range as one frame is about 0.7seconds, I think 🤔

DJ2LS, 2022-04-10:
> A random timer is probably the only way to do it. There will be inevitable collisions but I don't know of another way, unless we do something based on a has of the station's callsign or location. [...]

I suggest modifying helpers.check_callsign(self.mycallsign, bytes(bytes_out[1:3])) within helpers.py for doing this from my actual point of view. It returns true if we received a valid crc and also adds additional crc for different SSIDs

We also should use  a longer delay range as one frame is about 0.7seconds, I think 🤔

kronenpj, 2022-04-10:
I was wondering why I wrote that :). I'll set the step to be 8 and maybe increase the top to 240. 

I'm currently having trouble having the station decode the PING ACK packet now that I've extended it to 24 bytes and added static.MYCALLSIGN to bytes 13-23. I see that it sends but never decodes on the other end.

DJ2LS, 2022-04-10:
Oh, yes because a signalling frame has a payload of 14 bytes. And we have not that much left. We are also working at this area with a special encoding for Callsign compression.

So in this case we need to modify the receiving modem which is a little bit tricky and should be avoided in this case.

kronenpj, 2022-04-10:
What I have now, without the responding callsign in the packet, works:

```
2022-04-10 18:48:04 [debug    ] [SCK] CMD                      command={'type': 'ping', 'command': 'ping', 'dxcallsign': ''}
2022-04-10 18:48:04 [info     ] [TNC] PING REQ [W4MCO-0] >>> [-0]
2022-04-10 18:48:04 [info     ] ENABLE FSK                     state=False
2022-04-10 18:48:04 [debug    ] TRANSMIT                       mode=14 payload=14
8.9
self.n_retries_per_burst = 0
bytearray(b'W4MCO-0')
2022-04-10 18:48:07 [debug    ] PING ACK RECEIVED....
2022-04-10 18:48:07 [info     ] [TNC] PING ACK [W4MCO-0] >|< [-0] snr=8.9
```

But the benefit of seeing who responded is being able to start a chat with them. :)

kronenpj, 2022-04-10:
This is what I have so far. 

```patch
diff --git a/tnc/data_handler.py b/tnc/data_handler.py
index 02c21c7..3c7d10b 100644
--- a/tnc/data_handler.py
+++ b/tnc/data_handler.py
@@ -206,7 +206,7 @@ class DATA():
 
 
         #if bytes(bytes_out[1:3]) == self.mycallsign_CRC or bytes(bytes_out[2:4]) == self.mycallsign_CRC or frametype == 200 or frametype == 250:
-        if helpers.check_callsign(self.mycallsign, bytes(bytes_out[1:3])) or helpers.check_callsign(self.mycallsign, bytes(bytes_out[2:4])) or frametype == 200 or frametype == 250:
+        if helpers.check_callsign(self.mycallsign, bytes(bytes_out[1:3])) or helpers.check_callsign(self.mycallsign, bytes(bytes_out[2:4])) or frametype in [200, 210, 250]:
         
             # CHECK IF FRAMETYPE IS BETWEEN 10 and 50 ------------------------
             frame = frametype - 10
@@ -1475,9 +1475,13 @@ class DATA():
         static.INFO.append("PING;RECEIVING")
 
         # check if callsign ssid override
-        mycallsign = helpers.check_callsign(self.mycallsign, data_in[1:3])[1]        
-        
-        structlog.get_logger("structlog").info("[TNC] PING REQ [" + str(mycallsign, 'utf-8') + "] <<< [" + str(static.DXCALLSIGN, 'utf-8') + "]", snr=static.SNR )
+        try:
+            mycallsign = helpers.check_callsign(self.mycallsign, data_in[1:3])[1]        
+            structlog.get_logger("structlog").info("[TNC] PING REQ [" + str(mycallsign, 'utf-8') + "] <<< [" + str(static.DXCALLSIGN, 'utf-8') + "]", snr=static.SNR )
+        except TypeError:
+            structlog.get_logger("structlog").info("[TNC] PING REQ [" + "(NOCALL)" + "] <<< [" + str(static.DXCALLSIGN, 'utf-8') + "]", snr=static.SNR )
+            # Sleep a random amount of time before responding to make it more likely to be heard when many stations respond.
+            time.sleep(randrange(1, 240, 8) / 100.0)
 
         ping_frame      = bytearray(14)
         ping_frame[:1]  = bytes([211])
@@ -1485,6 +1489,7 @@ class DATA():
         ping_frame[3:5] = static.MYCALLSIGN_CRC
         ping_frame[5:11] = static.MYGRID
         ping_frame[11:13] = frequency_offset.to_bytes(2, byteorder='big', signed=True)
+        # ping_frame[13:23] = helpers.callsign_to_bytes(static.MYCALLSIGN)
 
         txbuffer = [ping_frame]
         static.TRANSMITTING = True
@@ -1493,6 +1498,7 @@ class DATA():
             modem.MODEM_TRANSMIT_QUEUE.put(['FSK_LDPC_0',1,0,txbuffer])
         else:
             modem.MODEM_TRANSMIT_QUEUE.put([14,1,0,txbuffer])
+        structlog.get_logger("structlog").info("[TNC] SEND PING ACK [" + str(self.mycallsign, 'utf-8') + "] >|< [" + str(static.DXCALLSIGN, 'utf-8') + "]")
         # wait while transmitting
         while static.TRANSMITTING:
             time.sleep(0.01)
@@ -1509,6 +1515,10 @@ class DATA():
 
         static.DXCALLSIGN_CRC = bytes(data_in[3:5])
         static.DXGRID = bytes(data_in[5:11]).rstrip(b'\x00')
+        try:
+            static.DXCALLSIGN = helpers.bytes_to_callsign(bytes(data_in[13:23]))
+        except IndexError:
+            pass
         
         jsondata = {"type" : "ping", "status" : "ack",  "uuid" : str(uuid.uuid4()),  "timestamp": int(time.time()), "mycallsign" : str(self.mycallsign, 'utf-8'), "dxcallsign": str(static.DXCALLSIGN, 'utf-8'), "dxgrid": str(static.DXGRID, 'utf-8'), "snr": str(static.SNR)}
         json_data_out = json.dumps(jsondata)
```

kronenpj, 2022-04-10:
Here's an idea - If a station receives one of these PING packets, originate a PING packet directed to that station.

DJ2LS, 2022-04-10:
🥳

I like this part: ```
frametype in [200, 210, 250]:
``` as it's more clean as before. This should be updated! 

But I suggest adding a crc check for a special wildcard-callsign like "ALL(-SSID)" to the helpers function as it's more homogeneous to the other crc checks, instead of whitelisting the entire ping frame and checking for an exception. What happens, if an other error occurs during a Callsign check --> in this case I would prefer a more precise check. 
Do you know what I want to say? ( it's getting late here - brain not that fresh anymore )

kronenpj, 2022-04-10:
I had a thought - If a station receives one of these (empty or ALL) PING packets, originate a PING packet directed to that station. This obviates the need for creating a special case for the response, but it does add another packet to the airwaves.

I can do that. That would open up all packets crafted to the "ALL(-SSID)". If that's desired, easy enough. I'm not sure the overall ramifications.

Also, in the current code, if a PING packet is heard but it's not for this station, it's not reported in the GUI and the debug log shows this:
`2022-04-10 20:35:07 [debug    ] [TNC] Unknown frame received   frame=b'\xd2)B\x99\x17W4MCO\x00\x00\x00\x00'`. Would we not prefer to properly decode and display this packet in the "Heard Stations" box?

kronenpj, 2022-04-10:
Current state. I need to split this into two PRs. The change to `check_callsign` wasn't completed, which is another issue. For this discussion, the code is working as I'd hoped it would.
```patch
diff --git a/tnc/data_handler.py b/tnc/data_handler.py
index 02c21c7..49b10da 100644
--- a/tnc/data_handler.py
+++ b/tnc/data_handler.py
@@ -206,7 +206,9 @@ class DATA():
 
 
         #if bytes(bytes_out[1:3]) == self.mycallsign_CRC or bytes(bytes_out[2:4]) == self.mycallsign_CRC or frametype == 200 or frametype == 250:
-        if helpers.check_callsign(self.mycallsign, bytes(bytes_out[1:3])) or helpers.check_callsign(self.mycallsign, bytes(bytes_out[2:4])) or frametype == 200 or frametype == 250:
+        _valid1, _ = helpers.check_callsign(self.mycallsign, bytes(bytes_out[1:3]))
+        _valid2, _ = helpers.check_callsign(self.mycallsign, bytes(bytes_out[2:4]))
+        if _valid1 or _valid2 or frametype in [200, 210, 250]:
         
             # CHECK IF FRAMETYPE IS BETWEEN 10 and 50 ------------------------
             frame = frametype - 10
@@ -265,7 +267,7 @@ class DATA():
                 structlog.get_logger("structlog").debug("PING RECEIVED....")
                 # = self.get_frequency_offset(freedv)
                 # we need to fix this later
-                frequency_offset = 0
+                frequency_offset = "0"
                 #print("Freq-Offset: " + str(frequency_offset))
                 self.received_ping(bytes_out[:-2], frequency_offset)
                 
@@ -360,12 +362,17 @@ class DATA():
         # get received crc for different mycall ssids
         self.received_mycall_crc = data_in[2:4]  
         
+        global TESTMODE
+        
         # check if callsign ssid override
-        mycallsign = helpers.check_callsign(self.mycallsign, self.received_mycall_crc)[1] 
+        valid, mycallsign = helpers.check_callsign(self.mycallsign, self.received_mycall_crc)
         print(mycallsign)
+        if not valid:
+            # ARQ data packet not for me.
+            if not TESTMODE:
+                self.arq_cleanup()        
+            return
         
-        
-        global TESTMODE
         # only process data if we are in ARQ and BUSY state else return to quit
         if not static.ARQ_STATE and static.TNC_STATE != 'BUSY':
             return
@@ -569,7 +576,12 @@ class DATA():
                 
  
                 # check if callsign ssid override
-                mycallsign = helpers.check_callsign(self.mycallsign, self.received_mycall_crc)[1] 
+                valid, mycallsign = helpers.check_callsign(self.mycallsign, self.received_mycall_crc)
+                if not valid:
+                    # ARQ data packet not for me.
+                    if not TESTMODE:
+                        self.arq_cleanup()        
+                    return
                 
                 base64_data = base64.b64encode(data_frame)
                 base64_data = base64_data.decode("utf-8")
@@ -1341,7 +1353,12 @@ class DATA():
         helpers.add_to_heard_stations(static.DXCALLSIGN,static.DXGRID, 'DATA-CHANNEL', static.SNR, static.FREQ_OFFSET, static.HAMLIB_FREQUENCY)
         
         # check if callsign ssid override
-        mycallsign = helpers.check_callsign(self.mycallsign, data_in[1:3])[1] 
+        valid, mycallsign = helpers.check_callsign(self.mycallsign, data_in[1:3])
+        if not valid:
+            # ARQ connect packet not for me.
+            if not TESTMODE:
+                self.arq_cleanup()        
+            return
             
         structlog.get_logger("structlog").info("[TNC] ARQ | DATA | RX | [" + str(mycallsign, 'utf-8') + "]>> <<[" + str(static.DXCALLSIGN, 'utf-8') + "]", bandwith="wide")
            
@@ -1475,8 +1492,19 @@ class DATA():
         static.INFO.append("PING;RECEIVING")
 
         # check if callsign ssid override
-        mycallsign = helpers.check_callsign(self.mycallsign, data_in[1:3])[1]        
-        
+        valid, mycallsign = helpers.check_callsign(self.mycallsign, data_in[1:3])
+        if not valid:
+            # PING packet not for me.
+            # But is it a "NO-CALL" PING? Check against an 'empty' callsign.
+            valid, _ = helpers.check_callsign(bytes(), data_in[1:3])
+            if valid:
+                structlog.get_logger("structlog").info("[TNC] PING HEARD [" + "(NOCALL)" + "] <<< [" + str(static.DXCALLSIGN, 'utf-8') + "]", snr=static.SNR )
+                # Sleep a random amount of time before responding to make it more likely to be heard when many stations respond.
+                time.sleep(randrange(1, 240, 16) / 100.0)
+                self.transmit_ping(static.DXCALLSIGN)
+            # Either way, skip the remainder of received_ping.
+            return
+
         structlog.get_logger("structlog").info("[TNC] PING REQ [" + str(mycallsign, 'utf-8') + "] <<< [" + str(static.DXCALLSIGN, 'utf-8') + "]", snr=static.SNR )
 
         ping_frame      = bytearray(14)
@@ -1493,6 +1521,7 @@ class DATA():
             modem.MODEM_TRANSMIT_QUEUE.put(['FSK_LDPC_0',1,0,txbuffer])
         else:
             modem.MODEM_TRANSMIT_QUEUE.put([14,1,0,txbuffer])
+        structlog.get_logger("structlog").info("[TNC] SEND PING ACK [" + str(self.mycallsign, 'utf-8') + "] >|< [" + str(static.DXCALLSIGN, 'utf-8') + "]")
         # wait while transmitting
         while static.TRANSMITTING:
             time.sleep(0.01)
```

DJ2LS, 2022-04-10:
I'm not sure if I understood you correctly - when receiving a wildcard ping, initiate a ping ( and wait a ping ack ) to the station which sent the wildcard ping? If so, why not running this when receiving CQ frames? This is maybe easier to understand than a wildcard ping --> "setting: respond to CQ with ping"

And yes, you're right- we should add those "unknown frames" to the Heard- stations list!

kronenpj, 2022-04-10:
Yes, not only does it cleanly re-use existing code, we don't need to do anything special except to decide that the un-directed / wildcard PING is something the operator has decided to respond to.

To me, answering a CQ might be interpreted by the operator as intent to start a chat, as is the usual intent. For instance, I may want to leave my station on and respond to PING requests but not be interested in a chat. For those familiar with networking PING is a diagnostic tool, albeit a primitive one. I'd prefer to keep it as such, rather than responding to a CQ or "I'm looking for someone to talk to" with a diagnostic message.  I realize the difference is subtle but, to me, the intent is different.

kronenpj, 2022-04-10:
Another way to state my concern is to imagine this becoming popular and being used on a crowded frequency - lots of stations within range and lots of people wanting to talk. Do we want to give people the option to respond to these CQ transmits with PING requests? 

I believe it's far better for people who don't read directions to not produce more QRM than necessary by accidentally turning on respond to PING with PING rather than respond to CQ with PING.

DJ2LS, 2022-04-11:
> Another way to state my concern is to imagine this becoming popular and being used on a crowded frequency - lots of stations within range and lots of people wanting to talk. Do we want to give people the option to respond to these CQ transmits with PING requests?
> 
> I believe it's far better for people, who don't read directions, to not produce more QRM than necessary by accidentally turning on respond to PING with PING rather than respond to CQ with PING. PING should be used far less than CQ in this future.

Yes, that's what I meant here https://github.com/DJ2LS/FreeDATA/pull/158#issuecomment-1094377165

A Ping is always bidirectional, where we want a behavior which is more a broadcast and we answering it. So a "wildcard"-Ping, answering with a Ping can be confusing.
We could do this by responding to a CQ with a "wake up frame", maybe? A wakeup frame could be a broadcast without any specific destination callsign. So its more like "CQ: Are you there?" "WakeUp: Jup, I'm here - but not only for you - for all others out there as well"

kronenpj, 2022-04-15:
I'm reconsidering using specific frames for this feature. Dedicated frame types makes the intention clear and responses optional on a granular level. I've closed the pull request with the original ping-based implementation while I work on this approach.

kronenpj, 2022-04-15:
I have a proof of concept using the existing BEACON frame (#250) and extended with a (to-be optional) BREPLY frame (#251), spaced out at random 50ms intervals with allowance for 10 "time slots". With my two soundcard-only "stations" it appears to work. The UI will need to be extended to enable the option and, I would suggest, increasing the minimum BEACON interval.

```patch
diff --git a/tnc/data_handler.py b/tnc/data_handler.py
index 7cf2a72..29e6478 100644
--- a/tnc/data_handler.py
+++ b/tnc/data_handler.py
@@ -208,7 +208,7 @@ class DATA():
         #if bytes(bytes_out[1:3]) == self.mycallsign_CRC or bytes(bytes_out[2:4]) == self.mycallsign_CRC or frametype == 200 or frametype == 250:
         _valid1, _ = helpers.check_callsign(self.mycallsign, bytes(bytes_out[1:3]))
         _valid2, _ = helpers.check_callsign(self.mycallsign, bytes(bytes_out[2:4]))
-        if _valid1 or _valid2 or frametype in [200, 210, 250]:
+        if _valid1 or _valid2 or frametype in [200, 210, 250, 251]:
         
             # CHECK IF FRAMETYPE IS BETWEEN 10 and 50 ------------------------
             frame = frametype - 10
@@ -332,6 +332,11 @@ class DATA():
                 structlog.get_logger("structlog").debug("BEACON RECEIVED")
                 self.received_beacon(bytes_out[:-2])
 
+            # this is outdated and we may remove it
+            elif frametype == 251:
+                structlog.get_logger("structlog").debug("BEACON REPLY!")
+                self.received_beacon_reply(bytes_out[:-2])
+
             # TESTFRAMES
             elif frametype == 255:
                 structlog.get_logger("structlog").debug("TESTFRAME RECEIVED", frame=bytes_out[:])

@@ -1636,7 +1641,7 @@ class DATA():
         Returns:
 
         """
-    # here we add the received station to the heard stations buffer
+        # here we add the received station to the heard stations buffer
         dxcallsign = helpers.bytes_to_callsign(bytes(data_in[1:9]))
         dxgrid = bytes(data_in[9:13]).rstrip(b'\x00')
 
@@ -1648,6 +1653,53 @@ class DATA():
         structlog.get_logger("structlog").info("[TNC] BEACON RCVD [" + str(dxcallsign, 'utf-8') + "]["+ str(dxgrid, 'utf-8') +"] ", snr=static.SNR)
         helpers.add_to_heard_stations(dxcallsign,dxgrid, 'BEACON', static.SNR, static.FREQ_OFFSET, static.HAMLIB_FREQUENCY)
 
+        # TODO: Add configuration option to enable responding to beacons.
+        # Sleep a random amount of time before responding to make it more likely to be
+        # heard when many stations respond. Each DATAC0 frame is 0.44 sec (44ms) in 
+        # duration, plus an offset for TX->RX at the beacon station, set the wait 
+        # interval to be 6ms + (50ms * random).
+        time.sleep(randrange(6, 506, 50) / 100.0)
+        static.INFO.append("BEACON;SENDING")
+        structlog.get_logger("structlog").info("[TNC] Sending beacon response!", interval=self.beacon_interval)
+
+        beacon_frame = bytearray(14)
+        beacon_frame[:1]   = bytes([251])
+        beacon_frame[1:9]  = helpers.callsign_to_bytes(self.mycallsign)
+        beacon_frame[9:13] = static.MYGRID[:4]
+        txbuffer = [beacon_frame]
+
+        static.TRANSMITTING = True
+        structlog.get_logger("structlog").info("ENABLE FSK", state=static.ENABLE_FSK)
+        if static.ENABLE_FSK:
+            modem.MODEM_TRANSMIT_QUEUE.put(['FSK_LDPC_0',1,0,txbuffer])
+        else:
+            modem.MODEM_TRANSMIT_QUEUE.put([14,1,0,txbuffer])
+
+        # wait while transmitting
+        while static.TRANSMITTING:
+            time.sleep(0.01)
+
+    def received_beacon_reply(self, data_in:bytes):
+        """
+        Called if we receive a beacon reply
+        Args:
+          data_in:bytes:
+
+        Returns:
+
+        """
+        # here we add the received station to the heard stations buffer
+        dxcallsign = helpers.bytes_to_callsign(bytes(data_in[1:9]))
+        dxgrid = bytes(data_in[9:13]).rstrip(b'\x00')
+
+        jsondata = {"type" : "breply", "status" : "received",  "uuid" : str(uuid.uuid4()),  "timestamp": int(time.time()), "mycallsign" : str(self.mycallsign, 'utf-8'), "dxcallsign": str(dxcallsign, 'utf-8'), "dxgrid": str(dxgrid, 'utf-8'), "snr": str(static.SNR)}
+        json_data_out = json.dumps(jsondata)
+        sock.SOCKET_QUEUE.put(json_data_out)
+
+        static.INFO.append("BREPLY;RECEIVING")
+        structlog.get_logger("structlog").info("[TNC] BEACON REPLY RCVD [" + str(dxcallsign, 'utf-8') + "]["+ str(dxgrid, 'utf-8') +"] ", snr=static.SNR)
+        helpers.add_to_heard_stations(dxcallsign,dxgrid, 'BREPLY', static.SNR, static.FREQ_OFFSET, static.HAMLIB_FREQUENCY)
+
 
 
     def transmit_cq(self):
```

Let me know your thoughts on this approach.

DJ2LS, 2022-04-15:
Thanks, @kronenpj this is the way I thought about this as well with one small difference: 

**Thought**: Instead of using frametype `251` as a `BEACONREPLY`, what about using frametype `201` `QRV`?
**Reason**: A beacon for myself is a simplex transmission, which we are not going to reply as they are used for a propagation tests or broadcasting a "I'm there" signal. A possible problem could be,  because of the fact, that a beacon is an automated transmission, a high channel load could be caused, specially if multiple station are running a beacon. For this reason I would suggest triggering a `QRV Response` should only be possible by receiving a manually transmitted `CQ (frametype 200)`. 

**Random time offset:** DATAC0 has a "runtime" of about 0.44s WITHOUT postamble and preamble! This also depends on processing speed when generating modulation, so I suggest using a higher runtime of about 0,8-1 seconds ( we might need to test this with multiple station as well for time optimizing this ).

kronenpj, 2022-04-15:
Ok, more like this then.

```patch
diff --git a/tnc/data_handler.py b/tnc/data_handler.py
index 010446d..fbfdf46 100644
--- a/tnc/data_handler.py
+++ b/tnc/data_handler.py
@@ -208,7 +208,7 @@ class DATA():
         #if bytes(bytes_out[1:3]) == self.mycallsign_CRC or bytes(bytes_out[2:4]) == self.mycallsign_CRC or frametype == 200 or frametype == 250:
         _valid1, _ = helpers.check_callsign(self.mycallsign, bytes(bytes_out[1:3]))
         _valid2, _ = helpers.check_callsign(self.mycallsign, bytes(bytes_out[2:4]))
-        if _valid1 or _valid2 or frametype in [200, 210, 250]:
+        if _valid1 or _valid2 or frametype in [200, 210, 250, 251]:
         
             # CHECK IF FRAMETYPE IS BETWEEN 10 and 50 ------------------------
             frame = frametype - 10
@@ -262,6 +262,12 @@ class DATA():
                 structlog.get_logger("structlog").debug("CQ RECEIVED....")
                 self.received_cq(bytes_out[:-2])
 
+            # QRV FRAME
+            elif frametype == 201:
+                structlog.get_logger("structlog").debug("QRV RECEIVED!")
+                self.received_qrv(bytes_out[:-2])
+
+
             # PING FRAME
             elif frametype == 210:
                 structlog.get_logger("structlog").debug("PING RECEIVED....")
@@ -1628,7 +1639,7 @@ class DATA():
         Returns:
 
         """
-    # here we add the received station to the heard stations buffer
+        # here we add the received station to the heard stations buffer
         dxcallsign = helpers.bytes_to_callsign(bytes(data_in[1:9]))
         dxgrid = bytes(data_in[9:13]).rstrip(b'\x00')
 
@@ -1684,6 +1694,56 @@ class DATA():
         structlog.get_logger("structlog").info("[TNC] CQ RCVD [" + str(dxcallsign, 'utf-8') + "]["+ str(dxgrid, 'utf-8') +"] ", snr=static.SNR)
         helpers.add_to_heard_stations(dxcallsign, dxgrid, 'CQ CQ CQ', static.SNR, static.FREQ_OFFSET, static.HAMLIB_FREQUENCY)
 
+        # TODO: Add configuration option to enable responding to CQ.
+        # Sleep a random amount of time before responding to make it more likely to be
+        # heard when many stations respond. Each DATAC0 frame is 0.44 sec (44ms) in
+        # duration, plus an offset for TX->RX at the beacon station, set the wait
+        # interval to be random between 0 and 9 seconds.
+        time.sleep(randrange(0, 10))
+        static.INFO.append("QRV;SENDING")
+        structlog.get_logger("structlog").info("[TNC] Sending QRV!")
+
+        qrv_frame = bytearray(14)
+        qrv_frame[:1]   = bytes([201])
+        qrv_frame[1:9]  = helpers.callsign_to_bytes(self.mycallsign)
+        qrv_frame[9:13] = static.MYGRID[:4]
+        txbuffer = [qrv_frame]
+
+        static.TRANSMITTING = True
+        structlog.get_logger("structlog").info("ENABLE FSK", state=static.ENABLE_FSK)
+        if static.ENABLE_FSK:
+            modem.MODEM_TRANSMIT_QUEUE.put(['FSK_LDPC_0',1,0,txbuffer])
+        else:
+            modem.MODEM_TRANSMIT_QUEUE.put([14,1,0,txbuffer])
+
+        # wait while transmitting
+        while static.TRANSMITTING:
+            time.sleep(0.01)
+
+
+
+    def received_qrv(self, data_in:bytes):
+        """
+        Called if we receive a QRV frame
+        Args:
+          data_in:bytes:
+
+        Returns:
+
+        """
+        # here we add the received station to the heard stations buffer
+        dxcallsign = helpers.bytes_to_callsign(bytes(data_in[1:9]))
+        dxgrid = bytes(data_in[9:13]).rstrip(b'\x00')
+
+        jsondata = {"type" : "qrv", "status" : "received",  "uuid" : str(uuid.uuid4()),  "timestamp": int(time.time()), "mycallsign" : str(self.mycallsign, 'utf-8'), "dxcallsign": str(dxcallsign, 'utf-8'), "dxgrid": str(dxgrid, 'utf-8'), "snr": str(static.SNR)}
+        json_data_out = json.dumps(jsondata)
+        sock.SOCKET_QUEUE.put(json_data_out)
+
+        static.INFO.append("QRV;RECEIVING")
+        structlog.get_logger("structlog").info("[TNC] QRV RCVD [" + str(dxcallsign, 'utf-8') + "]["+ str(dxgrid, 'utf-8') +"] ", snr=static.SNR)
+        helpers.add_to_heard_stations(dxcallsign,dxgrid, 'QRV', static.SNR, static.FREQ_OFFSET, static.HAMLIB_FREQUENCY)
+
+
     # ------------ CALUCLATE TRANSFER RATES
     def calculate_transfer_rate_rx(self, rx_start_of_transmission:float, receivedbytes:int) -> list:
         """
```

DJ2LS, 2022-04-15:
Looks good. Does the code work? I'm actually at work so I can't test it. 

But the QRV was just a thought. If you have any other idea - let's talk about this 🙂👍

We could merge the code and start a first public test to see how the users/tester can work with this.

kronenpj, 2022-04-15:
I'm testing it on my setup. I like the idea of putting out to the larger community.

DJ2LS, 2022-04-15:
Yes, thanks to the fact FreeDATA has different Update channels we can do tests like this with the alpha releases 🙂

kronenpj, 2022-04-15:
It's partially working. The response is triggered once for each successfully decoded CQ frame. I'm trying to think of ways to only send once per station's CQ-triple frame.  The original CQ station is also not properly decoding the QRV frame but I'll figure that one out pretty easily.

DJ2LS, 2022-04-15:
We could add an counter to the CQ frames to indicate which CQ frame we received, then only reply to the third CQ.

kronenpj, 2022-04-15:
That only works if all three CQ frames are received and properly decoded. I assume the purpose for sending 3 is to increase the odds that at least one is decoded properly.

DJ2LS, 2022-04-15:
The reason for 3 bursts is, it's easier to detect the signal by ear if someone is listening to the audio signals, than only one burst. So we could reduce it to one - which may be the best in this situation because of the possible QRV responses, I guess  🤔

kronenpj, 2022-04-15:
If you're ok with reducing the CQ frames to 1, please take a look at PR #169 as a candidate for a test for a wider audience.

closed 2022-04-16 by DJ2LS