DJ2LS/FreeDATA #173: Session disconnect too short --> causes heartbeat toggle

DJ2LS, 2022-04-19:
Session disconnect too short --> causes heartbeat toggle.
We need to track down this problem as its causes a really bad tnc state.

DJ2LS, 2022-04-26:
We also need to avoid the toggle somehow in case of a disconnect frame loss

kronenpj, 2022-05-01:
Unfortunately this fix causes the receiving TNC to remain in BUSY / OPEN after experiencing the same RF / audio state which caused issue #179. In this state, the TNC does not properly respond to GUI commands until the session times out in ~12 minutes.  

I may start writing tests for the TNC state machine transitions. I need to understand the `ctest` approach first.

DJ2LS, 2022-05-01:
Thanks, @kronenpj for having a look!

On which side does the TNC not respond to commands? ISS or IRS ?

Regarding to the ctests, please also have a look at https://github.com/DJ2LS/FreeDATA/issues/102

kronenpj, 2022-05-01:
IRS. I implemented the TODO from this commit and it fixes this case as well.

DJ2LS, 2022-05-01:
mh, there seems to be another problem with connecting - actually we can't establish a ARQ connection anymore... 🤔

kronenpj, 2022-05-01:
This appears to be fixed with PR #183.

commit ffff2566 (2022-04-30): arq disconnect improvement

this should fix and closes #173
closed 2022-05-01 by kronenpj