DJ2LS/FreeDATA #174: sounddevice present multiple times in the dropdown list

dk5sm, 2022-04-19:
The speaker/microphone sounddevice is always present multiple times in the respective dropdown lost.
It doesn't seem to make a difference which one you choose, but it is irritating.

Problem occured on Windows 10 and Windows 11:

[image]

kronenpj, 2022-06-25:
@dk5sm - Are you able / willing to build from source? If so, please clone the repository, check out `issue_174_fix` branch and build it. Let us know if that solves the issue for you.

kronenpj, 2022-06-26:
@dk5sm I figured out how to create builds manually. I've created a release on my fork:
https://github.com/kronenpj/FreeDATA-experiments/releases/tag/v0.4.3-alpha

dk5sm, 2022-06-26:
Paul,
I installed your fork-release on my Windows 10 machine, but it seems not to fix the issue. Instead, I have no more audio device:
[image]

kronenpj, 2022-06-26:
Stefan,

Hmm. I'll try again :).

kronenpj, 2022-06-26:
@dk5sm - I've re-built and re-uploaded the packages to the release (same URL). This time I tested that sound cards are listed. I can't go further as I don't run Windows on a machine connected to a radio.

dk5sm, 2022-06-27:
This build shows audio devices again, but it seems that all devices ever connected and also currently not present devices are in the list:
[image]

This is what is shown in widows control panel & device manager:
[image]

kronenpj, 2022-06-30:
@dk5sm I've added another (replaced) the packages in my repository. There will be an "error" level message listing the devices along with all the attributes the `sounddevice` module collects. I'll try to use that to filter out non-installed devices - if enough information is provided.

kronenpj, 2022-09-17:
@dk5sm I've updated the build and have new packages for you to try. They're available for download here:
https://github.com/DJ2LS/FreeDATA/actions/runs/3071502608

Please try them and let me know if the problem is resolved.
@DJ2LS

DJ2LS, 2022-11-01:
Next release adds device checksums so they can be identified. 
A first test run shows, that we have unique checksums, therefore we have different devices. Next step will be, determining the differences between equal name

DJ2LS, 2022-11-08:
Solved, because it's a feature, not a bug. Added Host-API Names to Device Name

closed 2022-11-08 by DJ2LS