DJ2LS/FreeDATA #181: Where did the file transfer feature go?

pe1rrr, 2022-04-27:
I updated to the latest git main branch and a lot has changed, I can't find the file transfer system anymore, and the personal chat system doesn't appear to work at all so not quite sure what is left to actually test?

DJ2LS, 2022-04-27:
Hi @pe1rrr ,

Yes a lot of parts have been changed. The file transfer is now included within the chat. You can start a new chat by opening a new one on the Callsign list (green top button within chat interface ). 

It would be interesting to know, which parts are confusing, so we could improve them. And if there's a bug - don't hesitate opening an issue 🙂

Simon

pe1rrr, 2022-04-27:
That would explain why nothing works anymore :)

https://youtu.be/YwManXB4PQc

DJ2LS, 2022-04-30:
Hm, it seems my answer I wrote you via email isn't posted here 🤔

Did you run `npm i`?

pe1rrr, 2022-05-01:
> Hm, it seems my answer I wrote you via email isn't posted here 🤔
> 
> Did you run `npm i`?

Yes, I always do this after a fresh reinstall. The chat window doesn't respond to anything on either systems here. It must be related to the version of some lib that is used on Raspbian? Can you post an example of what is supposed to happen as that would be useful to see

DJ2LS, 2022-05-02:
There should be a new tab with the selected callsign 🤔
Could you open the development console with Ctrl+Shift+i then selecting "Console" and check for the error message when opening a new chat? 
This error is new to me 🤔

pe1rrr, 2022-05-02:
> There should be a new tab with the selected callsign 🤔
> Could you open the development console with Ctrl+Shift+i then selecting "Console" and check for the error message when opening a new chat? 
> This error is new to me 🤔

VM113 renderer_init:73 Unable to load preload script: /mnt/disk1/home/pi/git/FreeDATA/gui/preload-chat.js
VM113 renderer_init:73 Error: No native build was found for platform=linux arch=arm runtime=electron abi=103 uv=1 armv=default libc=glibc
    at Function.load.path (VM252 index.js:56:9)
    at load (VM252 index.js:20:30)
    at Object.<anonymous> (VM251 binding.js:1:198)
    at Object.<anonymous> (VM251 binding.js:3:3)
    at Module._compile (VM94 loader:1116:14)
    at Object.Module._extensions..js (VM94 loader:1169:10)
    at Module.load (VM94 loader:988:32)
    at Module._load (VM94 loader:829:12)
    at Function.c._load (VM110 asar_bundle:5:13343)
    at Function.i._load (VM113 renderer_init:33:356)

DJ2LS, 2022-05-03:
This might be a problem with raspberry pi 🤔 I need to test this the next days.

pe1rrr, 2022-05-18:
Any news on this? :)

The most recent pull today i find that the GUI is completely unresponsive (slow as heck, like- painfully), there's an error when I have [x] selected for rigctl (no rigctl) because neither rigctl network works.

```
2022-05-18 21:42:47 [debug    ] [SCK] CMD                      command={'type': 'set', 'command': 'stop_tnc', 'parameter': '---'}
2022-05-18 21:42:47 [warning  ] [DMN] Stopping TNC
2022-05-18 21:44:14 [debug    ] [SCK] CMD                      command={'type': 'set', 'command': 'start_tnc', 'parameter': [{'mycall': 'PE1RRR-0', 'mygrid': 'jo21lo', 'rx_audio': '1', 'tx_audio': '1', 'radiocontrol': 'rigctld', 'devicename': 'RIG_MODEL_DUMMY_NOVFO', 'deviceport': '/dev/ttyAMA0', 'pttprotocol': 'NONE', 'pttport': '/dev/ttyAMA0', 'serialspeed': '9600', 'data_bits': '8', 'stop_bits': '1', 'handshake': 'None', 'rigctld_port': '4599', 'rigctld_ip': '192.168.1.42', 'enable_scatter': 'False', 'enable_fft': 'True', 'enable_fsk': 'False', 'low_bandwith_mode': 'False', 'tuning_range_fmin': '-50.0', 'tuning_range_fmax': '50.0', 'tx_audio_level': '125', 'respond_to_cq': 'True'}]}
2022-05-18 21:44:14 [debug    ] [DMN] TNC Startup Config : mycall value=PE1RRR-0
2022-05-18 21:44:14 [debug    ] [DMN] TNC Startup Config : mygrid value=jo21lo
2022-05-18 21:44:14 [debug    ] [DMN] TNC Startup Config : rx_audio value=1
2022-05-18 21:44:14 [debug    ] [DMN] TNC Startup Config : tx_audio value=1
2022-05-18 21:44:14 [debug    ] [DMN] TNC Startup Config : radiocontrol value=rigctld
2022-05-18 21:44:14 [debug    ] [DMN] TNC Startup Config : devicename value=RIG_MODEL_DUMMY_NOVFO
2022-05-18 21:44:14 [debug    ] [DMN] TNC Startup Config : deviceport value=/dev/ttyAMA0
2022-05-18 21:44:14 [debug    ] [DMN] TNC Startup Config : pttprotocol value=NONE
2022-05-18 21:44:14 [debug    ] [DMN] TNC Startup Config : pttport value=/dev/ttyAMA0
2022-05-18 21:44:14 [debug    ] [DMN] TNC Startup Config : serialspeed value=9600
2022-05-18 21:44:14 [debug    ] [DMN] TNC Startup Config : data_bits value=8
2022-05-18 21:44:14 [debug    ] [DMN] TNC Startup Config : stop_bits value=1
2022-05-18 21:44:14 [debug    ] [DMN] TNC Startup Config : handshake value=None
2022-05-18 21:44:14 [debug    ] [DMN] TNC Startup Config : rigctld_port value=4599
2022-05-18 21:44:14 [debug    ] [DMN] TNC Startup Config : rigctld_ip value=192.168.1.42
2022-05-18 21:44:14 [debug    ] [DMN] TNC Startup Config : enable_scatter value=False
2022-05-18 21:44:14 [debug    ] [DMN] TNC Startup Config : enable_fft value=True
2022-05-18 21:44:14 [debug    ] [DMN] TNC Startup Config : enable_fsk value=False
2022-05-18 21:44:14 [debug    ] [DMN] TNC Startup Config : low_bandwith_mode value=False
2022-05-18 21:44:14 [debug    ] [DMN] TNC Startup Config : tuning_range_fmin value=-50.0
2022-05-18 21:44:14 [debug    ] [DMN] TNC Startup Config : tuning_range_fmax value=50.0
2022-05-18 21:44:14 [debug    ] [DMN] TNC Startup Config : tx_audio_level value=125
2022-05-18 21:44:14 [debug    ] [DMN] TNC Startup Config : respond_to_cq value=True
2022-05-18 21:44:14 [warning  ] [DMN] Starting TNC             port=/dev/ttyAMA0 rig=RIG_MODEL_DUMMY_NOVFO
2022-05-18 21:44:14 [error    ] [DMN] worker: Exception:       e=FileNotFoundError(2, 'No such file or directory')
2022-05-18 21:44:14 [info     ] [DMN] TNC started              path=source
2022-05-18 23:44.15 [info     ] [C2 ] Searching for libcodec2...
2022-05-18 23:44.15 [warning  ] [C2 ] Libcodec2 found but not loaded e=OSError('lib/codec2/macOS/libcodec2.1.0.dylib: invalid ELF header') path=lib/codec2/macOS/libcodec2.1.0.dylib
2022-05-18 23:44.15 [warning  ] [C2 ] Libcodec2 found but not loaded e=OSError('lib/codec2/win64/libcodec2.dll: invalid ELF header') path=lib/codec2/win64/libcodec2.dll
2022-05-18 23:44.15 [warning  ] [C2 ] Libcodec2 found but not loaded e=OSError('lib/codec2/win32/libcodec2.dll: invalid ELF header') path=lib/codec2/win32/libcodec2.dll
2022-05-18 23:44.15 [info     ] [C2 ] Libcodec2 loaded         path=lib/codec2/linux/armv7l/libcodec2.so.1.0
2022-05-18 23:44.15 [debug    ] [HLP] callsign_to_bytes: Exception converting callsign to bytes: e=TypeError('encoding without a string argument')
2022-05-18 21:44:15 [info     ] [TNC] Starting FreeDATA        author=DJ2LS version=0.4.0-alpha year=2022
2022-05-18 21:44:15 [debug    ] [C2 ] Create 48<->8 kHz resampler
2022-05-18 21:44:15 [debug    ] [C2 ] Creating audio buffer    size=9600
2022-05-18 21:44:15 [debug    ] [C2 ] Creating audio buffer    size=9600
2022-05-18 21:44:15 [debug    ] [C2 ] Creating audio buffer    size=9600
2022-05-18 21:44:15 [debug    ] [C2 ] Creating audio buffer    size=4800
2022-05-18 21:44:15 [debug    ] [C2 ] Creating audio buffer    size=4800
2022-05-18 21:44:15 [info     ] [MDM] init: opened audio devices
2022-05-18 21:44:15 [debug    ] [MDM] init: starting pyaudio callback
Traceback (most recent call last):
  File "/mnt/disk1/home/pi/git/FreeDATA/tnc/main.py", line 137, in <module>
    modem = modem.RF()
  File "/mnt/disk1/home/pi/git/FreeDATA/tnc/modem.py", line 186, in __init__
    self.hamlib.open_rig(devicename=static.HAMLIB_DEVICE_NAME, deviceport=static.HAMLIB_DEVICE_PORT, hamlib_ptt_type=static.HAMLIB_PTT_TYPE, serialspeed=static.HAMLIB_SERIAL_SPEED, pttport=static.HAMLIB_PTT_PORT, data_bits=static.HAMLIB_DATA_BITS, stop_bits=static.HAMLIB_STOP_BITS, handshake=static.HAMLIB_HANDSHAKE, rigctld_ip = static.HAMLIB_RGICTLD_IP, rigctld_port = static.HAMLIB_RGICTLD_PORT)
AttributeError: module 'static' has no attribute 'HAMLIB_RGICTLD_IP'
2022-05-18 21:44:24 [warning  ] [SCK] Closing client socket    ip=127.0.0.1 port=45324

```

I've no idea what to do, FreeDATA has not been usable for quite a while now and my enthusiasm is somewhat dropping because of these issues affecting the builds since the file transfer was moved to the chat (that doesn't work due).

I hope it is usable again soon.

kronenpj, 2022-05-18:
This has been fixed. I'm looking to see if I can create a release without Simon. :)

pe1rrr, 2022-05-19:
> I've managed to release alpha.7, that will fix the hamlib typo. Please try it out when you have a chance.

:S 

```
```

        self.hamlib.open_rig(devicename=static.HAMLIB_DEVICE_NAME, deviceport=static.HAMLIB_DEVICE_PORT, hamlib_ptt_type=static.HAMLIB_PTT_TYPE, serialspeed=static.HAMLIB_SERIAL_SPEED, pttport=static.HAMLIB_PTT_PORT, data_bits=static.HAMLIB_DATA_BITS, stop_bits=static.HAMLIB_STOP_BITS, handshake=static.HAMLIB_HANDSHAKE, rigctld_ip = static.HAMLIB_RIGCTLD_IP, rigctld_port = static.HAMLIB_RGICTLD_PORT)

```

I think the typo is still there,, I am not 100% on git usage but as far as I know my checkout is correct:

Can you let me know how to check out this specific tag in case I have done it wrong.

DJ2LS, 2022-05-19:
Hi @pe1rrr , sorry for the long time you have to wait until this is fixed ( I'm busy with some personal stuff, which needs my priorities, actually ). But I'll. be back, soon! @kronenpj is a great addition, so I'm happy he can help you as well. 
We are actually working on some background stuff which is helping making development more reliable, so bug hunting is a little bit stuck, actually. However, this phase is ending, soon!

DJ2LS, 2022-05-19:
> :S
>
> ```
>
>         self.hamlib.open_rig(devicename=static.HAMLIB_DEVICE_NAME, deviceport=static.HAMLIB_DEVICE_PORT, hamlib_ptt_type=static.HAMLIB_PTT_TYPE, serialspeed=static.HAMLIB_SERIAL_SPEED, [...]

I fixed this, this morning as it was breaking the entire release.

DJ2LS, 2022-05-19:
Its a Raspberry Pi problem / ARM Problem. 

```
VM372 renderer_init:73 Unable to load preload script: /home/pi/FreeDATA/gui/preload-chat.js
VM372 renderer_init:73 Error: No native build was found for platform=linux arch=arm runtime=electron abi=103 uv=1 armv=default libc=glibc
    at Function.load.path (VM511 index.js:56:9)
    at load (VM511 index.js:20:30)
    at Object.<anonymous> (VM510 binding.js:1:198)
    at Object.<anonymous> (VM510 binding.js:3:3)
    at Module._compile (VM353 loader:1116:14)
    at Object.Module._extensions..js (VM353 loader:1169:10)
    at Module.load (VM353 loader:988:32)
    at Module._load (VM353 loader:829:12)
    at Function.c._load (VM369 asar_bundle:5:13343)
    at Function.i._load (VM372 renderer_init:33:356)
```

DJ2LS, 2022-05-19:
```
pi@dn2ls-pi:~/FreeDATA/gui $ npx electron-builder --armv7l
  • electron-builder  version=23.0.3 os=5.10.63-v7l+
  • loaded configuration  file=package.json ("build" field)
  • writing effective config  file=dist/builder-effective-config.yaml
  • rebuilding native dependencies  dependencies=leveldown@5.6.0 platform=linux arch=armv7l
```

I already know this problem. Its caused as a dependency `leveldown` of `pouchdb` package, which is needed for the chat database. 
This might need more time, I think.
However, there's only one fast option, which is bringing back file transfer to RaspberryPi:
We need to enable the old file transfer again for RaspberryPi :thinking:

DJ2LS, 2022-05-19:
@pe1rrr I enabled the "old" file transfer again, so you can use the application. Please keep in mind, we changed a lot of things. I did a test here, and it seems still to be working, but I'm not sure. Specially the compatibility between file transfer and chat may not working.

DJ2LS, 2022-05-19:
@pe1rrr I updated the old chat again, so its now in one sidebar. This is a work in progress and more a temporary solution, so you are able doing tests.

DJ2LS, 2022-05-20:
for your tests, please keep in mind theres a minor protocol issue, which fill be possibly fixed by https://github.com/DJ2LS/FreeDATA/pull/198

DJ2LS, 2022-05-27:
@pe1rrr can we close this issue?
Is the re-enabled file transfer working for you?

pe1rrr, 2022-05-27:
I havent had time to test lately sorry about that- had to rebuild my RPI system due to a disk failure. I'll get on it this weekend

pe1rrr, 2022-05-27:
I've managed to test one way so far, I am having issues with very poor GUI performance on one of the boxes and I dont know why but it MIGHT be related to USB HUB configuration here, it is quite a heavily loaded system, - it used to work fine but some wiring has changed so its probably that.

You may close this issue as the file dialog from the sending station's gui interacted fine with the headless daemon.py running on the 'slow box'.

Thanks very much for adding the feature back. When i get some more time to figure a new spare Pi to use as a dedicated box I will give you a heads up for some hardcore testing again.

By the way, the new GUI is looking slick and very nice. Keep up the great work- I am looking forward to talking to the TNC in KISS mode at some point so I can run it transparently on Linux.

73
Red

closed 2022-05-27 by kronenpj

DJ2LS, 2022-10-11:
Ah interesting. 
Did you run a ‚npm i‘ within gui folder?