DJ2LS/FreeDATA #190: chat cannot be deleted

dk5sm, 2022-05-15:
Clicking on "Delete chat" leads to a chat window reload, but has no further effect:

[image]

DJ2LS, 2022-05-17:
We also should add an option deleting single messages as additional feature.

DJ2LS, 2022-05-19:
@dk5sm Could you press `Ctrl+Shift+i `for opening the GUIs debugging console while the chat is openend? Then Try deleting the chat again and write the debugging output here. Thanks!

DJ2LS, 2022-05-19:
Further inspection has been showing, we need to catch errors in `preload-chat.js:515`

DJ2LS, 2022-05-27:
> @dk5sm Could you press `Ctrl+Shift+i `for opening the GUIs debugging console while the chat is openend? Then Try deleting the chat again and write the debugging output here. Thanks!

@dk5sm , could you check this again with the new error message which should be now displayed ?

DJ2LS, 2022-05-28:
Lines of interest:

```
app.asar\preload-chat.js:516 TypeError: Cannot convert undefined or null to object
    at Function.keys (<anonymous>)
    at update_chat (VM425 C:

```

DJ2LS, 2022-06-05:
```
Object
VM115 preload-chat.js:515 error with database parsing...
VM115 preload-chat.js:516 TypeError: Cannot convert undefined or null to object
    at Function.keys (<anonymous>)
    at update_chat (VM115 preload-chat.js:470:28)
    at VM115 preload-chat.js:77:13
    at Array.forEach (<anonymous>)
    at VM115 preload-chat.js:75:21
    at processTicksAndRejections (VM320 task_queues:96:5)
```

closed 2022-06-15 by DJ2LS