DJ2LS/FreeDATA #194: New PyInstaller (v5.1) breaks build

kronenpj, 2022-05-18:
The github action to build our release fails (on the soundmodem package) with PyInstaller 5.1.

**To Reproduce**
Steps to reproduce the behavior:
1. Change 'pyinstaller<5.1' to 'pyinstaller'
2. Re-run the github action.
3. Observe failure.

**Expected behavior**
Build succeeds.

**Logfiles**
One example.
https://github.com/DJ2LS/FreeDATA/runs/6497911314?check_suite_focus=true#step:4:100

Workaround committed to main: Limit pyinstaller to versions < 5.1.

DJ2LS, 2022-06-16:
test run with pyinstaller 5.1 and python 3.9.7 succeeded on Ubuntu 20.04 🤔

kronenpj, 2022-06-17:
Interesting, all I can say is that it was failing to build the executables when I was trying to release alpha4-7. I was able to trace it back to 5.1 but I fully believe the real problem has been fixed and 5.1 is working again.

DJ2LS, 2022-06-17:
We could test this carefully with the next release. On success: 👍 , if it fails: Fall back to older Vera hin which will be released as a hot fix

closed 2022-07-13 by DJ2LS