DJ2LS/FreeDATA #197: session disconnect ignored

DJ2LS, 2022-05-19:
session disconnect will be ignored.
Maybe we need to add this to callsign whitelist

DJ2LS, 2022-05-20:
Not sure if really a big. Occurred once. We might keep this in mind

DJ2LS, 2022-05-20:
When closing the TNC on IRS and starting it again while an ongoing transmission, and then sending a stop/disconnect on ISS, the frame is unknown.

kronenpj, 2022-06-01:
If the STOP frame (type #249) is received via the frame allow list then we would need to do additional processing to decide whether to act upon that frame. Otherwise any / every STOP frame received will cause a disconnection of a session. We don't want that. :)

kronenpj, 2022-06-24:
Looking at this a little deeper. If the STOP frame has the checksum for `mycallsign` the frame should be processed. Let me create a test case to see how the TNC behaves in a similar scenario.

DJ2LS, 2022-06-24:
I also had the case, that we got a signalling frame toggle 🤔 but this doesn't occur often. However, we need a fox for this as soon as it's clear what exactly happens

kronenpj, 2022-06-24:
The TNC will process any `STOP` frame addressed to that station regardless of ARQ connection status. The positive test verifies this and the negative verifies that the station ignores stop frames not addressed to it. Code is in PR #227.

closed 2022-06-24 by kronenpj