DJ2LS/FreeDATA #199: Binary files like pictures transfering with RF chat

computer-wizard, 2022-05-20:
It should be very nice if you can transfer binary files like pictures, etc. using RF-chat.
I mean that a copied binary file is transmitted when it is pasted/dropped  in the RF-chat window. 
At the remote RF-chat window an icon will appear with the name of the file or picture what can be
opened to see the contents when the download is completed.

computer-wizard, 2022-05-20:
Or in chase of a picture, directly the picture is displayed in the receiving remote RF chat window.

DJ2LS, 2022-05-20:
Hi, @computer-wizard sending pictures and other files via RF Chat is already possible. Just select a file by clicking the staple right of the text field. Then you can send a message + a file. Or are you talking about "drag and drop" ? 

The other feature with image and file preview is planned, but we are focusing on core stability first.

computer-wizard, 2022-05-20:
Yes, I was also talking about drag and drop.

DJ2LS, 2022-05-21:
Okay, I'd like to rename this issue to "drag and drop like behavior for chat" so its more clear. 
This needs a big rewrite of the actual data transmission part within chat, so might need some time until this is implemented. But I like to see this feature as well :smile:

DJ2LS, 2022-11-20:
> Or in chase of a picture, directly the picture is displayed in the receiving remote RF chat window.

@computer-wizard this should be possible in a really basic way with the next release. However, images should be really small because of low speed. Theres also not yet a nice modal for displaying the images. This can be done in future releases.

DJ2LS, 2023-10-25:
The new gui supports drag n drop, when dragging a file to the input field for selecting files.

DJ2LS, 2024-04-17:
@Mashintime can you have a look regarding to this? It's time for closing this if it's working or adjusting it 🙂

DJ2LS, 2024-09-17:
implemented, working for "new message area"

[image]

there is no additional information for keeping things simple.

closed 2024-09-21 by DJ2LS