DJ2LS/FreeDATA #200: improve TNCs info fields

DJ2LS, 2022-05-21:
we need to move from 

https://github.com/DJ2LS/FreeDATA/blob/fa7fe909a0f79764bf6a388ec4847d5e981b02b3/tnc/data_handler.py#L450

to

https://github.com/DJ2LS/FreeDATA/blob/fa7fe909a0f79764bf6a388ec4847d5e981b02b3/tnc/data_handler.py#L616

for the entire TNC

Also we should create an own function for handling this (following) part more simple and flexible :
```
jsondata = {"arq": "received", "uuid": uniqueid, "timestamp": timestamp,
            "mycallsign": str(mycallsign, 'utf-8'), "dxcallsign": str(static.DXCALLSIGN, 'utf-8'),
            "dxgrid": str(static.DXGRID, 'utf-8'), "data": base64_data}
json_data_out = json.dumps(jsondata)
structlog.get_logger("structlog").debug("[TNC] arq_data_received:", jsondata=jsondata)
sock.SOCKET_QUEUE.put(json_data_out)
```

When doing this, we need to update the GUI as well:

https://github.com/DJ2LS/FreeDATA/blob/fa7fe909a0f79764bf6a388ec4847d5e981b02b3/gui/sock.js#L207

move to something like done here:

https://github.com/DJ2LS/FreeDATA/blob/fa7fe909a0f79764bf6a388ec4847d5e981b02b3/gui/sock.js#L240

kronenpj, 2022-05-30:
It'll look a lot like this, pending additions needed for current uses.:
```python
    def send_data_to_socket_queue(self, /, **kwargs):
        jsondata = kwargs
        self.log.debug("[TNC] send_data_to_socket_queue:", jsondata=jsondata)
        json_data_out = json.dumps(jsondata)
        sock.SOCKET_QUEUE.put(json_data_out)
```

DJ2LS, 2022-05-30:
That's a nice idea! I like it!

The old function can  be updated/redcued as well, I think.
https://github.com/DJ2LS/FreeDATA/blob/8b9b54591fbed934354ba89c88a8e9beefcb6d14/tnc/sock.py#L379

kronenpj, 2022-05-30:
> The old function can be updated/redcued as well, I think.
> 
> https://github.com/DJ2LS/FreeDATA/blob/8b9b54591fbed934354ba89c88a8e9beefcb6d14/tnc/sock.py#L379

That's a bit of a challenge and it wouldn't reduce the size much. The simple one doesn't support checking for duplicate messages, for example, since it immediately sends data to the queue. We could split it into two methods, but they'd be very small. and wouldn't make a lot of sense. 

If we wanted to introduce another `static` variable, we could incorporate it as an optional duplicate check.

DJ2LS, 2022-05-31:
The idea with reducing the information within `send_tnc_state` is, avoiding confusion for developers who want to control the tnc. But while having a second look at `send_tnc_state` I agree with you. I thought, theres still more ARQ information within this function from earlier days which doesn't make sense anymore with the new information system - sorry for the confusion!

closed 2022-06-15 by DJ2LS