DJ2LS/FreeDATA #202: Implement KISS mode interface

kronenpj, 2022-05-27:
**Is your feature request related to a problem? Please describe.**
KISS mode is used by a large number of applications and is a standardized interface.

**Describe the solution you'd like**
Implement a cross-platform interface to the TNC that implements most, if not all, of the KISS protocol.

**Describe alternatives you've considered**
Another feature request would be to implement the AGWPE interface.

**Additional context**
Requested by @pe1rrr in issue #181. This will be challenging given the size of the control frames and the differences in the way `codec2` behaves vs. AX.25 or other purely packet-based protocols typically used with KISS mode.

DJ2LS, 2022-05-28:
Combined with the arq session it might be possible creating a working solution. 
But I suggest we are going to make the tnc itself stable and reliable before working on this ( specially ARQ session needs to be optimised ).

On the other side there are already existing and working applications which are made explicitly for packet / KISS / ax25. 
So it would be interesting why FreeDATA needs this or what's the specific use case 🤔

DJ2LS, 2022-05-28:
https://github.com/xssfox/freedv-tnc

There's also an existing solution for this.

kronenpj, 2022-05-28:
This was meant more to capture the indirect request from PE1RRR's closing remark on issue #181 than anything else.

DJ2LS, 2022-05-28:
Yes, no problem! Understood it the same. However this might be a topic we have to discuss about, somewhen.

closed 2022-11-20 by DJ2LS