DJ2LS/FreeDATA #207: Final data packet in burst does not retry on NACK.

kronenpj, 2022-06-05:
**Describe the bug**
The chat / file transfer OTA protocol doesn't resend the last frame if a NACK is received. Instead the routine drops into the transmission failed state and closes the connection.

**To Reproduce**
Steps to reproduce the behavior:
1. Craft a data frame that is larger than one frame (>~560 bytes)
2. Send over unreliable or delayed channel like FIFOs.
3. Wait for file transfer to complete / run new `test_data_channel` test.
4. Observe debug output / notice large chat or file transfer failed.

**Expected behavior**
Final frame is properly transmitted when NACK is received.

**Logfiles**
```
2022-06-05 17:56:23 [info     ] [TNC] ARQ | DATA | TX | [AA2BB-0]>> <<[ZZ9YY-0] attempt=1/5
2022-06-05 17:56:23 [info     ] S1 TX:                         TX=225
2022-06-05 17:56:26 [info     ] [TNC] ARQ | DATA | TX | [AA2BB-0]>> <<[ZZ9YY-0] attempt=2/5
2022-06-05 17:56:26 [info     ] S1 TX:                         TX=225
2022-06-05 17:56:26 [info     ] S2 RX:                         RX=225
2022-06-05 17:56:26 [info     ] [TNC] ARQ | DATA | RX | [ZZ9YY-0]>> <<[AA2BB-0] bandwidth=wide
2022-06-05 17:56:26 [info     ] S2 TX:                         TX=228
2022-06-05 17:56:26 [info     ] [TNC] ARQ | DATA | RX | [ZZ9YY-0]>>|<<[AA2BB-0] bandwidth=wide snr=9.0
2022-06-05 17:56:27 [warning  ] [TNC] Frame timeout            attempt=0 max_attempts=50 speed_level=2
2022-06-05 17:56:27 [info     ] S2 TX:                         TX=64
2022-06-05 17:56:27 [info     ] S1 RX:                         RX=228
2022-06-05 17:56:27 [info     ] [TNC] ARQ | DATA | TX | [AA2BB-0]>>|<<[ZZ9YY-0] snr=10.1
2022-06-05 17:56:27 [info     ] [TNC] | TX | DATACHANNEL       Bytes=1530 mode=10
2022-06-05 17:56:27 [info     ] [TNC] ARQ | TX | FRAMES        fpb=1 mode=10 retry=0
2022-06-05 17:56:27 [info     ] S1 TX:                         TX=10
2022-06-05 17:56:27 [info     ] S2 RX:                         RX=225
2022-06-05 17:56:27 [info     ] [TNC] ARQ | DATA | RX | [ZZ9YY-0]>> <<[AA2BB-0] bandwidth=wide
2022-06-05 17:56:27 [info     ] S2 TX:                         TX=228
2022-06-05 17:56:27 [info     ] S1 RX:                         RX=64
2022-06-05 17:56:27 [info     ] [TNC] ARQ | DATA | RX | [ZZ9YY-0]>>|<<[AA2BB-0] bandwidth=wide snr=9.0
2022-06-05 17:56:27 [info     ] [TNC] ARQ | TX | FRAMES        fpb=1 mode=10 retry=1
2022-06-05 17:56:27 [info     ] S1 TX:                         TX=10
2022-06-05 17:56:27 [info     ] S2 RX:                         RX=10
2022-06-05 17:56:27 [info     ] [TNC] ARQ | RX | SENDING ACK
2022-06-05 17:56:27 [info     ] S2 TX:                         TX=60
2022-06-05 17:56:27 [info     ] S1 RX:                         RX=228
2022-06-05 17:56:27 [info     ] [TNC] ARQ | DATA | TX | [AA2BB-0]>>|<<[ZZ9YY-0] snr=10.1
2022-06-05 17:56:28 [warning  ] [TNC] Frame timeout            attempt=0 max_attempts=50 speed_level=2
2022-06-05 17:56:28 [info     ] S2 TX:                         TX=64
2022-06-05 17:56:28 [info     ] S1 RX:                         RX=60
2022-06-05 17:56:28 [info     ] [TNC] ARQ | TX | FRAMES        fpb=1 mode=10 retry=0
2022-06-05 17:56:28 [info     ] S1 TX:                         TX=10
2022-06-05 17:56:28 [info     ] S2 RX:                         RX=10
2022-06-05 17:56:28 [info     ] [TNC] ARQ | RX | Frame already received - sending ACK again
2022-06-05 17:56:28 [info     ] [TNC] ARQ | RX | SENDING ACK
2022-06-05 17:56:28 [info     ] S2 TX:                         TX=60

------------ BURST NACK == 64 ------------
2022-06-05 17:56:28 [info     ] S1 RX:                         RX=64
2022-06-05 17:56:29 [warning  ] [TNC] Frame timeout            attempt=0 max_attempts=50 speed_level=1
2022-06-05 17:56:29 [info     ] S2 TX:                         TX=64
2022-06-05 17:56:29 [info     ] S1 RX:                         RX=60
2022-06-05 17:56:29 [info     ] [TNC] ARQ | TX | TRANSMISSION FAILED OR TIME OUT! overflows=[0, 0, 0, 0, 0]
2022-06-05 17:56:29 [warning  ] [TNC] Stopping transmission!
2022-06-05 17:56:29 [info     ] S1 TX:                         TX=249
------------ Frame should have been re-sent, instead ARQ_STOP == 249 ------------

2022-06-05 17:56:29 [info     ] S2 RX:                         RX=10
2022-06-05 17:56:29 [info     ] [TNC] ARQ | RX | DATA FRAME SUCESSFULLY RECEIVED
2022-06-05 17:56:29 [info     ] [TNC] ARQ | RX | SENDING DATA FRAME ACK crc=67eb76ea snr=8.9
...
```

**Screenshots**
N/A

**Desktop (please complete the following information):**
N/A - occurs independent of platform.

DJ2LS, 2022-06-05:
Does this only occur if we have a data frame NACK ( not a burst NACK )?
If we have a data frame NACK, the crc check for the entire data frame failed. So we have to restart the entire transmission 🤔

kronenpj, 2022-06-05:
This occurs with a `BURST_NACK` (frame type 64). `FRAME_NACK` is frame type 63 and doesn't appear above. Looking at the code receipt of a `FRAME_NACK` causes the session to be terminated locally **without notifying the far side of the session closure**. This is probably another bug :(.

DJ2LS, 2022-06-05:
Ah, combined with the PR which fixes this, I understood the problem 🙂
So this is the case if a frame ACK is getting lost so we are going to request it again. 

The case of a frame NACK and therefore closing the session is necessary because of corrupted data. Here we could think about an improvement - resending the entire data for example

kronenpj, 2022-06-05:
The chat and file transfer from chat work as before. Is a  long message split into chunks before it gets to `arq_transmit`? In any event, my tests show the fix does not affect current functionality. It also works with an unmodified `data_handler`.

DJ2LS, 2022-06-05:
No, not exactly. Splitting data into chunk was the "old" behaviour some version ago. Meanwhile all of the data will be hold in a buffer, we are accessing the needed position in accordance to codec2 frame payload and transmission progress. This happens all in `arq_transmit`.

kronenpj, 2022-06-05:
Then the test I ran (pasting the big block of text that I used in `test_data_handler`) should provide a roughly equivalent test. The only thing left would be to test it on the air.

DJ2LS, 2022-06-05:
Okay, I could test this in about two days

kronenpj, 2022-06-05:
Sounds good. I don't believe this has been encountered "in real life" but it caused me a day or two during test case development and after examining the code it should have been a problem.

DJ2LS, 2022-06-05:
I had this problem several times, specially under weak hf circumstances, which is stressing the protocol much. So this improvement helps for sure!

closed 2022-06-21 by DJ2LS