DJ2LS/FreeDATA #262: Transceiver serial port not listed in Hamlib advanced settings (macOS)

arthar360, 2022-10-11:
**Describe the bug**
Advanced hamlib settings does not show transceiver in Port

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Advanced hamlib settings'
2. Click on 'Port' dropdown
3. The transceiver serial port is not listed

**Expected behavior**
The transceiver serial port should be shown in the dropdown

**Logfiles**
[freeDataLogs.zip](https://github.com/DJ2LS/FreeDATA/files/9758730/freeDataLogs.zip)

**Screenshots**
FreeData shows only one entry
[image]

Screenshot from wsjtx, shows all entries
[image]

**Desktop (please complete the following information):**
 - OS: macOS
 - Version 12.5.1 (21G83)

DJ2LS, 2022-10-11:
@arthar360 thanks for the issue. Does this also appear on normal settings? Advanced settings will be dropped with next version as a complete switch to rigctld will be forced

arthar360, 2022-10-11:
In the normal settings, the dropdown for Radio dont have any options

[image]

The trx is Icom IC-7100 which has a built-in sound card. The dropdown for Audio does not show any builtin/trx sources

DJ2LS, 2022-10-11:
Errors in Logfile

`[2022-10-11 18:47:08.809] [info]  (system information) FOUND SETTTING [rx_buffer_size]: 16
[2022-10-11 18:47:08.810] [info]  (system information) ------------------------------------------  
[2022-10-11 18:47:09.014] [info]  (main)               Starting freedata-daemon binary
[2022-10-11 18:47:09.772] [info]  (freedata-daemon)    dyld[1770]: Library not loaded: '/Users/runner/hostedtoolcache/Python/3.9.13/x64/lib/libpython3.9.dylib'
  Referenced from: '/Applications/FreeDATA.app/Contents/Resources/tnc/freedata-daemon'
  Reason: tried: '/Users/runner/hostedtoolcache/Python/3.9.13/x64/lib/libpython3.9.dylib' (no such file), '/usr/local/lib/libpython3.9.dylib' (no such file), '/usr/lib/libpython3.9.dylib' (no such file)

[2022-10-11 18:47:09.791] [warn]  (freedata-daemon)    daemonProcess exited with code null
[2022-10-11 18:47:10.567] [info]  (updater)            Checking for update
[2022-10-11 18:47:10.568] [info]  (main)               checking for update
[2022-10-11 18:47:11.261] [error] (updater)            Error: Error: No published versions on GitHub
    at Object.newError (/Applications/FreeDATA.app/Contents/Resources/app.asar/node_modules/builder-util-runtime/out/index.js:47:19)
    at GitHubProvider.getLatestVersion (/Applications/FreeDATA.app/Contents/Resources/app.asar/node_modules/electron-updater/out/providers/GitHubProvider.js:94:42)
    at process.processTicksAndRejections (node:internal/process/task_queues:96:5)
    at async MacUpdater.getUpdateInfoAndProvider (/Applications/FreeDATA.app/Contents/Resources/app.asar/node_modules/electron-updater/out/AppUpdater.js:305:19)
    at async MacUpdater.doCheckForUpdates (/Applications/FreeDATA.app/Contents/Resources/app.asar/node_modules/electron-updater/out/AppUpdater.js:319:24)
[2022-10-11 18:47:11.262] [info]  (main)               update error
[2022-10-11 18:47:11.262] [error] (main)               AUTO UPDATER : Error: No published versions on GitHub`

arthar360, 2022-10-11:
The `runner` user is not present locally. It seems like the path is hardcoded by Github Actions and the dependency is missed when it was built

DJ2LS, 2022-10-11:
Good point. We need to check if this is a possible issue caused by nuitka as this is new to me 🤔

DJ2LS, 2022-10-11:
https://github.com/Nuitka/Nuitka/issues/1260

arthar360, 2022-10-11:
May be something like this is a partial solution

`~/.github/workflows/build_multiplatform.yml` line `305`

```
          if [{{!startsWith(matrix.os, 'macos')}}]
          then
            python3 -m nuitka --enable-plugin=numpy --assume-yes-for-downloads --standalone --macos-create-app-bundle daemon.py
            python3 -m nuitka --enable-plugin=numpy --assume-yes-for-downloads --standalone --macos-create-app-bundle main.py
          else
            python3 -m nuitka --enable-plugin=numpy --assume-yes-for-downloads --standalone daemon.py
            python3 -m nuitka --enable-plugin=numpy --assume-yes-for-downloads --standalone main.py
          fi
```

On the gitlab runner do we already have venv from homebrew?

DJ2LS, 2022-10-11:
Yes, this could be a possible solution. I had similar thoughts.We could also use homebrew as suggested in nuitka issue. Von meinem iPhone gesendet

DJ2LS, 2022-10-13:
Latest test commit seems to fix this problem, but causes problems with macOS security system. Moving back to pyinstaller for macOS as this has been working last time.

DJ2LS, 2022-10-16:
@arthar360 testing is actually not possible, because of macOS security scans. A developer ID is needed, which costs 90€.

DJ2LS, 2022-11-20:
@arthar360 I have problems with Apple and can't order an Developer ID. So we have to wait until this is fixed

closed 2023-01-21 by DJ2LS