DJ2LS/FreeDATA #285: TNC API continues to process new connect commands while already in a connecting loop

pe1rrr, 2022-11-17:
Suggestion: Block or ignore further connect commands from API until either timeout or a DISC is issued.
This is to prevent a loop within a loop which currently occurs and causes a loss of state awareness of the TNC for anything connected to the API. This can be an issue for node users issuing attempting to abort a connection or making a mistake.

Reproduce:

ATTACH TNC
C GM8BPQ-2
<enter> observe TNC sending packets, wait an undefined time
C GM8BPQ-2
<enter> observe overlapping request interjecting with previous loop.
Manually abort connection.

Expected behavior:

Instead of processing the extraneous command, the TNC should instead return an error or force first connection to abort and disconnect everything.

If a connection is established, send status to API-> CONNECTED
API can then update connected driver with connect status.

DJ2LS, 2022-11-18:
That's exactly the kind of feedback I need, @pe1rrr ! Thanks, I will implement this asap.

Suggestion:
When running connect command with same callsigns and already connected --> error 

When running connect command with different callsigns and already connected 
--> disconnect and connect to new callsign 

Or would it be better, not implementing some logic and just dropping an error information?

DJ2LS, 2022-11-18:
Okay, I implemented it as following:

- if not connected, just connect
- if already connected with same callsign - drop error message
- if already connected with different callsign - disconnect then connect to new callsign

pe1rrr, 2022-11-19:
Not sure if the last is good to be different from the standard which is to help BBS software react predictably.

 If a BBS fwd script messed up a session connect, abort and disconnect - let the TNC user/BBS script behave predictably according to norms defined with protocols such as the AGWPE protocol.

DJ2LS, 2022-11-19:
Okay, so it's no problem changing this. Regarding to AGWPE I don't have experience, so I need your advice here, @pe1rrr .

So I'm going to keep the following:

- if not connected, connect
- if already connected  - drop error message

I'm going to remove this:
- if already connected with different callsign - disconnect then connect to new callsign

At which point ( except session lost ) should we automatically disconnect? Or is this a behavior, we want to avoid and disconnecting should only happen when session is getting lost or we are forcing a disconnect manually?

pe1rrr, 2022-11-19:
I think if the user/script puts in a command that isn’t disconnect/abort during a connect attempt, the TNC should be reset and disconnected from the client immediately to free the TNC up for the next try.

pe1rrr, 2022-11-19:
> I'm going to remove this:
> 
> * if already connected with different callsign - disconnect then connect to new callsign
> 

Perfect

DJ2LS, 2022-11-20:
> I think if the user/script puts in a command that isn’t disconnect/abort during a connect attempt, the TNC should be reset and disconnected from the client immediately to free the TNC up for the next try.

Okay, from my side I'd like to start with 

- if not connected, connect
- if already connected - drop error message

at first, so we don't have any additional behavior. If any other command will be passed to the TNC while connecting or connected and it disconnects then, it could cause confusion as well in my opinion. 

So would it be okay for you, if we start with the simple approach first, and then see how it is working, then deciding if we have to adjust this behavior?

pe1rrr, 2022-11-20:
Yes sounds like a good idea. Go for it.

DJ2LS, 2022-11-20:
@pe1rrr Can you confirm, this is actually working for you when running from source?

DJ2LS, 2022-11-20:
this should work now (hopefully), we need to to further tests.

@pe1rrr please feel free reopening this, if problem still exists.

closed 2022-11-20 by DJ2LS