DJ2LS/FreeDATA #287: rigctld crashes because of too much connection attempts

DJ2LS, 2022-11-18:
Opened rig model 3085, 'IC-705'
rigctl_parse: nothing to scan#1? retcode=-1
Peer lookup error: Temporary failure in name resolutionaccept: Network error 24: Too many open files
rigctl_parse: rig not open...trying to reopen
rigctl_parse: RIG_EIO?
handle_socket: i/o error

DJ2LS, 2022-11-18:
fixed

pe1rrr, 2022-11-18:
Daemon.py on Latest pull of ls-beta: 

```
Exception in thread HAMLIB_THREAD:
Traceback (most recent call last):
  File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.9/threading.py", line 892, in run
    self._target(*self._args, **self._kwargs)
  File "/mnt/disk1/home/pi/FreeDATA/tnc/modem.py", line 837, in update_rig_data
    static.HAMLIB_STATUS = self.hamlib.get_status()
AttributeError: 'radio' object has no attribute 'get_status'

```

Is related?

DJ2LS, 2022-11-18:
Fixed with next release.Rigctld used with headless tnc is not affected.Von meinem iPhone gesendet

DJ2LS, 2022-11-18:
But I’m wondering why your error occurs. 🤔 will have a look!

DJ2LS, 2022-11-18:
> Daemon.py on Latest pull of ls-beta:
>
> ```
> Exception in thread HAMLIB_THREAD:
> Traceback (most recent call last):
>   File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner [...]

This is definitely part of the code. So it could be possible, git didn't get all files?
I Introduced this so we have an rigctld connection state.

DJ2LS, 2022-11-20:
@pe1rrr fixed by https://github.com/DJ2LS/FreeDATA/pull/280/commits/a9019442687b3b81b445859fabe3a0adaffe4d57

closed 2022-11-20 by DJ2LS