DJ2LS/FreeDATA #288: Daemon.py retry count set via API (by count not a timer)

pe1rrr, 2022-11-19:
For doing automated mail forwarding the client using the TNC would need to tell it how many retries it should do in order to avoid jamming up the QRG unnecessarily.

Current behavior:

- TNC connect attempts continue until some global timeout check parameter is met.

Desired behavior:

- Client tells TNC how many retries it wants to do, TNC disconnects after defined retries is exceeded. Similar to AGWPE protocol such as that incorporated in UZ7HO sound modem and QtSoundModem.

This will reduce QRM as there are excessive attempts by default which is unwanted behavior when there is limited time during propagation paths when the mail forwarding could have already switched to another station that is reachable.

DJ2LS, 2022-11-19:
Agree! I could add this to the API when connecting or sending data. Then this parameter is flexible and can be adjusted easily.

DJ2LS, 2022-11-19:
Started working on this.

pe1rrr, 2022-11-20:
> Agree! I could add this to the API when connecting or sending data. Then this parameter is flexible and can be adjusted easily.
> 
> Is there a special reason for naming "Daemon.py" here? There could be also an option for setting this as startup parameter, but I would prefer doing this via API and network command.

Just makes it abundantly clear that I am not using the GUI for FreeDATA

pe1rrr, 2022-11-20:
Here is AGWPE spec https://www.on7lds.net/42/sites/default/files/AGWPEAPI.HTM#_Toc500723771

pe1rrr, 2022-11-20:
I think I am wrong with regards to retry count set over AGWPE API, may be a config file thing on startup as you mentioned.

DJ2LS, 2022-11-20:
> Just makes it abundantly clear that I am not using the GUI for FreeDATA

Understood 👍

> Here is AGWPE spec https://www.on7lds.net/42/sites/default/files/AGWPEAPI.HTM#_Toc500723771

Thanks I didn't find this yesterday evening, will have a closer look. 

> I think I am wrong with regards to retry count set over AGWPE API, may be a config file thing on startup as you mentioned.

I suggest let's implement this to the API as a start. This would be the most flexible way of using this. If it's a startup parameter, we would increase the already long list of parameters and making startup more complicated. If this is part of the API when connecting to a station, it could be set flexible. Say we have good channel conditions for one specific station, then use a low number of connection attempts. When then connecting to a station under weak channel conditions, we could increase the number of attempts for increasing chance of a connection.

DJ2LS, 2022-11-20:
Updated Wiki
https://wiki.freedata.app/en/development/network/network-tnc/connect
https://wiki.freedata.app/en/development/network/network-tnc/send_raw

DJ2LS, 2022-11-20:
@pe1rrr this can be used now via API, please check wiki (links provided above )for command details. If there are problems, please feel free reopening this issue so we can fix them!

closed 2022-11-20 by DJ2LS