DJ2LS/FreeDATA #289: Daemon.py Extraneous data transmitted to RF while unattended

pe1rrr, 2022-11-19:
This issue differs slightly depending on whether this is the first boot of the TNC.

After making a connection attempt to a dummy station, the TNC will reach the timeout and abort, it returns a Connection Failed message.

At this point, disconnect the client from the TNC and leave the TNC alone and do nothing for at least a minute. 

The TNC eventually pushes extra data out on to RF without any interaction. 

This behavior causes the rig to transmit unexpectedly. 

If attempting to do a connect command a second time, proceed as normal, then force an abort from the client (detach).

Attempting to send commands (i.e attach to the TNC) fails until the after the long gap of waiting and the anomalous extra data is sent out over RF.

The subsequent extra data burst marks the moment that the TNC accepts commands again.

DJ2LS, 2022-11-19:
Thanks for the detailed report! I will try to fix this! It might be possible I also had this issue as well while testing.

DJ2LS, 2022-11-20:
one issue found: arq_cleanup not working - error message "`already connected to station: ... `". Seems to me, `static.ARQ_SESSION_STATE` is not released correctly

DJ2LS, 2022-11-20:
Important note: 
If the connection fails, the tnc is sending a "disconnect burst" which is needed in case we already reached our destination, but we didn't get its response. By this I want to avoid, the other station is staying in connected state until timeout.

pe1rrr, 2022-11-20:
That’s normal, by then the remote station should be configured for a max of say 15 retries for HF anyways, so if it doesn’t hear the burst it will timeout on its own, this is similar to how AX25 works.

DJ2LS, 2022-11-20:
Okay, this behavior is already implemented. Theres a time based timeout for case being stuck and a retry based one, which is set to 15 retries.

DJ2LS, 2022-11-20:
@pe1rrr Can you confirm, this is actually working for you when running from source?

pe1rrr, 2022-11-20:
The extra data burst is still occurring even after sending the disconnect burst, same as original issue post.
Sounds like its got two disconnect bursts, one is immediate which should be sufficient for the remote to QRT, and then the other one which seems based on a timer which is the superfluous one.

pe1rrr, 2022-11-20:
```

Normal first burst:
2022-11-20 15:06:45 [debug    ] [TNC] send_data_to_socket_queue: jsondata={'freedata': 'tnc-message', 'arq': 'session', 'status': 'failed', 'reason': 'timeout'}
2022-11-20 15:06:45 [debug    ] [TNC] TX DISCONNECT...
2022-11-20 15:06:45 [info     ] [TNC] SESSION [PE1RRR-7]<<X>>[TEST-0] state=disconnecting
2022-11-20 15:06:45 [debug    ] [TNC] send_data_to_socket_queue: jsondata={'freedata': 'tnc-message', 'arq': 'session', 'status': 'close'}
2022-11-20 15:06:45 [debug    ] [TNC] enqueue_frame_for_tx     c2_mode=sig0
2022-11-20 15:06:45 [debug    ] [MDM] worker_transmit          mode=14
2022-11-20 15:06:45 [debug    ] [MDM] transmit                 mode=14
2022-11-20 15:06:45 [debug    ] [MDM] TRANSMIT                 mode=14 payload=14
2022-11-20 15:06:48 [debug    ] [MDM] ON AIR TIME              time=3.357998847961426
2022-11-20 15:06:48 [debug    ] [TNC] arq_cleanup

Anomalous/superfluous burst:

2022-11-20 15:07:14 [debug    ] [SCK] CMD                      command={'type': 'arq', 'command': 'disconnect'}
2022-11-20 15:07:14 [debug    ] [TNC] TX DISCONNECT...
2022-11-20 15:07:14 [info     ] [TNC] SESSION [PE1RRR-7]<<X>>[TEST-0] state=disconnecting
2022-11-20 15:07:14 [debug    ] [TNC] send_data_to_socket_queue: jsondata={'freedata': 'tnc-message', 'arq': 'session', 'status': 'close'}
2022-11-20 15:07:14 [debug    ] [TNC] enqueue_frame_for_tx     c2_mode=sig0
2022-11-20 15:07:14 [debug    ] [MDM] worker_transmit          mode=14
2022-11-20 15:07:14 [debug    ] [MDM] transmit                 mode=14
2022-11-20 15:07:14 [debug    ] [MDM] TRANSMIT                 mode=14 payload=14
2022-11-20 15:07:17 [debug    ] [MDM] ON AIR TIME              time=3.350687265396118
2022-11-20 15:07:17 [debug    ] [TNC] arq_cleanup
```

pe1rrr, 2022-11-20:
I forgot to mention, the above log is from aborting a connection/disconnecting the TNC.

DJ2LS, 2022-11-20:
okay, I possibly found another odd behavior. closing the session has been called, twice. I pushed a commit wich removed the double call. However, it's called directly after closing the session. So all about within 10seconds.  I hope we're coming closer to the issue. @pe1rrr could you have another look? Thanks!

pe1rrr, 2022-11-20:
Can confirm:

```

2022-11-20 15:43:28 [debug    ] [SCK] CMD                      command={'type': 'arq', 'command': 'disconnect'}
2022-11-20 15:43:28 [debug    ] [TNC] send_data_to_socket_queue: jsondata={'freedata': 'tnc-message', 'arq': 'session', 'status': 'failed', 'reason': 'timeout'}
2022-11-20 15:43:28 [debug    ] [TNC] TX DISCONNECT...
2022-11-20 15:43:28 [info     ] [TNC] SESSION [PE1RRR-7]<<X>>[GM8BPQ-2] state=disconnecting
2022-11-20 15:43:28 [debug    ] [TNC] send_data_to_socket_queue: jsondata={'freedata': 'tnc-message', 'arq': 'session', 'status': 'close'}
2022-11-20 15:43:28 [debug    ] [TNC] enqueue_frame_for_tx     c2_mode=sig0
2022-11-20 15:43:28 [debug    ] [MDM] worker_transmit          mode=14
2022-11-20 15:43:28 [debug    ] [MDM] transmit                 mode=14
2022-11-20 15:43:28 [debug    ] [MDM] TRANSMIT                 mode=14 payload=14
2022-11-20 15:43:32 [debug    ] [MDM] ON AIR TIME              time=3.9266810417175293
2022-11-20 15:43:32 [debug    ] [TNC] arq_cleanup

2022-11-20 15:43:59 [debug    ] [SCK] CMD                      command={'type': 'arq', 'command': 'disconnect'}
2022-11-20 15:43:59 [debug    ] [TNC] TX DISCONNECT...
2022-11-20 15:43:59 [info     ] [TNC] SESSION [PE1RRR-7]<<X>>[GM8BPQ-2] state=disconnecting
2022-11-20 15:43:59 [debug    ] [TNC] send_data_to_socket_queue: jsondata={'freedata': 'tnc-message', 'arq': 'session', 'status': 'close'}
2022-11-20 15:43:59 [debug    ] [TNC] enqueue_frame_for_tx     c2_mode=sig0
2022-11-20 15:43:59 [debug    ] [MDM] worker_transmit          mode=14
2022-11-20 15:43:59 [debug    ] [MDM] transmit                 mode=14
2022-11-20 15:43:59 [debug    ] [MDM] TRANSMIT                 mode=14 payload=14
2022-11-20 15:44:03 [debug    ] [MDM] ON AIR TIME              time=3.9182686805725098
2022-11-20 15:44:03 [debug    ] [TNC] arq_cleanup

```

DJ2LS, 2022-11-20:
Ah, maybe I'm coming closer to the problem 🤔 
after being disconnected because of a timeout `2022-11-20 15:43:32 [debug    ] [TNC] arq_cleanup` the tnc gets another disconnect command in tis case. But it looks like disconnect `2022-11-20 15:43:59 [debug    ] [SCK] CMD                      command={'type': 'arq', 'command': 'disconnect'}` is forced from network, not from protocol. 
--> If the connection can't be established, the connection closes automatically- no additional session close command is needed in this case.

DJ2LS, 2022-11-20:
We could do the following for avoiding additional executed commands:
- if already disconnecting: "don't process" additional close commands
- if already disconnected: "don't process" close commands

pe1rrr, 2022-11-20:
Sounds like it should work 👍

DJ2LS, 2022-11-20:
okay, @pe1rrr I added this check. Lets see if this is running better

pe1rrr, 2022-11-20:
```
2022-11-20 17:22:06 [debug    ] [MDM] ON AIR TIME              time=0.7300717830657959
2022-11-20 17:22:07 [debug    ] [SCK] CMD                      command={'type': 'arq', 'command': 'disconnect'}
2022-11-20 17:22:07 [debug    ] [TNC] send_data_to_socket_queue: jsondata={'freedata': 'tnc-message', 'arq': 'session', 'status': 'failed', 'reason': 'timeout'}
2022-11-20 17:22:07 [debug    ] [TNC] TX DISCONNECT...
2022-11-20 17:22:07 [info     ] [TNC] SESSION [PE1RRR-7]<<X>>[GM8BPQ-2] state=disconnecting
2022-11-20 17:22:07 [debug    ] [TNC] send_data_to_socket_queue: jsondata={'freedata': 'tnc-message', 'arq': 'session', 'status': 'close'}
2022-11-20 17:22:07 [debug    ] [TNC] enqueue_frame_for_tx     c2_mode=sig0
2022-11-20 17:22:07 [debug    ] [MDM] worker_transmit          mode=14
2022-11-20 17:22:07 [debug    ] [MDM] transmit                 mode=14
2022-11-20 17:22:07 [debug    ] [MDM] TRANSMIT                 mode=14 payload=14
2022-11-20 17:22:11 [debug    ] [MDM] ON AIR TIME              time=3.926328420639038
2022-11-20 17:22:11 [debug    ] [TNC] arq_cleanup

2022-11-20 17:22:39 [debug    ] [SCK] CMD                      command={'type': 'arq', 'command': 'disconnect'}
2022-11-20 17:22:39 [warning  ] [SCK] Disconnect command not possible command={'type': 'arq', 'command': 'disconnect'} state=disconnected
```

This is much better. Ok the network socket takes a few more seconds to die, but that is a lot better.

pe1rrr, 2022-11-20:
Bug: There is no longer an error when the connection retries exceed the maximum, so it will just stop TXing and sit waiting.

```
2022-11-20 17:25:32 [info     ] [TNC] SESSION [PE1RRR-7]>>?<<[GM8BPQ-2] a=14 state=connecting
2022-11-20 17:25:32 [debug    ] [TNC] enqueue_frame_for_tx     c2_mode=sig0
2022-11-20 17:25:32 [debug    ] [MDM] worker_transmit          mode=14
2022-11-20 17:25:32 [debug    ] [MDM] transmit                 mode=14
2022-11-20 17:25:32 [debug    ] [MDM] TRANSMIT                 mode=14 payload=14
2022-11-20 17:25:33 [debug    ] [MDM] ON AIR TIME              time=0.7358319759368896
2022-11-20 17:25:36 [info     ] [TNC] SESSION [PE1RRR-7]>>?<<[GM8BPQ-2] a=15 state=connecting
2022-11-20 17:25:36 [debug    ] [TNC] enqueue_frame_for_tx     c2_mode=sig0
2022-11-20 17:25:36 [debug    ] [MDM] worker_transmit          mode=14
2022-11-20 17:25:36 [debug    ] [MDM] transmit                 mode=14
2022-11-20 17:25:36 [debug    ] [MDM] TRANSMIT                 mode=14 payload=14
2022-11-20 17:25:37 [debug    ] [MDM] ON AIR TIME              time=0.7396197319030762
2022-11-20 17:25:40 [info     ] [TNC] SESSION [PE1RRR-7]<<X>>[GM8BPQ-2] state=disconnecting
2022-11-20 17:25:40 [debug    ] [TNC] send_data_to_socket_queue: jsondata={'freedata': 'tnc-message', 'arq': 'session', 'status': 'close'}
2022-11-20 17:25:40 [debug    ] [TNC] enqueue_frame_for_tx     c2_mode=sig0
2022-11-20 17:25:40 [debug    ] [MDM] worker_transmit          mode=14
2022-11-20 17:25:40 [debug    ] [MDM] transmit                 mode=14
2022-11-20 17:25:40 [debug    ] [MDM] TRANSMIT                 mode=14 payload=14
2022-11-20 17:25:44 [debug    ] [MDM] ON AIR TIME              time=3.999711036682129
2022-11-20 17:25:44 [debug    ] [TNC] arq_cleanup
2022-11-20 17:25:44 [debug    ] [TNC] send_data_to_socket_queue: jsondata={'freedata': 'tnc-message', 'arq': 'session', 'status': 'failed', 'reason': 'timeout'}

```

DJ2LS, 2022-11-20:
Warning message has been added if session failed

DJ2LS, 2022-11-20:
@pe1rrr can you confirm this is working? So we can close some issues. I also merged the big PR so we can soon test them over the air. If needed we can reopen or add further fixes, but then issue list is more clean

closed 2022-11-20 by DJ2LS

pe1rrr, 2022-11-21:
Ok I have some bad news regarding OTA testing, it doesn’t work, there’s something wrong with sequencing and callsign handling (I used two different callsigns, no SSID).
I’ll open a different issue for this.

Is it possible to eradicate the delay after the TNC is disconnected and the socket closing, because the socket remaining open for > 10s while the scripting attempting to re-attach the TNC for another retry will likely fail as the socket is still ‘busy’.

https://youtu.be/lTyZURY04VY - Demonstration

DJ2LS, 2022-11-21:
@pe1rrr yes, just open new issues for new problems so I can structure my work a little. 

I also want to create a new release so it can be tested and new issues can be found. 

The socket problem sounds really interesting to me as I was wondering why it occurs. But let's open new issues for this.

DJ2LS, 2022-11-21:
@pe1rrr , I‘m going to create the new release in about two days. Until then it can be tested from source. I’ve wrote you an email to your gmail address with the answer to your music question