DJ2LS/FreeDATA #290: TNC session connection establishment

pe1rrr, 2022-11-23:
Over the next few days John G8BPQ will be setting up his pi with the latest pull to help me work on the problem I have now establishing connections to the TNC from BPQ, once that is fixed I expect to be able to continue working on the issue of connections going into perpetual loop after a connection is established. 

While heartbeat packets are sent after connection, no actual data is acknowledged or received across the connection so something is stepping out of sequence here. I’ll post more details on this issue as it develops we can hopefully help get your upcoming build out with it working.

As it is right now, shortly after testing the TNC to TNC connections Over The Air, it was clearly apparent there is a bug in the ARQ handler.

👍

DJ2LS, 2022-11-23:
Great, @pe1rrr ! But I suggest also having a look at your network problems and why it is such delayed.  We might do this first 🤔

DJ2LS, 2022-11-23:
Also another important point: the actual session is just a session around an arq transmission which is not that efficient. Here it would be interesting to know, which data is going to be transmitted and whats the next layer application is doing. So we could redesign the connection and making it more efficient.

pe1rrr, 2022-11-24:
It seems that the connection doesn’t send the ‘connected’ message and jumps immediately into heartbeat mode so the TNCs will sit looping heartbeats until its force disconnected, can you confirm ?

DJ2LS, 2022-11-24:
I can try testing tomorrow when back at home after work. The actual „normal“ behavior is, that as soon as two stations are connected, a heartbeat will be sent.Questions:- So just the network message „connected“ is missing? - Is the heartbeat packet sent too often? Should we reduce this / increasing time between heartbeats? Simon Von meinem iPhone gesendet

DJ2LS, 2022-12-01:
> It seems that the connection doesn’t send the ‘connected’ message and jumps immediately into heartbeat mode so the TNCs will sit looping heartbeats until its force disconnected, can you confirm ?

Added more connection messages, available with next release

DJ2LS, 2022-12-24:
@pe1rrr maybe there are some issues, but I had to merge the corresponding PR. I'm reopening this

DJ2LS, 2023-01-16:
@pe1rrr meanwhile some bugs have been fixed ( again 😄 ) - does the problem still exists?

DJ2LS, 2023-05-08:
no answer yet. @pe1rrr feel free reopening this if it still exists

closed 2023-05-08 by DJ2LS