DJ2LS/FreeDATA #296: SSID: use base Callsign (AA0AA-0 -> AA0AA) for better readability

dk5sm, 2022-11-29:
In regard to the readability, I´d propose the usage of a base callsign without SSID (in case a user uses the default SSID -0 only).

So this means, that internally (protocol) the SSID -0 is always processed as any other SSID, but in the external presentation (e.g. in the chat and in the lastheard list), only the base callsign is displayed. Just for -0. Any other SSID´s should display as they already do.

DJ2LS, 2023-03-12:
@Mashintime any idea how we could do this inside GUI?
Maybe doing a Callsign check and stripping Callsign if SSID == -0 ?

Mashintime, 2023-03-12:
Sure it'd be easy to do, but would require extensive testing to ensure it doesn't break anything.

DJ2LS, 2023-10-16:
We could implement this into new gui. Keeping the SSID always attached in background but just change the displaying. 

When opening a new chat without attaching the SSID we should attach "-0" in background

closed 2024-01-06 by DJ2LS