DJ2LS/FreeDATA #305: Waterfall not displaying.

jpronans, 2022-12-22:
**Describe the bug**
TNC Debug message
calculate_fft: Exception: module 'numpy' has not attribute float

**To Reproduce**
Steps to reproduce the behavior:
Pop_OS 22.04, 

clean install of FreeDATA,
python version is 3.10.6

**Expected behavior**
Waterfall displayed

**Logfiles**
https://wiki.freedata.app/en/usage/logging

**Screenshots**

**Desktop (please complete the following information):**
Pop_OS 22.04
Toshiba Laptop.
Icom IC-7200

**Additional context**
I'm not seeing this message on my pi.  

Assume its a python3/python2/numpy issue

DJ2LS, 2022-12-22:
Thanks, @jpronans , I'm going to install PopOS to reproduce the issue. So I might need some days.

jpronans, 2022-12-22:
No hurry.. i'll have a poke myself, once I figure out how to get the remote GUI talking to the modem on the pi.

DJ2LS, 2022-12-22:
For remote operation:

- Start the daemon on your raspberry pi. 
- Look for the ip address with `ip a`
- start the GUI on your preferred computer and select on the top left the remote button, there enter the ip address of your pi. If it's not connecting, please restart the gui. This is a bug which will be fixed somewhen in near future. 
- Please note: rigctld must be started manually on your pi. Starting by gui is not supported yet for remote control. 

Running the tnc headless will become easier with merging the branch ls-arq. Then you can use a config file for easy starting the tnc on your pi.

jpronans, 2022-12-22:
Line 895 of Modem.py... I've changed the np.float to np.float32 and I've got a Waterfall.

DJ2LS, 2022-12-22:
Thanks for the possible fix 🎉 I'll have a look tomorrow after work.

DJ2LS, 2022-12-24:
merged with latest release

closed 2022-12-24 by DJ2LS