DJ2LS/FreeDATA #319: retry failed connection

DJ2LS, 2022-12-28:
Steps:

- On a failed transmission 
    - IRS stores already received data to buffer --> Now we know our buffer position
    - IRS stores crc information to buffer --> that's the minimum requirement, a fully received data header with crc information

- Next transmission:
    - ISS send channel opener
    - IRS send channel opener and buffer position (32bit int) and checksum (crc32) = 8bytes
    - ISS checks if these information are available against "0", then compares crc with own crc of data which will be sent. If equal override buffer position minus some bytes, depending on speed level,  which will be used as indicator for IRS, that we are continuing ( data will be overwritten in buffer then )

pe1rrr, 2022-12-28:
I am following all of this by email,  I don’t meant to patronise. Genuinely blown away with how much you have been putting in over Christmas. Really cool.

DJ2LS, 2023-06-16:
Not sure, if we need this, as auto retrying the entire message might be more stable

DJ2LS, 2024-02-08:
Closing this one as it's making more sense retrying entire data triggered by beacon

closed 2024-02-08 by DJ2LS