DJ2LS/FreeDATA #324: Many to Many chat.

jpronans, 2022-12-30:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
Multi-party chat. i.e. many to many.

DJ2LS, 2022-12-30:
Needed frames:

datac0 as "weakup"
datac3 as payload

Frame design:
Wakeup:
<1b frame type><3b chat-id><6b Callsign><1b mode?><3b message-id>
Payload:
<3b message-id>

sp2ong, 2022-12-31:
Nice idea to broadcast features that allow you to send asynchronous messages to one or all (group chat )

DJ2LS, 2022-12-31:
Yes, but its a future feature. Sending a broadcast isn't a big deal, yet. But I want to focus on protocol stability. So this feature needs to wait

DJ2LS, 2023-01-08:
Possible NORM oriented feature thoughts:

- ISS: data as stream, repeat it N times 
- ISS: Split data into chunks with number and max chunks 
- IRS: missing chunks requested in NACK frame 
- N Stations receiving NACK: transmit missing bursts 
- We need to make sure IRS must receive at least one burst from ISS, otherwise it would be Message forwarding which isn't permitted by default in some countries 
- We need to avoid packet collision somehow. Therefore we should use a mechanism for random delayed transmission like used for QRV frames 
- We need to hold last bursts in separate broadcast buffer for making them available for other stations 
-  We need to set N retries of sending a NACK
- Should we use a wake up frame or trying to run a data mode in parallel all the time - this should be possible with vectorisation of codec2 ofdm acquisition code

DJ2LS, 2024-09-27:
Meanwhile we have more options. 

We could design a mode, working on low SNR, holding 128bytes or 256bytes of data. 

Send super short wake-up burst, followed by payload data mode.

We might resume it 3 times.
Header maybe contains: 
1Byte identifier --- 6Bytes Destination --- 6Bytes Origin --- 4Bytes Timestamp ––– Payload 

So it acts like short messaging based on  broadcast allowing some basic group chat

DJ2LS, 2025-03-14:
Saving a NORM thought: 

Database structure:
- message ID
- group ID reference 
- origin Callsign
- timestamp 
- direction 
- message 
- message chunks {1:{}, 2:....}
- total chunks 
- statistics  
- priority? 
- is read 

- group ID
- group NAME