DJ2LS/FreeDATA #325: wrong byte order bug

DJ2LS, 2023-01-02:
Wrong byte order occurs.

- Burst "A" is getting lost - tnc runs into timeout
- TNC reduced speed level, receiving 2 bursts at lower speed level
- TNC increases speed level, datac1 enabled, old burst "A" will be decoded

Does this occur also with datac3?

```
2023-01-02 11:14:45 [info     ] [TNC] ARQ | RX | SENDING ACK   bytesperminute=1553 finished=3910
2023-01-02 11:14:45 [debug    ] [MDM] [demod_audio] Pushing received data to received_queue nbytes=512
2023-01-02 11:14:45 [debug    ] [MDM] worker_received: received data!
2023-01-02 11:14:46 [info     ] [MDM] calculate_snr:           snr=8.0
2023-01-02 11:14:48 [debug    ] [TNC] enqueue_frame_for_tx     c2_mode=sig0
2023-01-02 11:14:48 [debug    ] [MDM] TRANSMIT                 mode=14 payload=14
2023-01-02 11:14:49 [debug    ] [MDM] ON AIR TIME              time=0.6921732425689697
2023-01-02 11:14:49 [debug    ] [MDM] self.modem_transmit_queue qsize=0
2023-01-02 11:14:49 [debug    ] [TNC] send_data_to_socket_queue: jsondata={"freedata":"tnc-message","arq":"transmission","status":"receiving","uuid":"367e7a21-1169-4180-ae2e-2485cce697d7","percent":1,"bytesperminute":1473,"mycallsign":"EI7IG-0","dxcallsign":"EI3HIB-0","finished":4122}
2023-01-02 11:14:49 [debug    ] [TNC] process_data:            n_retries_per_burst=0
2023-01-02 11:14:49 [debug    ] [HLP] check_callsign: Checking: callsign=b'EI7IG-0'
2023-01-02 11:14:49 [debug    ] [HLP] check_callsign: Checking: callsign=b'EI7IG-0'
2023-01-02 11:14:49 [debug    ] [HLP] check_sessionid: Checking: check=b'\x01' ownid=b'L'
2023-01-02 11:14:49 [debug    ] [HLP] check_sessionid: Checking: check=b'L' ownid=b'L'
2023-01-02 11:14:49 [debug    ] [TNC] RX SNR                   snr=8.0
2023-01-02 11:14:49 [debug    ] [TNC] static.RX_BURST_BUFFER   buffer=[b'aa']
2023-01-02 11:14:49 [warning  ] [TNC] ARQ | RX | wrong byteorder received - dropping data
2023-01-02 11:14:49 [debug    ] Time left until timeout        seconds=179.39059329032898
2023-01-02 11:14:54 [debug    ] [MDM] [demod_audio] Pushing received data to received_queue nbytes=512
2023-01-02 11:14:54 [debug    ] [MDM] worker_received: received data!
2023-01-02 11:14:54 [debug    ] [TNC] process_data:            n_retries_per_burst=0
2023-01-02 11:14:54 [debug    ] [HLP] check_callsign: Checking: callsign=b'EI7IG-0'
2023-01-02 11:14:54 [debug    ] [HLP] check_callsign: Checking: callsign=b'EI7IG-0'
2023-01-02 11:14:54 [debug    ] [HLP] check_sessionid: Checking: check=b'\x01' ownid=b'L'
2023-01-02 11:14:54 [info     ] [MDM] calculate_snr:           snr=7.8
```

```
2023-01-02 11:17:01 [debug    ] Time left until timeout        seconds=178.49662137031555
2023-01-02 11:17:03 [debug    ] [TNC] enqueue_frame_for_tx     c2_mode=sig0
2023-01-02 11:17:03 [debug    ] [MDM] TRANSMIT                 mode=14 payload=14
2023-01-02 11:17:03 [debug    ] [MDM] ON AIR TIME              time=0.6764054298400879
2023-01-02 11:17:03 [debug    ] [MDM] self.modem_transmit_queue qsize=0
2023-01-02 11:17:03 [debug    ] [TNC] send_data_to_socket_queue: jsondata={"freedata":"tnc-message","arq":"transmission","status":"receiving","uuid":"367e7a21-1169-4180-ae2e-2485cce697d7","percent":6,"bytesperminute":2116,"mycallsign":"EI7IG-0","dxcallsign":"EI3HIB-0","finished":2813}
2023-01-02 11:17:03 [debug    ] [TNC] process_data:            n_retries_per_burst=0
2023-01-02 11:17:03 [debug    ] [HLP] check_callsign: Checking: callsign=b'EI7IG-0'
2023-01-02 11:17:03 [debug    ] [HLP] check_callsign: Checking: callsign=b'EI7IG-0'
2023-01-02 11:17:03 [debug    ] [HLP] check_sessionid: Checking: check=b'\x01' ownid=b'L'
2023-01-02 11:17:03 [debug    ] [HLP] check_sessionid: Checking: check=b'L' ownid=b'L'
2023-01-02 11:17:03 [debug    ] [TNC] RX SNR                   snr=9.7
2023-01-02 11:17:03 [debug    ] [TNC] static.RX_BURST_BUFFER   buffer=[b'\x92\x0b\xb5O\x7fz\x0b\xd4\xf6>\xd3\xf2\x7f\x14\xed\xc0\xc3p\x93\xf3O\xd3\x11d\xc6>[d\x9f\xe8\xda\xa9\xc6\x00\x9e%\xf8\xcfgs*\xba\x82\xf2m\x06\x87\x17X\x9d \x8e\xcb\xd7\xe4\xa5\xd0\x0bK\x19\x91O\xf5\xfd}\xb1w\xaa"\x14i\xcb\xf8K\xea\xbbF(\x1fw!\xe0\xacHj\x9cj\x95\x7f\xcc`s\xbe\x8a\xaa -m\x9e\xee\xc8\xc7\x9e\xfe$@\x1e\xde\x90\x06q[\xc8\x11\r\xee\x96/\xb1\xd9$\xc8\xc7\x9aU`\xfa\xc2A\x1c\x05\xbd\x12+\x1d\x9a@>\x93\x8bl\x81JZp\xe0\x90\xa4\xda\xcd\xe1\x9f\xd4\x7fK`2\xb9=\xdbv^\x17\x03%\x9fVGX|\xd2\xaa\xa4\x14S\x13\xd2\xae\xa8\r\xc0\xa1i@\xeaZ!\x00\xf0\xb5\xe6\xa6\x10\r\xea\xf0\xbb@\xea\xd2\x854bh \xf7\xa7\xc0\x02\xaa\xd2\x9f:\xd7>\xa0\xa43\x83\xa8\xec\x9aY)\x17!\x9c?K\ngP\xd8\x7f$x\xf4\xd2\xb6%\xb5+\xb7\xc1v#\x86Mp\xb6\xfc\xf4\xf0\x0b\x88\x15\x11\xec:\xc2\xb6\xb1Em\x92\xc5\'\n\xbaN\xc1\x93\xda\x97\xd6Ya\x9dla{\x89\x01\xe8u[:\x9e\xa1\xdf\x90\x19\x10q\x05\xdfI)\xc4!\xd1(\x9c\xb5\xdc\x99s\x90\xde"uTH\xc2j\xa1!7\xe8\x10\xe7\x08`\xf2\xf26\xf5\x92\xc5O\x0e:\xd2l\xce\xb8\xa3\xb8 L\x08DO\x0e\xd7]\x06\xe6\x13\xd4\xbf\xe4\x00\xc9\xa6\x99\xa4m?M5\r\xd0\xa5dNX\'4\xba\xaa\xda"<\xb5\x87\x9d\xe5\x86\xd0\x1a\xdd\xfa\xa9\xf1\x88\xac\x8f\x9b\xdd\xf9z\xfbpL\xaa#\xfa\xd5V\x87\x88\xd5\xa8\x0b&\xd2\x95m\xe5\x96aE\xf2/\xe0\x1c\x01\x04,x }\x96\xe6\xc3\x87$\x9aH\xc8j\xaa\xf3M+\xa2w\x00\xa0>\x10\x94\xbd)L7\x8f\x13(Y\x03y\x15\xc3-\xd5h\x907q\xb4<|M\x15\xe5^s]r\x1cM\xd1\xe2\xc4v\xb9\xfe\xe8\xed\xf7nMZ\x94n\xd1\xf1e\x19\xc8|\x91\xf1\xc5\xf1F\x07\xc8)5O\xe8\x97\x80\xcb']
2023-01-02 11:17:03 [warning  ] [TNC] ARQ | RX | wrong byteorder received - dropping data
2023-01-02 11:17:06 [debug    ] Time left until timeout        seconds=176.37912464141846
2023-01-02 11:17:08 [debug    ] [MDM] [demod_audio] Pushing received data to received_queue nbytes=512
2023-01-02 11:17:08 [debug    ] [MDM] worker_received: received data!```

drowe67, 2023-01-03:
My theory is the packet was part way being received when the timeout occurred.  Suggest monitoring `rx_status`, and don't timeout if a packet is part way through being received.

DJ2LS, 2023-01-03:
@drowe67 the timeouts are actually about 6-8 seconds. Enough time for the entire burst ( N=1 frames per burst ) for being transmitted, I think. 
Let's wait some time until we got reports with latest debugging logs with enable rx_status information.

DJ2LS, 2023-01-04:
> My theory is the packet was part way being received when the timeout occurred. Suggest monitoring `rx_status`, and don't timeout if a packet is part way through being received.

This comment makes me thinking about moving from time based "time out" to "state out" or combining them, so we could not getting stuck in a possible state loop. 
We could make time out as short as possible, say timeout = "N frames per burst" * "transmission length per burst", but waiting until we are back to rx_status == 0. By doing this we could drastically increase time out in case of PER, but don't running into time out while data will be sent and therefore rx_status !== 0.

DJ2LS, 2023-01-16:
Problem didn't occur with latest version and fixed "timeout bug" .

DJ2LS, 2023-01-28:
There occurs a byteorder bug again, but it's not covered by actual logging.

DJ2LS, 2023-02-24:
Might be fixed: Problem was speed level increasing with use case of already received data of lower speed level

DJ2LS, 2023-03-12:
Fix will be confirmed if we have 100 successful transmissions without CRC error.

closed 2023-03-19 by DJ2LS