DJ2LS/FreeDATA #329: List of GUI optimisations and features

DJ2LS, 2023-01-16:
## List of todo's 

### Main view 
- [ ] Title - show tnc IP
- [ ] Reduce CPU load for progress bars
- [ ] Update package.json to latest versions
- [ ] Move transmission progress bar to bottom of footer bar with full width of screen 
- [ ] Create small info box with transmission progress separated from progress bar 
- [ ] Enable / disable progress bars? Maybe a low cpu mode 
- [ ] Make IRS-ISS speed chart looking equal 
- [ ] Disable CQ and Ping button when pressed for a specific time 
- [ ] Different colours for waterfall 
- [ ] Theme for color blind people 
- [ ] Export speed chart as CSV
- [ ] NACKs as line for speed chart 

### Chat
- [ ] Send voice message 
- [ ] Send location with map view 
- [ ] Image preview for all images 
- [ ] Image previewbox 
- [ ] Quote a Message

g0hww, 2023-01-21:
Might it be possible to run the waterfall display with fewer FFT bins? Maybe half the current number?

DJ2LS, 2023-01-22:
Duplicate issue. Going to close this.

closed 2023-01-22 by DJ2LS