DJ2LS/FreeDATA #331: GUI optimisations and features

DJ2LS, 2023-01-17:
## Main
- [ ] Display tnc ip in title 
- [ ] Reduce progressbar CPU load -> move to chartjs 
- [ ] Move to Vue framework 
- [ ] Make waterfall color adjustable 
- [ ] Waterfall optimisations 
- [ ] Simplify code by using more functions 
- [ ] Use electron progress bar 
- [ ] Show version mismatch correctly 
- [ ] Move progressbar of transmission to tnc-message instead of state message 
- [ ] Notification in settings to restart the tnc 
- [ ] Remove Scrollbar from splash screen 
- [ ] Disable ping and CQ button when pressed, then show a spinner

Mashintime, 2023-01-17:
Thanks for taking the time to create this list.

DJ2LS, 2023-01-24:
https://www.npmjs.com/package/browser-image-compression

DJ2LS, 2023-01-28:
- [ ] low cpu mode for chat
- [ ] fix version and callsign on title for ubuntu 
- [ ] show callsign we are connected to
- [ ] display ping ACKs in chat
- [ ] https://github.com/DJ2LS/FreeDATA/issues/264
- [ ] https://github.com/DJ2LS/FreeDATA/issues/268
- [ ] https://github.com/DJ2LS/FreeDATA/issues/296
- [ ] https://github.com/DJ2LS/FreeDATA/issues/178
- [ ] https://github.com/DJ2LS/FreeDATA/issues/140
- [ ] https://github.com/DJ2LS/FreeDATA/issues/60
- [ ] fix send on enter not working always

DJ2LS, 2023-01-28:
@Mashintime feel free checking the already working features

Mashintime, 2023-01-28:
@DJ2LS Are you good with marking the progress bar CPU as done?

DJ2LS, 2023-01-28:
@Mashintime yes. We've made minor improvements in this area. Adjustments can be implemented at every time

jpronans, 2023-05-11:
What about "Sorting function for Heard Station List". I.e sort be last heard, callsign, distance or SNR?

DJ2LS, 2023-08-06:
> What about "Sorting function for Heard Station List". I.e sort be last heard, callsign, distance or SNR?

implemented

DJ2LS, 2023-08-06:
@Mashintime could you check the list above and check the ones we/you already implemented?

closed 2024-01-06 by DJ2LS