DJ2LS/FreeDATA #332: ALC level

frspin, 2023-01-17:
I am using FreeDATA with Ft897D and Pulse audio device.

When I transmit (Beacon, CQ or simply Tune test) I am unable to lower ALC level without lowering power level.
Same behavior using pavucontrol or transmit level control of GUI.

I am using same setup with Fldigi, Wsjt-X and JS8Call and in all this programs I am able to get a volume adjustment when ALC is off or only 1 or 2 but power don't decrease.
This is simply not possible with FreeDATA using Appimage or using latest version cloned with GIT

After starting rigctld I am starting TNC with:

cd tnc
python3 main.py --use-config=my_config.py

and, after this, 
cd gui
npx electron main.js

I am not able to do a "tune" or a "Beacon" tx without using GUI so I don't know if this a GUI or a TNC problem.
No errors in console and in terminal window where programs are started.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS] - Ubuntu Linux 22.04
 - Version [e.g. 22] 0.6.11-alpha.2
 - Platform [Raspberry Pi, Desktop]  Desktop PC
 - Python 3.10
 
Regards

Franco Spinelli
IW2DHW

frspin, 2023-01-18:
I have done a small Python program for sending beacon and test command to TNC without using the GUI.
Problem with ALC persists.

So the problem is in TNC and not in GUI

Is this a codec2 behavior or is this a bug?
Any other user is seeing this?

Regards
Franco Spinelli
IW2DHW

DJ2LS, 2023-01-18:
Hi @frspin  there's also a cli tool in tools folder.

I'm still not sure what's the problems origin as it isn't occurring for other users. So the feature itself is working. 
Did you try setting level to 0 and maximum and checking the output? 
On 0 there shouldn't be a audio output, on max it should be too much. You could also try testing it with audio loopbacks or using a audio cable for connecting microphone and speakers. 
Audio level depends on audio devices, so it's not equal or comparable with other levels. 

Again, thanks for your feedback! Every bug we can close is a good one 🙂

frspin, 2023-01-18:
Il 18/01/23 12:49, DJ2LS ha scritto:
> Hi @frspin <https://github.com/frspin> there's also a cli tool in tools 
> folder.
> 

Yes. I am started from your tool integrating Ping, Audio test, Audio 
Level, Cq, Start and Stop Beacon in a simple GTK gui.

> I'm still not sure what's the problems origin as it isn't occurring for
> other users. So the feature itself is working.
> Did you try setting level to 0 and maximum and checking the output?
> On 0 there [...]

I have added Audio level adjustment to my simple interface.
Now if I set level to 0 i get no output. 54 show RF power and no ALC, 
100 show same RF power and high ALC.

With this setup I get same result using your GUI started after TNC, as 
you suggested yesterday.

The remaining problem is low power shown in my MFJ 945E
With same rig and same rig setup (RF power set to 30W) I get only 20W on 
MFJ. With WSPR and other digital modes I get 30-35W.

Can it be due to codec2 modem?

Regards

Franco Spinelli
IW2DHW

DJ2LS, 2023-01-19:
@frspin 

https://github.com/DJ2LS/FreeDATA/blob/97143f25f6ad6a0574c918dfec1902cc93ea14e3/tnc/modem.py#L548

Instead of static.AUDIO_LEVEL try using a higher value. Maybe you need a much higher range of audio level

frspin, 2023-01-19:
Il 19/01/23 08:21, DJ2LS ha scritto:
> @frspin <https://github.com/frspin>
>
> https://github.com/DJ2LS/FreeDATA/blob/97143f25f6ad6a0574c918dfec1902cc93ea14e3/tnc/modem.py#L548 [...]

static.TX_AUDIO_LEVEL is the value in config file, AUDIO section, 
'txaudiolevel' key.

Increasing this value increase ALC but power remain at same value.

The actual value (54 for my rig) is the value for getting an ALC value 
low (1 or 2 notches) without reducing power level.

Regards

Franco Spinelli
IW2DHW

DJ2LS, 2023-01-19:
@drowe67 do you have any further thoughts about this?

frspin, 2023-01-21:
Some other test.
Using WSJT-X and WSPR I have transmitted with same rig, same power and same antenna.
Reception results as per WsprNet is attached to this message.
After this I fired up FreeDATA and send beacon.
Power measured by my power meter was approximately one half and nobody received my beacon.
EI7IG-0 and G0HWW where online and, according to WsprNet map, my WSPR signal is reaching UK

Modifying audio level don't change power level but only ALC level.

Regards
Franco Spinelli
IW2DHW

[image]

DJ2LS, 2023-01-21:
Hi @frspin please note, that WSPR and other super low SNR modes can't be compared with wide bandwidth ofdm modes. The protocol also needs a two way communication which is another difference compared to broadcasting modes. 

Did you increase the audio level as mentioned above within code to a higher value than 255 for example?

I'm not sure, but I think I already got a beacon from you, some weeks ago, so that's working okay I would say. Sure, the ALC level is important, but not as important as low EMI and good antennas, for example.

frspin, 2023-01-21:
> Hi @frspin <https://github.com/frspin> please note, that WSPR and other 
> super low SNR modes can't be compared with wide bandwidth ofdm modes. 
> The protocol also needs a two way communication which is another 
> difference compared to broadcasting modes.

Yes, I know. But something is changed in recent releases.

> 
> Did you increase the audio level as mentioned above within code to a 
> higher value than 255 for example?

Yes. After 20-25W only ALC is increasing. I have also used pavucontrol 
and increased audio level with this. Both adjustments have same effect

Can be a codec2 library problem? I have codec2 in tnc/lib/codec2 folder, 
as suggested in wiki. And codec2 library is changed in last months.

> 
> I'm not sure, but I think I already got a beacon from you, some weeks 
> ago, so that's working okay I would say. Sure, the ALC level is 
> important, but not as important as low EMI and good antennas, for example.

Also I remember this. And some months ago I was able to heard my beacons 
using and online receiver in Nederland and in UK.
Now I am unable to do this.

My setup, my rig and my antennas are unchanged in the last 2 o 3 years!

Regards

Franco Spinelli

DJ2LS, 2023-01-21:
Okay, Franco,

its really hard for me reproducing this, because the problem only occurs on your setup. 
Also there wasn't a change in audio processing part the last months which would affect this behavior. Codec2 is running really stable, so I don't think it's the problem, but we want to eliminate its is.

Okay, we need to make sure, things are running as expected so we have no caching error something like that: 

1) What's the version, the tnc displays on startup?
2) What's the version, the gui displays on top?
3) When setting the tune level to highest amount, what's the tnc responding in CLI ?
4) When setting the tune level to highest amount, what's ALC level displaying on your radio? 
5) Can you use a loopback device, for moving audio back to the tnc, when transmitting, so we can be sure, decoding is working ( and therefore codec2 ).
6) Can please you attach log files of the tests above?  https://wiki.freedata.app/en/usage/logging

Thanks, 

Simon

frspin, 2023-01-21:
Il 21/01/23 15:14, DJ2LS ha scritto:
> Okay, Franco,
>
> its really hard for me reproducing this, because the problem only occurs
> on your setup.
> Also there wasn't a change in audio processing part the last months
> which would affect this [...]

2023-01-21 16:30.10 [C2 ] Searching for libcodec2...
2023-01-21 16:30.10 [C2 ] Libcodec2 loaded 
path=lib/codec2/build_linux/src/libcodec2.so.1.0
2023-01-21 16:30.11 [CFG] logfile init             file=my_config.ini
2023-01-21T15:30:11.176142Z [info     ] [TNC] Starting FreeDATA 
author=DJ2LS version=0.6.11-alpha.2 year=2022

>  2. What's the version, the gui displays on top?

There is no version on top of GUI. I am starting TNC (main.py) with my 
config file before starting the GUI

This is SYSTEM INFORMATION at startup

16:30:11.425 (system information) › SYSTEM INFORMATION 
-----------------------------
16:30:11.429 (system information) › APP VERSION : 22.0.2
16:30:11.429 (system information) › PLATFORM    : linux
16:30:11.430 (system information) › ARCHITECTURE: x64
16:30:11.430 (system information) › FREE  MEMORY: 3079401472
16:30:11.430 (system information) › TOTAL MEMORY: 8011259904
16:30:11.431 (system information) › LOAD AVG    : 2.59,2.83,2.88
16:30:11.431 (system information) › RELEASE     : 5.15.0-58-generic
16:30:11.431 (system information) › TYPE        : Linux
16:30:11.432 (system information) › VERSION     : #64-Ubuntu SMP Thu Jan 
5 11:43:13 UTC 2023
16:30:11.432 (system information) › UPTIME      : 115032.29

And this is messages when GUI is connecting to TNC (without daemon)

16:30:13.937 (tnc)                › TNC connection established
(node:85248) MaxListenersExceededWarning: Possible EventEmitter memory 
leak detected. 11 end listeners added to [Socket]. Use 
emitter.setMaxListeners() to increase limit
(Use `electron --trace-warnings ...` to show where the warning was created)
(node:85248) MaxListenersExceededWarning: Possible EventEmitter memory 
leak detected. 11 connect listeners added to [Socket]. Use 
emitter.setMaxListeners() to increase limit
(node:85248) MaxListenersExceededWarning: Possible EventEmitter memory 
leak detected. 11 error listeners added to [Socket]. Use 
emitter.setMaxListeners() to increase limit

>  3. When setting the tune level to highest amount, what's the tnc
>     responding in CLI ?

2023-01-21T15:38:06.170056Z [debug    ] [SCK] CMD 
command={'type': 'set', 'command': 'tx_audio_level', 'value': '250'}

>  4. When setting the tune level to highest amount, what's ALC level
>     displaying on your radio?

70%. On Ft897D there is no numeric value.

>  5. Can you use a loopback device, for moving audio back to the tnc,
>     when transmitting, so we can be sure, decoding is working ( and
>     therefore codec2 ).

With Pulse I have a "null device" and a "monitor of null device"
Using this devices for I/O I can send a beacon ad decode the beacon

2023-01-21T15:50:34.356487Z [debug    ] [TNC] send_data_to_socket_queue: 
jsondata={"freedata":"tnc-message","beacon":"transmitting","dxcallsign":"None","interval":5,"mycallsign":"IW2DHW-0"}

2023-01-21T15:50:35.471236Z [debug    ] [TNC] send_data_to_socket_queue: 
jsondata={"freedata":"tnc-message","beacon":"received","uuid":"8e752949-83e3-4cad-8c83-bafa6482a4a9","timestamp":1674316235,"dxcallsign":"IW2DHW-0","dxgrid":"JN45OP","snr":"7.9","mycallsign":"IW2DHW-0"}

>  6. Can please you attach log files of the tests above?
>     https://wiki.freedata.app/en/usage/logging
>     <https://wiki.freedata.app/en/usage/logging>

See attached

Thanks

Franco Spinelli
IW2DHW

DJ2LS, 2023-01-21:
okay, @frspin please check the following line when running from source:

https://github.com/DJ2LS/FreeDATA/blob/0635d354437560e60745bae17826e77ad9b39aa2/gui/src/index.html#L1151

Please change "max" to a higher value, say 300 or more. Then try again with setting a higher value in gui. Maybe we can increase audio level for you by this. Please note, you need to restart the app.

frspin, 2023-01-21:
Il 21/01/23 17:06, DJ2LS ha scritto:
> okay, @frspin <https://github.com/frspin> please check the following
> line when running from source: [...]

No change setting this value to 400 and setting TX audio level to max.

For getting and indication of 30 W, as configured in the rig, I am 
required to set at 100% Pulse volume with pavucontrol.

After this I get 30 W and an ALC level approx of 75%

For reducing ALC level I need to lower FreeDATA level to 40-45 and power 
go down to 20 W

Done also a test using audio card without pulse. Values are changing 
but, for getting 30 W of power I get 75% of ALC al lowering ALC to 10% I 
lower power to 20 W.

Now 7063000 frequency is busy and in Explorer I can see only 3 active 
station other me. No station is receiving or decoding my beacon.

As a last test I will test another band if there is someone on 30 or 20 
meters.

Regards

Franco Spinelli
IW2DHW

DJ2LS, 2023-01-21:
Franco, why can't you set the value to 400?

frspin, 2023-01-21:
I have set this value to 400 but there is no change in power level without setting audio level to 100% in pavucontrol.
Using audio card without pulse maximum power of 30 W is reached with a value of 150 of this parameter.

Franco Spinelli
IW2DHW

DJ2LS, 2023-01-21:
Aaah, Franco, maybe pulse could be a problem? Portaudio is still waiting for pulse audio implementation.Can you use the regular device instead of using pulse for testing purposes?Von meinem iPhone gesendet

frspin, 2023-01-21:
Il 21/01/23 18:10, DJ2LS ha scritto:
> Aaah, Franco, maybe pulse could be a problem? Portaudio is still waiting 
> for pulse audio implementation.Can you use the regular device instead of 
> using pulse for testing purposes?Von meinem iPhone gesendetAm 21.01.2023 
> um 17:57 schrieb frspin ***@***.***>:﻿

Now that there is level adjustment in FreeDATA I can use regular device 
instead of pulse.

With 30 W on rig and value of 150 for parameter output, power is 
correct, ALC is very high but my beacon was received from DK5SM (-0.2 
dB). G0HWW (-2.1 dB) and EI7IG (-0.2 dB).

I will do some test with 40 W on rig but we need to test on 30m, if possible

Regards

Franco Spinelli
IW2DHW

drowe67, 2023-01-21:
> When I transmit (Beacon, CQ or simply Tune test) I am unable to lower ALC level without lowering power level. Same behavior using pavucontrol or transmit level control of GUI.

This is normal and expected.
 
> I am using same setup with Fldigi, Wsjt-X and JS8Call and in all this programs I am able to get a volume adjustment when ALC is off or only 1 or 2 but power don't decrease. This is simply not possible with FreeDATA using Appimage or using latest version cloned with GIT

You can't compare these waveforms directly, they are FSK, we use OFDM.

frspin, 2023-01-22:
Il 21/01/23 21:30, drowe67 ha scritto:
>     When I transmit (Beacon, CQ or simply Tune test) I am unable to
>     lower ALC level without lowering power level. Same behavior using
>     pavucontrol or transmit level control of GUI.
> 
> This is normal and expected.

If this is normal and expected is also correct an ALC level of 75% of 
full scale (my rig don't have a numeric indicator for ALC but only a 
graphic indicator).

There are signal distortions?
There are spurious signals?

Regards

Franco Spinelli
IW2DHW

DJ2LS, 2023-01-22:
Morning Franco,

If you have distortions or audio isn't like expected: did you see the record button? It will record audio. You could record and then do some transmission, then sending it to me so I can inspect them. Also important: audio also needs 48000Hz of sample rate!

frspin, 2023-01-22:
I don't know if I have distortions!

I only know that a high ALC level can cause distortion and spurious transmitted signal

Attached is a record for few beacons and a cqcqcq transmission, compressed as zip as required from group policy
[1674377304_audio_recording.zip](https://github.com/DJ2LS/FreeDATA/files/10474052/1674377304_audio_recording.zip)
.
Obviously all my audio are at 48000 Hz

Regards
Franco Spinelli
IW2DHW

DJ2LS, 2023-01-22:
Franco, I received your QRV with my small magnetic loop! My antenna is really bad, so it's nice hearing your signal here! Audio can't be that bad.

frspin, 2023-01-22:
Il 22/01/23 18:08, DJ2LS ha scritto:
> Franco, I received your QRV with my small magnetic loop! My antenna is 
> really bad, so it's nice hearing your signal here! Audio can't be that bad.

Now I have increased rig power level from 30 to 50 W
TX level is 120 for getting all 50 W and ALC is as usual al 75%
Happy to see that signal is not so bad.

My QRV transmitted was for DK5SM CQCQCQ

I have also received some ARQ data from EI7IG and G0HWW but, as far as I 
can see, there is no data received in directory (~/tmp) configured for 
received files. I suppose no ACK was received by then from my station.

How can they send an ARQ request and data?

Regards

Franco Spinelli
IW2DHW

DJ2LS, 2023-01-22:
Franco, only data which is addressed to your callsign will be processed. The other data wich is decoded and not for your station is just a debugging output. 
Broadcasts like CQ and QRV can be processed by every station

frspin, 2023-01-22:
Il 22/01/23 18:32, DJ2LS ha scritto:
> Franco, only data which is addressed to your callsign will be processed. 
> The other data wich is decoded and not for your station is just a 
> debugging output.
> Broadcasts like CQ and QRV can be processed by every station

As per messages in log I see:

   freedata: 'tnc-message',
   arq: 'transmission',
   status: 'received',
   uuid: 'e629b77d-1544-438f-9600-fcba6c34d1bd',
   timestamp: 1674408941,
   mycallsign: 'IW2DHW-0',
   dxcallsign: 'EI7IG-0',
   dxgrid: '------',
   data: 
'bQA7ATtzZW5kX21lc3NhZ2UAOwE7MAA7ATtmZDM0NmFiYS1mNjI5LTRkYTEtYTg1Ny0zNmM3OWFkODJkOWUAOwE7MTY3NDQwODgzOAA7ATt0ZXN0MgoAOwE7ADsBO3BsYWluL3RleHQAOwE7'
}

and, after this, many

2023-01-22T17:37:20.875118Z [debug    ] [MDM] [demod_audio] modem state 
mode=sig0-datac0 rx_status=3 sync_flag=--ST

And ACK or NACK messages send my my rig

Is this a transmission not to my call?

Regards

Franco Spinelli
IW2DHW

g0hww, 2023-01-22:
@frspin 

Hi Franco. I hope I managed to send a message of greetings to you today. It is a funny thing, but EI7IG and I both tried to send messages to you yesterday, Saturday, at exactly the same time, thereby causing some confusion in the TNC, as it does not yet have the ability to manage or prevent overlapping sessions. We managed to do exactly the same thing again today! After that, I think we both managed to send messages to you at different times. It may not have been obvious what was happening during these periods of silliness.

Regarding ALC setup, here is a quick summary of how things look with my system. The radio is an IC-7200. It is said that about 2 bars of ALC indication is about right for this radio model. I arrived at these settings more or less by accident. None of the values have been very deliberately chosen. This seems to work well when the path to Ireland is usable on 40m.

I have alsamixer set for a playback level of 31 on the CODEC audio device (disabled in Pulse audio). In the FreeData GUI i have tx level set to 234. IC-7200 USB level set in the menu to 100 (so all level control is external, in software).  RF Power set to 30W.  When I Transmit in Audio Tuning dialog, I see about 30W indicated on the IC-7200s power bar, 3 decreasing to 2 on the ALC bar, and about 8W average power on my external power meter.

I hope this might be useful.

frspin, 2023-01-22:
Il 22/01/23 20:46, Darren Long ha scritto:
> @frspin <https://github.com/frspin>
>
> Hi Franco. I hope I managed to send a message of greetings to you today.
> It is a funny thing, but EI7IG and I both tried to send messages to you
> yesterday, [...]

I am very new to FreeDATA and don't know if this message where received 
or not and, if received, where was stored.
Please let me know some info on this and how can try to send a message.

> Regarding ALC setup, here is a quick summary of how things look with my
> system. The radio is an IC-7200. It is said that about 2 bars of ALC
> indication is about right for this radio model. I [...]

No problem at all on receiving side. Audio level, as shown in GUI, is 
half of scale.
Problem is on transmitting side. I am using an Ft897D connected to a 
MFJ-945E and a fan dipole for 40/30/20 meters.
RF power is set in rig, now at 50W, yesterday at 30W.
If I set tx level in FreeDATA for a low ALC (2-3 bar) as I do with other 
programs, I get only 2/3 of power on my MFJ.
For reading full power (as I set it in rig) I need to increase tx level 
and power increase but also ALC increase to 75% of full ALC scale.

Despite this, my signal is decoded (almost for beacon).

Is this correct?

Regards

Franco Spinelli
IW2DHW

g0hww, 2023-01-22:
@frspin 
In the FreeData GUI near the top right hand corner is the RF Chat button. This will open a chat dialog window. Remote stations that have been heard will be listed. You can send messages to them, using separate ARQ sessions for each message. Messages received in ARQ sessions from other stations will be listed also.. 

Regarding your transmit levels, this is probably to be expected. David Rowe is clearly the expert in this area, but my limited understanding is that the OFDM modulation scheme has a high Peak to Average Power Ratio (PAPR).  The transmit power dialled into the radio limits the peak power, i suppose.  The audio drive levels you set to obtain the desired minimal ALC action in the radio, to avoid distortions, etc. and the relatively low average power you measure on your meter, is because of the high PAPR.

Hope this helps.

jpronans, 2023-01-22:
@frspin I sent you a message earlier, did you receive it? Your audio sounded fine at the time. Also, looking at the 897 handbook, it says 7-8 'dots' of ALC is what you should be aiming for in AFSK modes.

drowe67, 2023-01-23:
> Regarding ALC setup, here is a quick summary of how things look with my system. The radio is an IC-7200. It is said that about 2 bars of ALC indication is about right for this radio model. I arrived [...]

Actually I run an IC-7200 too.  The general advice ([FreeDV manual](https://github.com/drowe67/freedv-gui/blob/master/USER_MANUAL.md#sound-card-levels)) is adjust the audio drive level (via the laptop of radios USB level) to "just tickle the ALC".  With the radio cranked up to 100W I get roughly 40Wrms on the radios power meter (and an external LED bar graph power meter).  With compression enabled (the default) the peak to average ratio of the FreeDATA waveforms is about 4dB so 40Wrms for a 100W PEP radio is about right.

However I'm aware of other hams who adjust for "max power" and TBH no one has really sat down and systematically explored this.  The only thing I would avoid is cranking up the Tx audio level so far that the input waveform is really distorted.

Also please note power meters can vary wildly with OFDM as the waveform (like speech) has occasional peaks that confuse them.

frspin, 2023-01-23:
Il 23/01/23 08:26, drowe67 ha scritto:
> Actually I run an IC-7200 too. The general advice (FreeDV manual
> <https://github.com/drowe67/freedv-gui/blob/master/USER_MANUAL.md#sound-card-levels>) is adjust the audio drive level (via the [...]

So, if I have understood:

I can set tx level for a low ALC indicator and read on MFJ-945E average 
value of RF power 4 dB less than max power.

Alternatively I can set tx level for reading on MFJ-945E max power, 
increasing ALC level at risk of distorted waveform.

It is possible to test it on-line with someone receiving my signal?

Regards

Franco Spinelli
IW2DHW

DJ2LS, 2023-01-23:
Franco, for testing I suggest having a look here 

https://explorer.freedata.app/

Looks promising in your case 🙂

frspin, 2023-01-23:
Il 23/01/23 10:39, DJ2LS ha scritto:
> Franco, for testing I suggest having a look here
> 
> https://explorer.freedata.app/ <https://explorer.freedata.app/>
> 
> Looks promising in your case 🙂

I am already using this.

I meant test with different adjustments and get s/n received.

But I can try it, if 40 meters band allows it!

Franco Spinelli
IW2DHW

DJ2LS, 2023-01-23:
Okay, Franco, then we will wait for 40m.

There's also a Discord Server some hams are active there. Feel free joining the party! https://discord.gg/A84dBrXy

frspin, 2023-01-23:
Il 23/01/23 11:49, DJ2LS ha scritto:
> Okay, Franco, then we will wait for 40m.

Found a setting for tx level with low ALC and power indication approx 
30W RMS with 50 W configured in the rig (PEP)

With this setup I have a chat with DL1DJH exchanging messages and a 1K file

So this issue can be considered resolved.

> 
> There's also a Discord Server some hams are active there. Feel free 
> joining the party! https://discord.gg/A84dBrXy <https://discord.gg/A84dBrXy>

Joined

Thanks

Franco Spinelli
IW2DHW

closed 2023-01-25 by DJ2LS