DJ2LS/FreeDATA #338: Emoji/smiley/emoticon chooser in chat window can get stuck and will not close

g0hww, 2023-01-22:
**Describe the bug**
It is possible to get the smiley chooser stuck open in the chat dialog. Interaction with the chat dialog becomes awkward. It is possible, but not easy, to select another station from the list on the left. The main window is unaffected. CPU load is unaffected. Station seems to function normally.

**To Reproduce**
Steps to reproduce the behavior:
1. I did insert a smiley into a message I had written, but it didn't go where i wanted. I had set the character insertion point a few words back in the message. 
2. i tried to get out of the situation but couldn't make the smiley chooser close. 
3. now i have clicked on a different station and added smileys into a message i didn't want to write, to a station I didn't want to send a message to. 
4. closing and reopening the chat window doesn't help. 
5. The smiley chooser seems to be permanently stuck on the screen

eenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Pop!_OS Linux
 - Version 22.04 LTS
 - Platform x86_64

**Additional context**
Perhaps hitting Esc could make the smiley chooser close, if there are no better ideas. I'm not a GUI expert so I don't know what is normal. Perhaps mouse clicking outside the chooser, too. Maybe closing the chat dialog.
[image]

DJ2LS, 2023-01-23:
@Mashintime the chat needs some big improvements. As soon as we have reduced cpu load for the main gui, we need to adjust the chat

Mashintime, 2023-01-23:
Not able to reproduce, but if it happens again; does clicking the smiley icon (the one that opens the menu) also close the menu?

DJ2LS, 2023-01-24:
Yes, it does. But we need to add further close actions like we know from modern instant messengers. 
Maybe we should do some brainstorming,first

DJ2LS, 2023-01-31:
@g0hww can you test this? Just git pull from main.

When clicking on an emoji, the emoji window should close now.

g0hww, 2023-02-01:
I can now dismiss the smiley chooser by selecting a smiley. That seems to work.
However, once I have done that, and hit enter to transmit the message, it is not sent. The pending message is handled by the UI as if it was sent and moves up into the chat history.
I can then force it to be sent by clicking the little circular arrow to the left of the message in the chat history.
If i enter a message and insert no smiley characters, the message is sent immediately when i press Enter, as expected.
If I add a smiley into a message and then delete it, and press Enter, the message is sent immediately when i press Enter, as expected.
I have performed these actions many times and confirm that this is not a result of a busy channel or some other system state.

DJ2LS, 2023-02-01:
I can confirm this behaviour as I had it as well. @Mashintime we could close the emoticon chooser within „return listener“ as well and have a look there if this handler is getting called or not in this case

Mashintime, 2023-02-02:
Yes, was able to reproduce.  I believe it's related to the btoa function.  Investigating.

DJ2LS, 2023-02-02:
The btoa function is a critical area. Maybe we need some minor redesign of the chat part. But I’m not sure. At least simplifying it would help I think. But that’s a separate step with a huge amount of work I think

Mashintime, 2023-02-02:
The btoa function is throwing an error since the emoji is unicode(?--whatever char code it is is greater than a byte).  The resend doesn't use btoa so it works.  Still trying to come up with fix that works.

DJ2LS, 2023-02-02:
@Mashintime important side-node: When resending data, it will be processed

Mashintime, 2023-02-02:
@DJ2LS yes, that's the problem I'm running into, the receiving end isn't decoding the emojis correctly.

DJ2LS, 2023-02-02:
This would explain, why we had some lost messages in past.

Mashintime, 2023-02-04:
I was able to reproduce the window becoming stuck open, and now know why.  So I'll be reverting my attempted fix of closing the window after selecting an emoji.

PR #352 (2023-02-06): GUI Optimization
- Added settings for auto tune and explorer stats
- Further tweaks to low graphics mode
- Reworked footer, now responsive
- 'Connected to' field to show who you're connected with
- Further code clean up and optimizations; probably nothing noticeable
- Fixed #338
closed 2023-02-06 by DJ2LS