DJ2LS/FreeDATA #340: wrong locator in heard-list while connected to next station

dk5sm, 2023-01-24:
Yesterday I noticed the following, presumably reproducible behaviour (0.6.12.-a.6, Windows):

There was a chat transmission to DL1DJH (1st connect, successful). Afterwards also successfully a chat to IW2DHW (2nd connect). During the connect to Italy I noticed that the displayed locator in the heard list is the same as for DL1DJH (JN47SU, although in the chat window for IW2DHW -> JN50OP is displayed).

After connecting to G0HWW (3rd connect), displaying the wrong locator in the heard list was repeated: G0HWW was listed with the *correct* locator of IW2DHW (see picture).

A caching problem?

[image]

Mashintime, 2023-01-25:
I assume it's coming from the TNC, but could be the GUI.  Not sure if the data sent by the TNC gets logged when it sends a TNC state message to the GUI.  @DJ2LS thoughts?

DJ2LS, 2023-01-25:
Problem mit be using a dx grid global:

`        helpers.add_to_heard_stations(
            static.DXCALLSIGN,
            static.DXGRID,
            "DATA-CHANNEL",
            snr,
            static.FREQ_OFFSET,
            static.HAMLIB_FREQUENCY,
        )`

We should change this where possible or then better disable the heard station list for avoiding confusion

DJ2LS, 2023-01-26:
@dk5sm should be fixed with ls-arq merge

closed 2023-01-26 by DJ2LS