DJ2LS/FreeDATA #342: PTT problem with version cloned from GIT

frspin, 2023-01-27:
Using latest version cloned today from GIT with my Ubuntu 22.04 and my Ft897D.

When doing Beacon or CQ, transmit begins ok but PTT remain ON

Using Appimage all is working corrected, using same rig, same rigctld command and same audio card.

What is difference from Appimage 0.6.13-alpha1 and version cloned from GIT?

Regards

Franco Spinelli
IW2DHW

DJ2LS, 2023-01-28:
Hi Franco, 

please check if you're on "main" branch by using `git status` and then `git checkout main `

frspin, 2023-01-28:
I am on main branch.

git status
Sul branch main
Il tuo branch è aggiornato rispetto a 'origin/main'.

And the problem is present also with a fresh clone from GIT

Regards
Franco Spinelli
IW2DHW

DJ2LS, 2023-01-28:
Franco, sometimes it occurs there's a caching error with git. But in case of a new gig clone it should work. 

Can you attach a small log file of the area where the problem occurs and getting stuck with PTT?
Are there any errors shown in cli ?

frspin, 2023-01-28:
I am running with a proc that start rigctld, tnc with my config file and gui.

This is what is shown in cli:

2023-01-28 09:37.20 [C2 ] Searching for libcodec2...
2023-01-28 09:37.20 [C2 ] Libcodec2 loaded         path=lib/codec2/build_linux/src/libcodec2.so.1.0
2023-01-28 09:37.21 [CFG] logfile init             file=my_config.ini
2023-01-28T08:37:21.130510Z [info     ] [TNC] Starting FreeDATA        author=DJ2LS version=0.6.13-alpha.1 year=2022
2023-01-28T08:37:21.131715Z [debug    ] [C2 ] Create 48<->8 kHz resampler
2023-01-28T08:37:21.132271Z [debug    ] [C2 ] Creating audio buffer    size=9600
2023-01-28T08:37:21.132674Z [debug    ] [C2 ] Creating audio buffer    size=9600
2023-01-28T08:37:21.133314Z [debug    ] [C2 ] Creating audio buffer    size=9600
2023-01-28T08:37:21.133805Z [debug    ] [C2 ] Creating audio buffer    size=9600
2023-01-28T08:37:21.134113Z [debug    ] [C2 ] Creating audio buffer    size=9600
2023-01-28T08:37:21.134441Z [debug    ] [C2 ] Creating audio buffer    size=9600
2023-01-28T08:37:21.138303Z [info     ] [MDM] init: opened audio devices
2023-01-28T08:37:21.138544Z [debug    ] [MDM] init: starting pyaudio callback
2023-01-28T08:37:21.152772Z [info     ] [RIGCTLD] Connected PTT instance to rigctld! ip=127.0.0.1 port=4532
2023-01-28T08:37:21.153263Z [info     ] [RIGCTLD] Connected DATA instance to rigctld! ip=127.0.0.1 port=4532
09:37:21.316 (system information) › SYSTEM INFORMATION  -----------------------------  
09:37:21.320 (system information) › APP VERSION : 20.3.11
09:37:21.320 (system information) › PLATFORM    : linux
09:37:21.321 (system information) › ARCHITECTURE: x64
09:37:21.321 (system information) › FREE  MEMORY: 3053416448
09:37:21.322 (system information) › TOTAL MEMORY: 8011268096
09:37:21.322 (system information) › LOAD AVG    : 0.8,0.71,0.76
09:37:21.322 (system information) › RELEASE     : 5.15.0-58-generic
09:37:21.323 (system information) › TYPE        : Linux
09:37:21.323 (system information) › VERSION     : #64-Ubuntu SMP Thu Jan 5 11:43:13 UTC 2023
09:37:21.323 (system information) › UPTIME      : 2911.19
09:37:21.324 (system information) › CONFIG VALIDATION  -----------------------------  
09:37:21.325 (system information) › FOUND SETTTING [tnc_host]: 127.0.0.1
09:37:21.325 (system information) › FOUND SETTTING [tnc_port]: 3000
09:37:21.325 (system information) › FOUND SETTTING [daemon_host]: 127.0.0.1
09:37:21.325 (system information) › FOUND SETTTING [daemon_port]: 3001
09:37:21.326 (system information) › FOUND SETTTING [mycall]: IW2DHW-0
09:37:21.326 (system information) › FOUND SETTTING [mygrid]: JN45OP
09:37:21.326 (system information) › FOUND SETTTING [radiocontrol]: rigctld
09:37:21.326 (system information) › FOUND SETTTING [hamlib_deviceid]: RIG_MODEL_FT897D
09:37:21.327 (system information) › FOUND SETTTING [enable_hamlib_deviceport]: true
09:37:21.327 (system information) › FOUND SETTTING [hamlib_deviceport]: /dev/ttyS4
09:37:21.327 (system information) › FOUND SETTTING [enable_hamlib_stop_bits]: False
09:37:21.327 (system information) › FOUND SETTTING [hamlib_stop_bits]: 1
09:37:21.327 (system information) › FOUND SETTTING [enable_hamlib_data_bits]: False
09:37:21.327 (system information) › FOUND SETTTING [hamlib_data_bits]: 8
09:37:21.328 (system information) › FOUND SETTTING [enable_hamlib_handshake]: true
09:37:21.328 (system information) › FOUND SETTTING [hamlib_handshake]: None
09:37:21.328 (system information) › FOUND SETTTING [enable_hamlib_serialspeed]: true
09:37:21.328 (system information) › FOUND SETTTING [hamlib_serialspeed]: 38400
09:37:21.328 (system information) › FOUND SETTTING [enable_hamlib_pttprotocol]: true
09:37:21.329 (system information) › FOUND SETTTING [hamlib_dtrstate]: OFF
09:37:21.329 (system information) › FOUND SETTTING [enable_hamlib_dtrstate]: true
09:37:21.329 (system information) › FOUND SETTTING [hamlib_pttprotocol]: RIG
09:37:21.329 (system information) › FOUND SETTTING [enable_hamlib_pttport]: False
09:37:21.329 (system information) › FOUND SETTTING [hamlib_pttport]: /dev/ttyS4
09:37:21.330 (system information) › FOUND SETTTING [hamlib_dcd]: 
09:37:21.330 (system information) › FOUND SETTTING [hamlbib_serialspeed_ptt]: 9600
09:37:21.330 (system information) › FOUND SETTTING [hamlib_rigctld_port]: 4532
09:37:21.330 (system information) › FOUND SETTTING [hamlib_rigctld_ip]: 127.0.0.1
09:37:21.330 (system information) › FOUND SETTTING [hamlib_rigctld_path]: /usr/bin/rigctld
09:37:21.331 (system information) › FOUND SETTTING [hamlib_rigctld_server_port]: 4532
09:37:21.331 (system information) › FOUND SETTTING [spectrum]: waterfall
09:37:21.331 (system information) › FOUND SETTTING [tnclocation]: localhost
09:37:21.331 (system information) › FOUND SETTTING [enable_scatter]: False
09:37:21.331 (system information) › FOUND SETTTING [enable_fft]: True
09:37:21.331 (system information) › FOUND SETTTING [enable_fsk]: False
09:37:21.332 (system information) › FOUND SETTTING [low_bandwidth_mode]: False
09:37:21.332 (system information) › FOUND SETTTING [theme]: default
09:37:21.332 (system information) › FOUND SETTTING [screen_height]: 776
09:37:21.332 (system information) › FOUND SETTTING [screen_width]: 1092
[... 57 lines cut ...]
closing all windows.....
09:37:51.037 (main)               › closing sub processes
09:37:51.038 (main)               › closing tnc and daemon
09:37:51.063 (main)               › quitting app

Today log attached

[tnc.log](https://github.com/DJ2LS/FreeDATA/files/10526806/tnc.log)

Appimage is started with a proc like GIT version, starting rigctld and Appimage

Two more small problems, not related to this:

- please rotate also tnc.log as you do for main.log and renderer.log
- is this config section required or can be simplified?
  [AUDIO]
#audio settings
rx = USB Device 0x41e:0x30d3: Audio (hw:5,0)
tx = USB Device 0x41e:0x30d3: Audio (hw:5,0)

(hw:5,0) can change after a reboot or an insertion of another USB device

Regards
Franco Spinelli
IW2DHW

frspin, 2023-01-28:
Another test done:

Starting daemon.py and gui instead of starting main.py with my config file work OK

But there is no difference between my config file and config.ini in tnc folder. Only audio device is changed

Regards
Franco Spinelli
IW2DHW

Mashintime, 2023-01-28:
@frspin there were some changes pushed to main recently.  I'd recommend doing the git checkout as suggested earlier to see if you still have this issue.

If manually starting the TNC, only start daemon.py; it in return controls main.py.

I typically, launch daemon, then launch gui.

Let us know if the issue reoccurs with the updated git pull.

frspin, 2023-01-28:
Il 28/01/23 14:00, Mashintime ha scritto:
> @frspin <https://github.com/frspin> there were some changes pushed to 
> main recently. I'd recommend doing the git checkout as suggested earlier 
> to see if you still have this issue.

Already done but this don't solve the problem.

> 
> If manually starting the TNC, only start daemon.py; it in return 
> controls main.py.
> 
> I typically, launch daemon, then launch gui.

A suggested by DJ2LS I am starting directly main.py with my config.ini 
file. This because the GUI changed my audio setup at each start.

Now GUI seems correctly remember my audio setup. But Gui store audio 
device number, not audio device name. And device number are changing!

Starting rigctld, daemon.py and GUI is working correctly
Starting rigctld, main.py and GUI no!
So there is some changes to main.py or to daemon.py when start main.py

Regards
Franco Spinelli
IW2DHW

frspin, 2023-01-28:
Just updated to 0.7.1-alpha.1 and the problem seem vanished.
When frequency 7053000 become free I will do some more test

Regards

Franco Spinelli
IW2DHW

closed 2023-02-06 by DJ2LS