DJ2LS/FreeDATA #350: als auto tune

DJ2LS, 2023-01-31:
ALC auto-tune feature:
checking for ALC  - if too low, increase audio level

Todo: check why audio level is str and not int

DJ2LS, 2023-03-19:
Latest state is, audio level will be reduced to total silence...bad..needs a fix

DJ2LS, 2023-08-21:
In general this is a tricky feature as it depends on hamlib. So I'm not sure about the effort we should spend for this...
@dk5sm @LA3QMA  @Mashintime what do you think about this?

LA3QMA, 2023-08-22:
I guess this is a tricky part. i-e Elecraft KX3 requires the ALC to be 4-5 in digimodes and regular mode its 0.
It also have two different DIG modes.

Not sure how hamlib support this on different radios.

Can we use a different approach to get the same result?
i.e transmit a tone and increase the level and then read the power out?

DJ2LS, 2024-02-17:
It feels a bit like we should remove this feature as it's causing more problems. Better working on a reliable tuning mechanism

jpronans, 2024-03-03:
Maybe with some defaults for known/tested radios?

DJ2LS, 2024-03-03:
I also tend to remove this feature as it’s really depending on the corresponding radio. Von meinem iPhone gesendet

closed 2024-03-13 by DJ2LS