DJ2LS/FreeDATA #362: QRM avoidance ideas

g0hww, 2023-02-15:
**The problem**
Often the HF channel is busy with  other "traffic", i.e. perhaps RTTY or off-frequency opposite sideband. Such QRM may only occupy a small region of the full 1700Hz pass band, but it can prevent successful ARQ operation by either:
a) Causing pathological backoff due to a busy channel condition, that eventually results in a timeout, either at IRS or ISS or both.
b) Coinciding in frequency and time and "clobbering" traffic from IRS or ISS or both.

**Background info**
I would like to propose some related ideas aimed at mitigating these kind of issues.
Both ideas involve dividing thinking of the 1700Hz passband as 3 distinct sub-channels.
They are formed by the 4 sub-bands separated by vertical overlay lines in the GUI waterfall.
Let us call those sub-channels A, B and C, from left (LF) to right (HF). I think they are each around 566Hz wide, totalling around 1700Hz.

The narrow modem waveform currently can occupy sub-channel C, while the wide waveform occupies all three sub-bands.

**1. Idea for an improvement to the Busy Channel detection feature.**
My first proposal is that the Busy Channel detection feature be modified to only consider the set of sub-bands needed for a pending transmission.  If the pending transmission is to use the narrow waveform, then only the energy in sub-band B needs to be considered. If the pending transmission is to use a wide waveform, then all three sub-band must be assessed.

**2. Idea for IRS consideration of sub-band state when choosing the mode for the next ISS transmission.**
The ISS could consider the busy state of the 3 sub-bands and command the ISS to use narrow mode if either sub-band A or C have more energy than B, as measured over some interval. I expect that the view from the ISS might be different though, due to hidden transmitter, etc.

**3. Idea for narrow mode sub-band agility**
This idea might be a bit harder to implement, but if we are able to implement 1 and 2 above, then might it also be possible to choose to operate the narrow waveform in sub-channels A or B or C dynamically, depending on the state of the 3 sub-channels. There are two things to consider here,
a) Which sub-band will cause the least amount of back off due to busy channel detection affecting both ISS and IRS.
b) Which sub-band produces the best SNR for the ISS or signals.
I suppose that the ISS might use all 3 in wide mode, and the ISS could use whichever sub-band appears to be best from a channel busy and/or reported SNR perspective.

DJ2LS, 2023-02-27:
@g0hww Didn't forget this one. Reading again and again over it. Thinking from time to time about it.

DJ2LS, 2023-11-18:
1 and 2 are already implemented, 3 needs more effort as we need audio pitch shifting.