DJ2LS/FreeDATA #364: Revoke message - remote delete

Mashintime, 2023-02-15:


Mashintime, 2023-02-15:
I don't think is something we should do.  A station who could hear both sides of the conversation could be a bad actor and do this.  We would need some way of verifying a remote station is really who they say they are.

DJ2LS, 2023-02-16:
Yea, message signing is an important area somewhen. Not sure how to realise this efficient

DJ2LS, 2023-03-12:
@Mashintime @dk5sm  I'm in general not sure if this feature would work that well for HF. Even with message signing. What do you think? Should we close this issue/feature?

DJ2LS, 2023-08-21:
@Mashintime with message signing enabled, we could now work on this I think.

on ISS:
-- Additional button next to message: "revoke"
-- Change color of message and make the message itself filled with placeholders so its clear its revoked
-- find a way of making sure, the revoke command has been sucessfully transmitted to IRS

on IRS:
-- check if message is signed
----- if yes, mark message as revoked and make it filed with text placeholders
----- if no, do nothing and ignore command

DJ2LS, 2024-09-17:
I think we should close this one, @Mashintime  

Not sure if it's an good idea doing this as it can cause some trouble. Also message signing would be needed. But this needs to be implemented first

closed 2024-09-21 by DJ2LS