DJ2LS/FreeDATA #371: winlink KISS  / TCP connectivity

SpudGunMan, 2023-02-28:
**Describe the solution you'd like**
enable a data mode/pipe to be used as a TNC KISS on AX.25 encapsulation. (for winlink and TNC-chess)

**Describe alternatives you've considered**
vara

DJ2LS, 2023-03-01:
@SpudGunMan thank you for opening this steatite request, this is a planned feature and I'm thinking from time to time about it. Nevertheless focus is at the moment on making basic operation on HF more stable. 

Personally I'm thinking about a TCP/IP bridge, but not exactly sure about its implementation details. 

Further comments and thoughts and use-cases are appreciated, as well as feedback.

Simon

SpudGunMan, 2023-03-03:
off hand to hit the ground running? KISS and AGW packet compatibility via advertised TCP port would be the "big win" allowing use of just about every ham app-mode today with little or no changes to them.

I would keep the dream big and ignore politics, having a symbol rate limit on/off or just a warning you need to check local laws before use dialog may suffice? for the VHF question?

DJ2LS, 2023-03-05:
@SpudGunMan I did some research about KISS and AX.25. 

- It is possible I think, implementing KISS into the actual socket, so it can handle both - JSON encoded strings and KISS as well. 
- There are also some AX.25 libraries I could use, so I just need to get more familiar with the protocol. Where I also might need some advice is how ARQ behaves/interacts with AX.25
- All of this is a future topic, but I can have a look from time to time.

DJ2LS, 2023-09-30:
We need to check for another way of transmitting data. Its codec2 related topic as well I think

SpudGunMan, 2023-10-01:
I agree, and I assume M17

DJ2LS, 2023-10-01:
Yes, the modem related use case is a different one for packet. Not sure if it’s making sense if there are alternatives like M17

DJ2LS, 2023-10-27:
Closing this for now. Nevertheless the IP over HF topic is still in mind. But the KISS topic isn't making that much sense actually. There are plans of using an additional mode with different FEC which would fit for VHf. But that's a future topic and needs more time.

closed 2023-10-27 by DJ2LS